wmbusmeters/wmbusmeters #72: iM871A-USB and wmbusmeters capturing problems

diegomjb94, 2020-02-06:
Hi @weetmuts,

I am trying to collect the data from a "Qundis Q caloric 5.5" to a RaspberryPi 4. I am using the iM871A-USB dongle from the IMST company. I had installed the wmbusmeters as a daemon. It seems that the dongle is detected by the software:

`~ tail -f /var/log/wmbusmeters/wmbusmeters.log`
`(wmbusmeters) logging started 2020-02-06 11:18:03`
`(wmbusmeters) waiting for telegrams`

However, it does not receive any message. In the beginning, I thought that the problem was the "key" value in the configuration of the meter file in path `/etc/wmbusmeters.d/` since I do not know the value of the "key". So, I tried to isolate the error. Firstly, I tried to check that the "id" value was the correct one; hence, I run the following commands:

`~ wmbusmeters --listento=t1 /dev/im871a_0`
`No wmbus device found! Exiting!`
`~ wmbusmeters --listento=c1 /dev/im871a_0`
`No wmbus device found! Exiting!`

As can be seen, no wmbus device was found. So, what is the problem? Does "Qundis Q caloric 5.5" have different types of telegrams (no C1 or T1)? Do I need the "key" value to receive the telegrams? 

I am extremely new in this area and I am a bit lost. Thanks in advance.

Greetings,

Diego

weetmuts, 2020-02-06:
Hi! The daemon has probably taken the device so you cannot manually listen to it.
First `sudo systemctl stop wmbusmeters@-dev-im871a_0.service`
Now the device /dev/im871a_0 i free and you can do:
`wmbusmeters --listento=c1 /dev/im871a_0`

diegomjb94, 2020-02-10:
Hi again and thanks for your answer. But still, no wmbus devices were found. What is generating this problem?

Greetings,

Diego

weetmuts, 2020-02-10:
Try: mbusmeters - - debug --listento=c1 auto
and post the output here.

diegomjb94, 2020-02-10:
The output for the command `wmbusmeters --debug --listento=c1 auto` is the following one:

`(wmbusmeters) version:` 
`(config) using device: auto`
`(config) number of meters: 0`
`(detect) "auto" ""`
`(detect) driver: auto`
`(im871a) exists? /dev/im871a`
`(im871a) exists? /dev/im871a_0`
`You are not in the same group as the device /dev/im871a_0`
`(serial) stopping manager`

weetmuts, 2020-02-12:
Add your linux user to the wmbusmeters group:
usermod -a -G wmbusmeters diegomjb94

diegomjb94, 2020-02-12:
After running the command `usermod -a -G wmbusmeters username` and rerunning the command `wmbusmeters --debug --listento=c1 auto` the output is the following one:

`(wmbusmeters) version:` 
`(config) using device: auto`
`(config) number of meters: 0`
`(detect) "auto" ""`
`(detect) driver: auto`
`(im871a) exists? /dev/im871a`
`(im871a) exists? /dev/im871a_0`
`(im871a) checking /dev/im871a_0`
`(im871a) exists? /dev/im871a_1`
`(im871a) exists? /dev/im871a_2`
`(im871a) exists? /dev/im871a_3`
`(im871a) exists? /dev/im871a_4`
`(im871a) exists? /dev/im871a_5`
`(im871a) exists? /dev/im871a_6`
`(im871a) exists? /dev/im871a_7`
`(im871a) exists? /dev/im871a_8`
`(amb8465) exists? /dev/amb8465`
`(amb8465) exists? /dev/amb8465_0`
`(amb8465) exists? /dev/amb8465_1`
`(amb8465) exists? /dev/amb8465_2`
`(amb8465) exists? /dev/amb8465_3`
`(amb8465) exists? /dev/amb8465_4`
`(amb8465) exists? /dev/amb8465_5`
`(amb8465) exists? /dev/amb8465_6`
`(amb8465) exists? /dev/amb8465_7`
`(amb8465) exists? /dev/amb8465_8`
`(rfmrx2) exists? /dev/rfmrx2`
`(rfmrx2) exists? /dev/rfmrx2_0`
`(rfmrx2) exists? /dev/rfmrx2_1`
`(rfmrx2) exists? /dev/rfmrx2_2`
`(rfmrx2) exists? /dev/rfmrx2_3`
`(rfmrx2) exists? /dev/rfmrx2_4`
`(rfmrx2) exists? /dev/rfmrx2_5`
`(rfmrx2) exists? /dev/rfmrx2_6`
`(rfmrx2) exists? /dev/rfmrx2_7`
`(rfmrx2) exists? /dev/rfmrx2_8`
`(cul) exists? /dev/ttyUSB0`
`(cul) checking /dev/ttyUSB0`
`(rtlsdr) exists? /dev/rtlsdr`
`(rtlsdr) exists? /dev/rtlsdr_0`
`(rtlsdr) exists? /dev/rtlsdr_1`
`(rtlsdr) exists? /dev/rtlsdr_2`
`(rtlsdr) exists? /dev/rtlsdr_3`
`(rtlsdr) exists? /dev/rtlsdr_4`
`(rtlsdr) exists? /dev/rtlsdr_5`
`(rtlsdr) exists? /dev/rtlsdr_6`
`(rtlsdr) exists? /dev/rtlsdr_7`
`(rtlsdr) exists? /dev/rtlsdr_8`
`No wmbus device found! Exiting!`

According to the output it seems that I do not have any meter, however, I had configured a meter in the path `/etc/wmbusmeters.d/`.

weetmuts, 2020-02-12:
Ok, can you checkout the branch BetterDetection build and run the same command again?

diegomjb94, 2020-02-12:
After installing the BetterDetection branch and run the command, the output is the following one:

`(wmbusmeters) version:` 
`(config) using device: auto`
`(config) number of meters: 0`
`(detect) "auto" ""`
`(detect) driver: auto`
`(im871a) exists? /dev/im871a`
`(im871a) exists? /dev/im871a_0`
`(im871a) checking /dev/im871a_0`
`(serialtty) could not open "/dev/im871a_0" with 57600 baud N81`
`(im871a) could not open "-1093681776"`
`(im871a) exists? /dev/im871a_1`
`(im871a) exists? /dev/im871a_2`
`(im871a) exists? /dev/im871a_3`
`(im871a) exists? /dev/im871a_4`
`(im871a) exists? /dev/im871a_5`
`(im871a) exists? /dev/im871a_6`
`(im871a) exists? /dev/im871a_7`
`(im871a) exists? /dev/im871a_8`
`(amb8465) exists? /dev/amb8465`
`(amb8465) exists? /dev/amb8465_0`
`(amb8465) exists? /dev/amb8465_1`
`(amb8465) exists? /dev/amb8465_2`
`(amb8465) exists? /dev/amb8465_3`
`(amb8465) exists? /dev/amb8465_4`
`(amb8465) exists? /dev/amb8465_5`
`(amb8465) exists? /dev/amb8465_6`
`(amb8465) exists? /dev/amb8465_7`
`(amb8465) exists? /dev/amb8465_8`
`(rfmrx2) exists? /dev/rfmrx2`
`(rfmrx2) exists? /dev/rfmrx2_0`
`(rfmrx2) exists? /dev/rfmrx2_1`
`(rfmrx2) exists? /dev/rfmrx2_2`
`(rfmrx2) exists? /dev/rfmrx2_3`
`(rfmrx2) exists? /dev/rfmrx2_4`
`(rfmrx2) exists? /dev/rfmrx2_5`
`(rfmrx2) exists? /dev/rfmrx2_6`
`(rfmrx2) exists? /dev/rfmrx2_7`
`(rfmrx2) exists? /dev/rfmrx2_8`
`(cul) exists? /dev/ttyUSB0`
`(cul) checking /dev/ttyUSB0`
`(serialtty) could not open "/dev/ttyUSB0" with 38400 baud N81`
`(cul) could not open tty /dev/ttyUSB0 for reading.`
`(rtlsdr) exists? /dev/rtlsdr`
`(rtlsdr) exists? /dev/rtlsdr_0`
`(rtlsdr) exists? /dev/rtlsdr_1`
`(rtlsdr) exists? /dev/rtlsdr_2`
`(rtlsdr) exists? /dev/rtlsdr_3`
`(rtlsdr) exists? /dev/rtlsdr_4`
`(rtlsdr) exists? /dev/rtlsdr_5`
`(rtlsdr) exists? /dev/rtlsdr_6`
`(rtlsdr) exists? /dev/rtlsdr_7`
`(rtlsdr) exists? /dev/rtlsdr_8`
`No wmbus device found! Exiting!`

weetmuts, 2020-02-18:
Edit /etc/wmbusmeters.conf and change loglevel to debug.
Edit your meter definition in /etc/wmbusmeters.d/YourMeter
Unplug and insert the imst dongle again.

Post the debug output here!

weetmuts, 2020-02-18:
Oh, and download and build the latest master from github. Not the branch, it has been merged.

weetmuts, 2020-02-23:
Did it work?

diegomjb94, 2020-02-24:
Sorry for the delay. The output is the following one:

`(wmbusmeters) version: `
`(config) using device: auto`
`(config) number of meters: 0`
`(detect) "auto" ""`
`(detect) driver: auto`
`(im871a) exists? /dev/im871a`
`(im871a) exists? /dev/im871a_0`
`(im871a) checking /dev/im871a_0`
`Device /dev/im871a_0 is already in use and locked.`
`(im871a) exists? /dev/im871a_1`
`(im871a) exists? /dev/im871a_2`
`(im871a) exists? /dev/im871a_3`
`(im871a) exists? /dev/im871a_4`
`(im871a) exists? /dev/im871a_5`
`(im871a) exists? /dev/im871a_6`
`(im871a) exists? /dev/im871a_7`
`(im871a) exists? /dev/im871a_8`
`(amb8465) exists? /dev/amb8465`
`(amb8465) exists? /dev/amb8465_0`
`(amb8465) exists? /dev/amb8465_1`
`(amb8465) exists? /dev/amb8465_2`
`(amb8465) exists? /dev/amb8465_3`
`(amb8465) exists? /dev/amb8465_4`
`(amb8465) exists? /dev/amb8465_5`
`(amb8465) exists? /dev/amb8465_6`
`(amb8465) exists? /dev/amb8465_7`
`(amb8465) exists? /dev/amb8465_8`
`(rfmrx2) exists? /dev/rfmrx2`
`(rfmrx2) exists? /dev/rfmrx2_0`
`(rfmrx2) exists? /dev/rfmrx2_1`
`(rfmrx2) exists? /dev/rfmrx2_2`
`(rfmrx2) exists? /dev/rfmrx2_3`
`(rfmrx2) exists? /dev/rfmrx2_4`
`(rfmrx2) exists? /dev/rfmrx2_5`
`(rfmrx2) exists? /dev/rfmrx2_6`
`(rfmrx2) exists? /dev/rfmrx2_7`
`(rfmrx2) exists? /dev/rfmrx2_8`
`(cul) exists? /dev/ttyUSB0`
`(cul) checking /dev/ttyUSB0`
`Device /dev/ttyUSB0 is already in use and locked.`
`(rtlsdr) exists? /dev/rtlsdr`
`(rtlsdr) exists? /dev/rtlsdr_0`
`(rtlsdr) exists? /dev/rtlsdr_1`
`(rtlsdr) exists? /dev/rtlsdr_2`
`(rtlsdr) exists? /dev/rtlsdr_3`
`(rtlsdr) exists? /dev/rtlsdr_4`
`(rtlsdr) exists? /dev/rtlsdr_5`
`(rtlsdr) exists? /dev/rtlsdr_6`
`(rtlsdr) exists? /dev/rtlsdr_7`
`(rtlsdr) exists? /dev/rtlsdr_8`
`No wmbus device found! Exiting!`

weetmuts, 2020-02-25:
It seems like your daemon is running since since it says:
"Device /dev/.... is already in use and locked."

Stop the daemon:
sudo systemctl stop wmbusmeters@-dev-im871a_0.service

Make sure no wmbusmeters or wmbusmetersd are running.
Then try again.

diegomjb94, 2020-02-25:
The output is the same one that I copied in the last message.

weetmuts, 2020-02-26:
Run the command: sudo lsof /dev/im871a_0
to see which process is using the device.

That process (probably another wmbusmeters) is blocking your attempts to start wmbusmeters.

diegomjb94, 2020-02-26:
You were right. Another wmbusmeters was using the device. So, I had run `sudo systemctl stop wmbusmeters@-dev-im871a_0.service` command and after that I ran `wmbusmeters --debug --listento=c1 auto`, the output has changed:

```(wmbusmeters) version: 
(config) using device: auto
(config) number of meters: 0
(detect) "auto" ""
(detect) driver: auto
(im871a) exists? /dev/im871a
(im871a) exists? /dev/im871a_0
(im871a) checking /dev/im871a_0
(serialtty) opened /dev/im871a_0
(serial) received binary 
(im871a) are you there?
(serial /dev/im871a_0) sent "A5010100"
(serial) received binary A58102004CA3
(serialtty) closed /dev/im871a_0
(im871a) checkIM871AFrame "A58102004CA3"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 2
(im871a) got crc16 a34c expected a34c
(im871a) received full frame
(im871a) on /dev/im871a_0
(serialtty) opened /dev/im871a_0
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(im871a) set link mode 06
(serial /dev/im871a_0) sent "A5010306000206300101"
(serial) received binary A58104009CF7
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(im871a) get config
(serial /dev/im871a_0) sent "A5010500"
(serial) received binary A5810617FF000600B32532201000010001FF0702320001010000008CF4
(im871a) checkIM871AFrame "A5810617FF000600B32532201000010001FF0702320001010000008CF4"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 f48c expected f48c
(im871a) received full frame
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 06
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00102032
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) listen to link modes: c1
(serial) waiting for stop
```

weetmuts, 2020-03-08:
Seems like it is working now? Can we close this issue?

diegomjb94, 2020-03-11:
Yes, it seems to be working properly. Thank you! However, it doesn't get any telegrams, what could be the reason?

Thanks in advance.

weetmuts, 2020-03-11:
Have you tried both --listento=c1 and --listento=t1

diegomjb94, 2020-03-11:
Yes. Also, no files listed in `/var/log/wmbusmeters/meter_readings/` path.

weetmuts, 2020-03-16:
The meter_readings will only contain data when you have configured a meter.

You should see (wmbusmeters) waiting for telegrams
in the log.

Do you have that in your wmbusmeters.log?

weetmuts, 2020-04-02:
Closing this.

closed 2020-04-02 by weetmuts