wmbusmeters/wmbusmeters #73: docker wmbusmeters

lsw271, 2020-02-10:
hello @BIBOLV 
i'm trying to use wmbusmeters in docker 
i've got RPI 3b+ /raspian buster > docker > hassio

i'm trying to run docker with one image (wmbusmeters 177mb file) but there are two images, what should i do with mqtt (lightest file) when i've got my own mqtt container from hassio add-ons?

i can't run the container because of error:
**standard_init_linux.go:211: exec user process caused "exec format error"**

maybe it will be helpful:
```
pi@raspberrypi:~ $ docker version
Client: Docker Engine - Community
 Version:           19.03.5
 API version:       1.40
 Go version:        go1.12.12
 Git commit:        633a0ea
 Built:             Wed Nov 13 07:37:22 2019
 OS/Arch:           linux/arm
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.5
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.12.12
  Git commit:       633a0ea
  Built:            Wed Nov 13 07:31:17 2019
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.2.10
  GitCommit:        b34a5c8af56e510852c35414db4c1f4fa6172339
 runc:
  Version:          1.0.0-rc8+dev
  GitCommit:        3e425f80a8c931f88e6d94a8c831b9d5aa481657
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683
```
without the docker i can read water meter but i've got a problem with mqtt because i've got mqtt broker as another container and i could't send any message on host, because i don't have mqtt broker there. I would like to use my mqtt container from hassio add-ons to send messages

BIBOLV, 2020-02-10:
Hi @lsw271 
Please refer to https://github.com/weetmuts/wmbusmeters/issues/63#issuecomment-579091386 for docker docs. 
But in short - you do not need to build docker image locally (if you do not have specific need), but download and run it using either docker cmd or docker-compose.yml.
Please let me know if you have any questions after reviewing documentation.
Regarding MQTT - do you use hassio integrated one or seperate docker image with MQTT server (for instance Mosquitto)? If you are running integrated then you should check if mqtt port is exposed to outside from container and in theory it should be possible to send messages to it (have not tried). If you are using standalone mqtt instance image then everything should work also with wmbusmeters running on host (your raspberry) and moving to docker will not solve your issue.

lsw271, 2020-02-10:
@BIBOLV  i"ve done this with docs, add rules and then trying to run the container. This error `standard_init_linux.go:211: exec user process caused "exec format error"` is from portainer 

> seperate docker image with MQTT server (for instance Mosquitto)?

yes, i've got mosquitto as docker container

i don't have mosquitto_pub in below localization 
`shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/"$METER_ID" -m "$METER_JSON"`
maybe i should use another path? but probably i don't have moquitto service on host - i'll check this tonight

BIBOLV, 2020-02-10:
Could you share more details how you created container in portainer? 
Did you create new stack and added docker-compose file content there or some other way? Did you edited parameters like bind mount and device path for your needs?
If using portainer please try creating new stack and in Build method web editor paste
`version: "2"
services:
  wmbusmeters:
    image: weetmuts/wmbusmeters
    container_name: wmbusmeters
    restart: always
    volumes:
      - /opt/wmbusmeters:/wmbusmeters_data
      - /etc/localtime:/etc/localtime:ro
    devices:
      - /dev/im871a`

Adjust /opt/wmbusmeters if needed and confirm that it exists, also adjust device path if needed. When done press Deploy the Stack

Or other option on host device console (your raspberry) run:
`docker run -d \
    --name=wmbusmeters \
    --restart=always \
    -v /etc/localtime:/etc/localtime:ro \
    -v /opt/wmbusmeters:/wmbusmeters_data \
    --device=/dev/im871a \
    weetmuts/wmbusmeters `

Also adjusting same parameters. 

Path to mosquitto_pub is inside wmbusmeters docker image and it is there so you need to change only localhost to your actual IP of MQTT server so MQTT client would be able to communicate with MQTT server.

BIBOLV, 2020-02-10:
Did a quick googling for your error and most of articles reffer to platform cpu arch issues, I will need to double check that with local raspberry, hopefully I will be able to do it this week. 
Meanwhile we could try to debug/fix issues with your standard wmbusmeters instalation and MQTT conectivity if you are up for it.

lsw271, 2020-02-11:
> Could you share more details how you created container in portainer?

i've created container by command in putty, in portainer i only looks for logs

```
pi@raspberrypi:~ $ docker run -d \
>     --name=wmbusmeters \
>     --restart=always \
>     -v /etc/localtime:/etc/localtime:ro \
>     -v /opt/wmbusmeters:/wmbusmeters_data \
>     --device=/dev/rtlsdr_4 \
>     weetmuts/wmbusmeters
7dc6eabea36d68d66014ae462988ebae2c3e4da6f7fad8fbf2d1a3fb6a9d7012
```
but i've got the same error "standard_init_linux.go:211: exec user process caused "exec format error"

> Path to mosquitto_pub is inside wmbusmeters docker image and it is there so you need to change only localhost to your actual IP of MQTT server so MQTT client would be able to communicate with MQTT server.

my configuration of wmbusmeters.conf:
```
loglevel=debug
device=rtlwmbus:868.9M
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
meterfilesnaming=name
meterfilestimestamp=day
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=/usr/bin/mosquitto_pub -h 192.168.xxx.xxx -p 1883 -u "mqtt" -P "xxxxxx" -t wmbusmeters/"$METER_ID" -m "$METER_JSON"
```

configuration of Mosquitto broker 5.1 (hassio add-on)
```
logins:
  - username: mqtt
    password: xxxxxx
anonymous: true
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false
log_level: trace
```
when i'm trying by hand, the command is not correct

```
pi@raspberrypi:/usr/bin $ mosquitto_pub -h 192.168.xxx.xxx -p 1883 -u "mqtt" -P "xxxxxx" -t wmbusmeters
-bash: mosquitto_pub: command not found
```
so i think its a problem with mqtt on host

BIBOLV, 2020-02-11:
Existing docker image is built for x86 cpu arch and will not run on arm. While I`m working to solve it I would sugest to connect you local running wmbusmeters instance to mqtt.
Looking at your hassio config looks like mqtt server is in same container with hassio, so question - is mqtt available externaly. Try to run `telnet 192.168.xxx.xxx 1883` from raspberry and see if port is open, if it is then install mqtt client by running `sudo apt-get install mosquitto mosquitto-clients` on raspberry and see if wmbusmeters can send info to mqtt.

lsw271, 2020-02-11:
Of course! :) i didn't have a mosquitto-client on host. After `sudo apt-get install mosquitto-clients` everything it's ok with mqtt. Thanks @BIBOLV !
i think we can close this issue

closed 2020-02-11 by lsw271

BIBOLV, 2020-03-08:
@lsw271 you can now try to run docker on raspberry if your`e up for it.
Please check https://hub.docker.com/r/weetmuts/wmbusmeters