wmbusmeters/wmbusmeters #74: Obviously Hydrus t1 telegram misinterpretation

superfino, 2020-02-11:
Hi again, just after receiving my first hydrus watermeter telegrams i suffered the next problem. It seems that there is a problem with the offset in dismantle the telegram. Wmbusmeters rejects all my frames because it doesn't identify my ID. my ID is 00260003. I think the meter ID starts 2 bytes later as you can see from my logfile attached. I have opened a new issue since the problem is different from the last one.  So long... Jo

[hydrus_t1.txt](https://github.com/weetmuts/wmbusmeters/files/4187056/hydrus_t1.txt)

weetmuts, 2020-02-12:
The number printed on the meter is not always the number in the wmbus telegram.

The Hydrus meter is mostly wmbus standards compliant, so use the id that you see in the wmbus telegram, which in your case is: 00030724

The manufacturer can do anything with the 6 bytes that make up the address-field.
In your case: mediatype=0 (non-compliant, this usually tells what kind of meter it is)
version=26 (any meter version from this manufacturer) id=00030724

If they want to print all six bytes on the meter, fine. If the want to print just the id, very fine, this
is normal. If they want to print type,version,id (without the two last id bytes because 0724 is static for all their meters) errr, not fine, but who cares....

superfino, 2020-02-12:
Thanks for your clarification. I've tried your recommendation but wmbusmeters still has problems to handle the Telegram.
'Unknown tpl-ci-field 00' and 'expected telegram with version 0x70, but got version 0x26 !' are the messages. Have attached a logfile for more details.

[mylog2.txt](https://github.com/weetmuts/wmbusmeters/files/4190863/mylog2.txt)

weetmuts, 2020-02-12:
Interesting, these telegrams look like telegrams sent to the meter.
If you check your log you will see: (hydrus) 01: 24 dll-c (to meter SND_NR)
Normally all telegrams that wmbusmeters listen for are 44 (not 24)

Do you know what is transmitting these telegrams? 

If you listen for awhile, do you see any 
01: 44 dll-c (from meter SND_NR)
in the log?

superfino, 2020-02-12:
Indeed, i have not recognised that there are two kind of frames. But the ones send by the meter show the same behavior.

[log2.txt](https://github.com/weetmuts/wmbusmeters/files/4194212/log2.txt)

weetmuts, 2020-02-12:
log2.txt only contains a telegram to meter dll-c = 24
do you see any from_meter dll-c=44 ? I.e. the second byte in the telegram must be 44,
otherwise wmbusmeters wont work.

Do you know what is sending to the meter?

superfino, 2020-02-12:
Have checked now hundreds of frames but all with dll-c=24, nothing else. I won't belive that there are frames send to the meter. I've to stay nearby the meter (approx. 8m) to receive the frames. When there is another RF source outside my sinkhole, i should receive the frames all over. So i think this is not quiet likely.

superfino, 2020-02-14:
I've logged now my Diehl Hydrus 171A for some hours beside the meter. There are also some other meters in this uncut log for comparison. None of them show dll-c = 44 so i think dll-c = 24 is normal behaviour.

[mylog3.txt](https://github.com/weetmuts/wmbusmeters/files/4204329/mylog3.txt)

superfino, 2020-02-15:
Some interrresting news here. Have today catured some t1 telegrams with a plain serial terminal. The result was surprisingly. All frames have dll-c=44. I changed back to wmbusmeters and all frames come with dll-c=24. Now what? How can this happen?

[serial.txt](https://github.com/weetmuts/wmbusmeters/files/4208595/serial.txt)

weetmuts, 2020-02-15:
Yes, there is a bug in wmbus_cum.cc i am fixing it!

weetmuts, 2020-02-15:
It is a problem with the CRCs.

@chester444 received C1 telegrams with frame B, 
where the CRCs are easy to remove if the telegram is shorter than 128 bytes.
Its just the two last bytes....

However @superfino wants to recive T1 telegrams using frame A
which is more complicated.

I have rewritten wmbus_cul.cc quite a lot and added
trimCRCsFrameFormatA
and trimCRCsFrameFormatB.

Trim format B seems to works. But please test @chester4444.
However format A, I have got problems with the last block.

I also see that CUL seems to send extra bytes at the end of the telegram.

Do you have documentation for this @superfino or @chester4444?

Also this Hydurs meter does not talk the same telegram protocol
as the existing Hydrus meter driver. Quite different actually.
It might not even be wmbus compatible...

@superfino do you have an decryption key for this meter?
If so, can you mail it to me?

chester4444, 2020-02-16:
which branch to test?

weetmuts, 2020-02-16:
In this case, it is master.

superfino, 2020-02-16:
Hello, just tested your fix right now and it looks much better, whenever i think it's not finished. Regarding your questions, a link to all me known CUL docu = http://culfw.de/culfw.html ,
attached the WMBus docu for the Hydrus meter and at least a new log.

[mylog5.txt](https://github.com/weetmuts/wmbusmeters/files/4209812/mylog5.txt)
[Hydrus 1.2 WMBus.pdf](https://github.com/weetmuts/wmbusmeters/files/4209813/Hydrus.1.2.WMBus.pdf)

chester4444, 2020-02-16:
> But please test @chester4444.

everything still ok here

weetmuts, 2020-02-16:
Thank you @chester4444 for the confirmation and thank you @superfino. 
With this information I should be able to get it working.

weetmuts, 2020-02-16:
Ok, @superfino I managed to fix the CUL driver. It seems to receive and trim the crc checksum bytes properly for T1 telegrams. (The CUL is one weird dongle, for C1 telegrams the len byte includes the crc bytes, and there are 2 extra crc bytes at the end. But for T1 telegrams the len bytes does not include the crc bytes and there are no 2 extra crc bytes at the end.....)

Your meter seems to be an older version of the Diehl Hydrus meter...

Your telegrams does not say that it measures water, it says (00) which means other....
The manufacturer is HYD which is Hydrometer which is the old name for Diehl.
But more problematic the tpl-config field that explains the encryption is just
weird and changes.  Here it is 3ABF, which is not a known encryption mode.
484424232407030026(00) >>7A 7800 >>3ABF?B32704147C...

Other Hydrus meter that I have in simulations/simulation_t1.txt
the manufacturer is DME which is the new name Diehl Metering.
Here the tpl-config is a nice 4005, which means security mode 5.
4E44A5116464646470(07) >>7A ED00 >>4005|2F2F01FD08...

The documentation you referred to clearly gives an example
of a telegram using 4005.

Could it be that you have a very old meter which is not OMS compliant?
Or should we listen for some other telegram?

Could you listen more to see if you see any telegram with 4005 in them?

superfino, 2020-02-17:
Hmm, it seems you're right, weird and not complient are the right words. My meter is 6 years old, the moment as Hydrometer merged to Diehl. The interiority is Hydrometer and the outside is Diehl. Since my meter's ID or S/Nr. is 00260003 and my neighbors is 00260007 the ID must be moved two bytes left, the 07 before represents the media and the 24 the Generation ID. Maybe, my theory. I don't know. I logged today for some hours with 2 abortions. 
Unsupported dif format for extraction to double! dif=3d
(dvparser) warning: unexpected end of data
(dvparser) warning: unexpected end of data (vif expected)
But unfortunately i can't serve a 4005 encrytion mode. The actual results with your new version are very different. My current meter reading is around 580+ m3 but everything in the log is far away. I admire your constant efforts but unfortunately can't be of little help. If you have the time and the passion, i've attached a log and my meter! For nerds only !!!

[mylog6.txt](https://github.com/weetmuts/wmbusmeters/files/4214430/mylog6.txt)
[image]

weetmuts, 2020-02-18:
Yes, unfortunately, its going to be tricky to decode this meter. My personal focus is on wmbus-compliant meters. 

But, we got T1 mode added  to the CUL driver and proper CRC hash removing! This is very valuable!

chester4444, 2020-02-18:
currently i am testing  0.9.26-18-gf1da5b4:

now i get an crc error:

`(serial) received binary 3941324644363232323030384439353938464644363032343836463535383434423437383630363930344246433831353739370D0A
(cul) checkCULFrame "bY25442D2C250635741B168D20C9A2FD6222008D9598FFD602486F55844B478606904BFC815797<0D><0A>"
(wmbus) trimming frame B "25442D2C250635741B168D20C9A2FD6222008D9598FFD602486F55844B478606904BFC81"
(wmbus) ff b dll crc (calculated 8c26) did not match (expected fc81) for bytes 0-34!
(cul) dll C1 (frame b) crcs failed check! Ignoring telegram!
(cul) error in received message.
`

weetmuts, 2020-02-18:
Try this change i wmbus_cul.cc line 307:
-        hex.insert(hex.end(), data.begin()+2, data.begin()+eolp-eof_len-4); // Remove CRLF
+        hex.insert(hex.end(), data.begin()+2, data.begin()+eolp-eof_len); // Remove CRLF

See if it works for you. I hope it does, because then the CUL driver is at least consistent
between C1 and T1 mode. The -4 are there because previous examples had two extra bytes
(4 hex digits) at the end.... but maybe the examples were bad.

Anyway please test this change. If it works I will push it to master.

chester4444, 2020-02-18:
yes, I it works -  👍 (i had found the same solution in this second)

The CUL driver can append RSSI info when starting the receiver with X21
but we do it with X01 - so i think you have some inconsistent examples.

weetmuts, 2020-02-18:
Ah, makes sense, thanks. I have pushed the fix to master.

weetmuts, 2020-02-23:
Assuming this meter sends totally wmbus-noncompliant data, it could
be that it lies about being encrypted.

If you collect a lot of telegrams when your meter consumes water, 
and if we are lucky, the data is not encrypted, then we can see
which bytes continously increment. 

We did this for the apator162 meter. :-)

superfino, 2020-02-23:
If this is a chance, i'll arrange it and deliver the data soon. Thanks for your suggestions.

superfino, 2020-02-24:
Hello, i started the log this morning with a meter count of 590.660 m3 and stopped it with 590.727 m3.
Some other maybe interesting readings are the software version 'F4-005' the batterie lifetime 02.09.26 and another date (31.12.19) which changes with the value (571.392) . Thats what the meter shows me. Hope this helps for further analysis.  

[mylog8.txt](https://github.com/weetmuts/wmbusmeters/files/4244030/mylog8.txt)

weetmuts, 2020-02-26:
Thanks, I have now looked at the log and unfortunately it is definitely encrypted 
and has probably some sort of dependency on a iv/salt and/or the acc (counter).

The length is not a multiple of the AES blocks so it is unclear how to even attempt
a decryption. It is probably a aes-cbc encrypted, but how is the IV generated?

The wmbus standard nicely specifies this, but alas the meter does not comply with the standard.
So, I cannot make any more progress on this meter without some hints on how they use AES. :-/

superfino, 2020-02-26:
Too bad, i feared that it is not possible under this circumstances. Nevertheless, many thanks again for
your strenuous efforts in this case, which i think can be closed now.

closed 2020-02-26 by superfino