wmbusmeters/wmbusmeters #79: Suggestion : Check if rtl_sdr / rtl_wmbus exist

anonymous-one, 2020-02-18:
I know there are probably many things that need to be done before something of this nature, but a small suggestion:

For someone playing with sdr for the first time, check to see if the rtl_sdr / rtl_wmbus binaries exist, if not, show something in the error logs. Currently silent

Once again, fairly elementary... But besides this, the daemon spawned up without ANY issues on my RPI3B+ running Stretch.

weetmuts, 2020-02-18:
Actually, I just pushed a fix for this yesterday. :-)

When starting wmbusmeters as a daemon it will check
that /usr/bin/rtl_sdr and /usr/bin/rtl_wmbus exists.
If not found, an error is printed in the wmbusmters.log file
before exiting.

When starting from the command line it uses the PATH
to find the binaries, so no such check is done.

anonymous-one, 2020-02-18:
:)

closed 2020-02-18 by anonymous-one