wmbusmeters/wmbusmeters #80: Packet is not fully decoded neither json output generated

dwrobel, 2020-02-18:
I'm trying to figure out how to convert input data from `stdin` in a form of _(T1 packet)_ from the `rtlwmbus` into the json format sent to `stdout`. Based on documentation I'm using the following command line:

```
$ echo "T1;1;1;2020-02-18 17:43:08.000;64;128;00111979;0x1844ae4c791911007c077a2900000004139d190100023b0000" |./build/wmbusmeters  --debug --verbose --format=json stdin:rtlwmbus
(wmbusmeters) version: 0.9.26-18-gf1da5b4
(config) using device: stdin
(config) with: rtlwmbus
(config) number of meters: 0
(detect) "stdin" "rtlwmbus"
(detect) is_tty=0 is_stdin=1 is_file=0
(serial) override with devicefile: stdin
(serialfile) reading from stdin
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop
(serial) no more data on fd=0
(serial) received binary 54313B313B313B323032302D30322D31382031373A34333A30382E3030303B36343B3132383B30303131313937393B307831383434616534633739313931313030376330373761323930303030303030343133396431393031303030323362303030300A
(serial) no devices working
(serial) stopping manager
(serialtty) closed stdin -1
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-02-18 17:43:08.000;64;128;00111979;0x1844ae4c791911007c077a2900000004139d190100023b0000<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 25
(wmbus) parseELL @10 15
(wmbus) parseAFL @10 15
(wmbus) parseTPL @10 15
(dvparser) found new format "0413023B" with hash c944, remembering!
(telegram) DLL L=18 C=44 (from meter SND_NR) M=4cae (SEN) A=00111979 VER=7c TYPE=07 (Water meter)
(telegram) TPL CI=7a ACC=29 STS=00 CFG=0000 ()
Received telegram from: 00111979
          manufacturer: (SEN) Sensus Metering Systems, Germany
           device type: Water meter
(wmbus) 00: 18 length (24 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: ae4c dll-mfct (SEN)
(wmbus) 04: 79191100 dll-id (00111979)
(wmbus) 08: 7c dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: 29 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 0000 tpl-cfg 0000 ( )
(wmbus) 0f: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 10: 13 vif (Volume l)
(wmbus) 11: 9D190100
(wmbus) 15: 02 dif (16 Bit Integer/Binary Instantaneous value)
(wmbus) 16: 3B vif (Volume flow l/h)
(wmbus) 17: 0000
(wmbus) "telegram=||1844AE4C791911007C077A2900000004139D190100023B0000|+0"
(serial) event loop stopped!
(serial) closing 0 devices
```
However I don't see the packet to be fully decoded _(despite using --debug and --verbose)_ nor converted into json format. Am I doing something wrong or I simple misunderstood the documentation or purpose of this program?

FYI, I verified that the input packet seems to be correct as when I'm passing the same data into [jmbus](https://github.com/kaikreuzer/jmbus) based java program the list of measurements are as follows:

```
DIB:04, VIB:13 -> descr:VOLUME, function:INST_VAL, scaled value:72.093, unit:CUBIC_METRE, m³
DIB:02, VIB:3B -> descr:VOLUME_FLOW, function:INST_VAL, scaled value:0.0, unit:CUBIC_METRE_PER_HOUR, m³/h
```
So the question is. Is it possible to get from the `wmbusmeters` the json output containing the following measurements: 72.093m³ and 0.0m³/h?

weetmuts, 2020-02-18:
You only need to add: MyWater iperl 00111979 NOKEY
to the end of your commandline.
Then it will print:
(iperl) expected telegram with version 0x68, but got version 0x7c !
{"media":"water","meter":"iperl","name":"MyWater","id":"00111979","total_m3":72.093,"max_flow_m3h":0,"timestamp":"2020-02-18T20:26:46Z"}

What is the exact version of your sensus iperl meter? I will add 0x7c to the iperl driver.

dwrobel, 2020-02-18:
> You only need to add: MyWater iperl 00111979 NOKEY
> to the end of your commandline.
> Then it will print:

Indeed it works now. Thank you very much for the support.

> What is the exact version of your sensus iperl meter? I will add 0x7c to the iperl driver.

It is: [Sensus 640](https://sensus.com/products/640-water-meter/).

dwrobel, 2020-02-18:
There is one minor issue. It seems to be not possible to combine `wmbusmeters` into more compound shell commands e.g.:
```
$ echo "T1;1;1;2020-02-18 17:43:08.000;64;128;00111979;0x1844ae4c791911007c077a2900000004139d190100023b0000" |./build/wmbusmeters --debug --verbose --format=json stdin:rtlwmbus MyWater iperl 00111979 NOKEY  | jq
parse error: Invalid numeric literal at line 1, column 14
$
```

Mainly because the debug and verbose messages are combined with the real output and all are send to the `stdout`. It would be more usable if the debug and verbose messages will be send to e.g. `stderr` and the real output e.g. json to the `stdout`.

weetmuts, 2020-02-22:
I have added 0x7c version to the iperl meter and I have added --usestderr as a command line option.
This option will print all debug/verbose and log messages on stderr.

Please test!

dwrobel, 2020-02-25:
> Please test!

I tested it and it works well.
Thank you!

weetmuts, 2020-02-26:
Great!

closed 2020-02-26 by weetmuts