wmbusmeters/wmbusmeters #81: Apator  AT-WMBUS-08

henrykmal, 2020-02-22:
Hello

Any chance for working At-Wmbus-08?

weetmuts, 2020-02-22:
I do not know. If you can collect some telegrams while you at the same time consume water.
Then we might have a chance to find the value in the telegram. :-)
Assuming it can be decrypted etc etc. 

Apator is unfortunately not wmbus compliant, so there are no guarantees that they will work.

henrykmal, 2020-02-23:
In Attachment are a few logs. My apator is 000404d9 othera are neighbors.
My water consumption is now 871,850 m3

T1;1;1;2020-02-23 09:14:01.000;71;76;000404D9;0x73441486d90404000303a0a8e527004c4034b31ced0106ff01d093270065f022009661230054d02300ec49240018b424005f012500936d2500ffd525000e3d26001eac26000b2027000300000000371d0b2000000000000024000000000000280000000000002c0033150c010d2f000000000000
T1;1;1;2020-02-23 09:17:19.000;76;72;000404D9;0x73441486d90404000303a0b9e527004c4034b31ced0106ff01d093270065f022009661230054d02300ec49240018b424005f012500936d2500ffd525000e3d26001eac26000b2027000300000000371d0b2000000000000024000000000000280000000000002c0033150c010d2f000000000000
T1;1;1;2020-02-23 09:18:38.000;71;71;000404D9;0x73441486d90404000303a0b9e527004c4034b31ced0106ff01d093270065f022009661230054d02300ec49240018b424005f012500936d2500ffd525000e3d26001eac26000b2027000300000000371d0b2000000000000024000000000000280000000000002c0033150c010d2f000000000000
T1;1;1;2020-02-23 09:20:31.000;68;79;000404D9;0x73441486d90404000303a0b9e527004c4034b31ced0106ff01d093270065f022009661230054d02300ec49240018b424005f012500936d2500ffd525000e3d26001eac26000b2027000300000000371d0b2000000000000024000000000000280000000000002c0033150c010d2f000000000000
T1;1;1;2020-02-23 09:22:09.000;69;77;000404D9;0x73441486d90404000303a0b9e527004c4034b31ced0106ff01d093270065f022009661230054d02300ec49240018b424005f012500936d2500ffd525000e3d26001eac26000b2027000300000000371d0b2000000000000024000000000000280000000000002c0033150c010d2f000000000000
T1;1;1;2020-02-23 09:40:21.000;68;74;000404D9;0x73441486d90404000303a0d2e527004c4034b31ced0106ff01d093270065f022009661230054d02300ec49240018b424005f012500936d2500ffd525000e3d26001eac26000b2027000300000000371d0b2000000000000024000000000000280000000000002c0033150c010d2f000000000000

T1;1;1;2020-02-23 09:18:18.000;27;55;00042E36;0x73441486362e04000303a038e20700404034e1ffed0106ff01c2d207004c380700ba420700d84f0700c8580700d5660700c2760700af87070085950700e3a6070039b1070053bf0700030000000037090b2000000000000024000000000000280000000000002c0000000000002f000000000000
T1;1;1;2020-02-23 09:18:37.000;76;131;0004021E;0x734414861e0204000303a00fa319004c40340a1bed0106ff0126681900bd70160040b91600ec0017001c481700dc8b170013c61700bb081800064e1800be911800a8d41800e71d19000300000000371e0b2000000000000024000000000000280000000000002c00281003070c2f000000000000
T1;1;1;2020-02-23 09:18:38.000;31;37;0004054D;0x734414864d0504000303a03e8b0b004c6034ad1ced0106ff013a760b004a570a0081760a0062960a0037b50a008ad50a006ee40a0048f60a001f160b0002340b00384d0b0027630b000300000000371d0b2000000000000024000000000000280000000000002c002e051706112f000000000000
T1;1;1;2020-02-23 09:18:38.000;71;71;000404D9;0x73441486d90404000303a0b9e527004c4034b31ced0106ff01d093270065f022009661230054d02300ec49240018b424005f012500936d2500ffd525000e3d26001eac26000b2027000300000000371d0b2000000000000024000000000000280000000000002c0033150c010d2f000000000000
T1;1;1;2020-02-23 09:18:44.000;93;128;000406CE;0x73441486ce0604000303a0e0610c00484034b41ced0106ff0111500c0003c10a0057d90a0084f50a0050120b0060350b002e680b00f8a30b006ddc0b005e050c007e1d0c0006370c000300000000371c0b2000000000000024000000000000280000000000002c0000000000002f000000000000
T1;0;0;2020-02-23 09:18:46.000;17;24;0004053C;0x73ff14863c0504000303a0edbc14004c4034c91ced0106ff01ff901400a60912ff3d431200a576120057aa120015e91200d626130061681300a7a21300a8dd130024151400245514000300000000ff1d0b20000000000000240000000000ff280000000000002c0029010201102f000000000000
T1;1;1;2020-02-23 09:18:49.000;49;79;0004363E;0x734414863e3604000303a07ac70100404034f9ffed0106ff010dc6010013b50100b0b50100b1b6010077b7010047b9010039bb010040bd01006cbf010062c1010019c2010087c40100030000000037060b2000000000000024000000000000280000000000002c0000000000002f000000000000
T1;1;1;2020-02-23 09:18:55.000;36;66;000404EA;0x73441486ea0404000303a0727008004c4034c01ced0106ff01727008007e920700c3a40700c7ba0700e9d10700ddea070049fb0700a7080800e3220800623e0800f7570800796b08000300000000371d0b2000000000000024000000000000280000000000002c0002131d05102f000000000000
T1;1;1;2020-02-23 09:18:56.000;32;57;0004050B;0x734414860b0504000303a067160200484034c51ced0106ff01171102002ccc0100e8d10100b2d70100b9dd0100ffe6010031ee01003ef4010074fb0100b100020050050200680b02000300000000371d0b2000000000000024000000000000280000000000002c0000000000002f000000000000
T1;1;1;2020-02-23 09:19:04.000;72;84;000400E3;0x73441486e30004000303a02daf12004c40342a1bed0106ff015b8212005dcc0f00440c1000b94d1000a08c100063c91000500a1100dd4d11006f871100f0c91100a30312004e4712000300000000371e0b2000000000000024000000000000280000000000002c0024040102112f000000000000
T1;1;1;2020-02-23 09:19:05.000;40;75;0004312F;0x734414862f3104000303a0e2582600404034e8ffed0106ff01582c26004d922300c8c72300d80a2400c14224003f792400f0bd240009f4240055332500386e25007ea72500efec2500030000000037080b2000000000000024000000000000280000000000002c0000000000002f000000000000
T1;1;1;2020-02-23 09:19:21.000;76;144;00040082;0x73441486820004000303a03c5c19004840340cceec0106ff016b2c190035d315004d1b16005c6416009daf16000a05170081591700acaa170031fa17004b4a1800d8951800e1e118000300000000371e0b2000000000000024000000000000280000000000002c0000000000002f000000000000
T1;0;0;2020-02-23 09:19:25.000;16;44;00043199;0x73441486993104000303a092391f004040ffe3ffed01ffff01ffee1efffed41a0018271b00ee721b0096c61b0064181c009b601c0059d31c00aa451d00e7b11d006b151e0061871e0003000000ff37080b20000000000000240000ff000000280000000000002c0000000000002f0000ff000000
[... cut ...]
T1;1;1;2020-02-23 09:41:36.000;73;140;0004021E;0x734414861e0204000303a01ca319004c40340a1bed0106ff0126681900bd70160040b91600ec0017001c481700dc8b170013c61700bb081800064e1800be911800a8d41800e71d19000300000000371e0b2000000000000024000000000000280000000000002c00281003070c2f000000000000
T1;1;1;2020-02-23 09:41:38.000;89;146;00040082;0x73441486820004000303a0435c19004840340cceec0106ff016b2c190035d315004d1b16005c6416009daf16000a05170081591700acaa170031fa17004b4a1800d8951800e1e118000300000000371e0b2000000000000024000000000000280000000000002c0000000000002f000000000000

weetmuts, 2020-02-23:
Ok, I pushed experimental support for apator08 to master. I am guessing that the water total data are the first 4 bytes after ...0303a0 xx xx xx xx 

Please test by monitoring your consumption and collect logfiles while your own meter
consumes water. The absolute value for your meter seems to be 2614.713 m3.
Which might or might not be correct.... (despite you wrote 871,850 m3) 
since the wmbus radio value can be different from the mechanical meter value.
However another meter has the total 811.488.

henrykmal, 2020-02-23:
Thank you so much for your time and work :-)

i compare logs from my mechanic meter and from wmbusmeter
and i came to the conclusion i must divide the result by 3
Now in wmbusmeter logs 2615,066
in mechanic meter 871,688
2615,066/3=871,666
Maybe for apator 3 impulses = 1 litr water

henrykmal, 2020-02-23:
871,680 - 2615,041
871,682 - 2615,047
871,683 - 2615,05

niedz., 23 lut 2020 o 17:03 Henryk Malysz <henrykmalysz@gmail.com>
napisał(a):

weetmuts, 2020-02-23:
Nice! I just pushed a divide by 3 to wmbusmeters. Odd thing to have it report the ticks, 
since even the simplest meter-cpu ought to be able to divide by 3. :-)

We will see if the /3 relation holds for other apator08 users.

weetmuts, 2020-02-26:
Can we close this now? :-) Does the divide by 3 work as expected?

henrykmal, 2020-02-26:
For me all works great bat sometimes  i was a result with six decimal
places. Like this : 872.755667

weetmuts, 2020-02-26:
That is an artifact of using double and divide by 3.

I have decided to not explicitly round the values and use
the c++ function to_string(v) but drop trailing zeroes.
to_string is the equivalent to the printf formatter %f
which does handle doubles and not just floats.

See the function "string valueToString(double v, Unit u)" in units.cc

So the number of decimals depend on how many significant
digits the double has and how the double is represented in the bits.... 
:-) Complicated stuff.

weetmuts, 2020-02-26:
Closing this for now. Open a new issue if you find problems.

closed 2020-02-26 by weetmuts