wmbusmeters/wmbusmeters #82: Mode C1 format A (OMS)?

GitTibbe, 2020-02-23:
Is there support for mode C1 format A? OMS does not support format B so I cannot test OMS meters correctly.

weetmuts, 2020-02-23:
Hmmm, I am not sure I follow. Do you mean frame format A and B?
(A having much more CRCs in the data link layer. B having much fewer.)

I think it depends a lot on the radio dongle you use. The hardware ones like
imst and amber strip the frame format automatically. You will not see any
difference when receiving the telegram.

The CUL dongle does not strip the crcs, so I need to use the methods 
trimCRCsFrameFormatA and trimCRCsFrameFormatB to remove the 
frame formats. Rtlwmbus does strip the crcs for both frame format A and B.

So, what is your use case? Receiving data for measurements or for testing?
Which meters are you interested in?  
Which radio dongle are you using?

weetmuts, 2020-02-23:
It seems like frame format B is recommended for C1 telegrams. 

If you really want to receive C1 telegrams with frame format A and the hardware dongles
do not support this. I have no idea if they do. :-)
Then try rtlwmbus (perhaps modify it) or the cul dongle.

GitTibbe, 2020-02-23:
Okay I see. I'm using this for testing (simulate) my meter device to be compatible with the open metering standard (OMS) which requires meters to send in format A for all modes... If its only the CRC fields that differs I could trim them away to test. But it's a shame that OMS doesn't support format B.  

I'm not using any dongle as I'm creating the frames myself.

weetmuts, 2020-02-23:
Hmm, I am not an expert, but isnt OMS is a subset of EN-13757 ?

Multical21 which is a very common meter i Sweden offers these configurations:
Wireless M-Bus, 868 MHz, mode C1
Wireless M-Bus, 868 MHz, mode T1 - OMS

So I assume, if you want to use C1 frame b, then you cannot call it OMS,
but that is about it. It is still EN13757 wmbus compatible.
So just dont call it OMS-compliant then... :-) Or do you have to?

Eventually the frame B stuff will get into the OMS I suppose?

Will you build the hardware for your meter device that you currently simulate?

GitTibbe, 2020-02-23:
Yes but most meter manufacturer brag so much about OMS that it felt as mandatory. However I also saw multical 21 to have either A or B versions.  I just sent an email to OMS asking them about format B. 

Yes I'm building it!

weetmuts, 2020-02-23:
Ok, you can use wmbusmeters to test your simulated frame A or B.

Put your own generated frame B C1 telegram in the file myc1telegram.txt
with this ASCII text: (This is the serial ascii protocol that the cul dongles sends over the serial tty.)

bY2544MMMMAAAAAAAA.....XXXX<CR><LF>

Where XXXX is the CRC. Then do 
wmbusmeters --debug myc1telegram.txt:cul
You will see things like trimming frame B.

Now put a T1 telegram in myt1telegram.txt
b4844MMMMAAAAAAAA....<0D><0A>
with all the CRCs in place. Then run
wmbusmeters --debug myt1telegram.txt:cul

You will see things like:
(wmbus) trimming frame A "4844......."
(wmbus) ff a dll crc 0-9 1e75 ok
(wmbus) ff a dll crc mid 12-27 442a ok
(wmbus) ff a dll crc mid 30-45 f294 ok
(wmbus) ff a dll crc mid 48-63 bbb6 ok
(wmbus) ff a dll crc final 66-80 e08d ok
(wmbus) trimmed 10 crc bytes from frame a and ignored 0 suffix bytes.
(wmbus) trimmed  frame A "4844....."

If any crc errors are found, it will print those.
If you find any errors in my code, let me know. :-)

weetmuts, 2020-02-26:
Closing this now. Please re-open if you need more help.

closed 2020-02-26 by weetmuts