wmbusmeters/wmbusmeters #83: Docker and RTLSDR

MTrab, 2020-02-24:
I'm trying to run wmbusmeters in Docker with a RTLSDR dongle.

Started the Docker container with this command, but it just stops instantly (if I append --restart=always it just goes into a restart loop):
`docker run --name=wmbusmeters -v /etc/localtime:/etc/localtime:ro -v /volume2/docker/wmbusmeters/config:/wmbusmeters_data --device=/dev/rtlsdr weetmuts/wmbusmeters`

wmbusmeters.conf:
```
loglevel=debug
device=auto
listento=c1,t1
logtelegrams=false
format=json
meterfiles=/wmbusmeters_data/logs/meter_readings
meterfilesaction=overwrite
logfile=/wmbusmeters_data/logs/wmbusmeters.log
```

Log:
```
(wmbusmeters) version: 0.9.21-1-geed4b17
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) using device: auto
(config) number of meters: 0
(detect) "auto" ""
(detect) driver: auto
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
(config) explicitly listening to: c1,t1
(config) listen link modes calculated to be: c1,t1
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii /bin/sh: rtl_sdr: not found<0A>
(serial) received ascii /bin/sh: rtl_wmbus: not found<0A>
(bgshell) 9 exited with return code 127
(serial) closing non working fd=3
(serial) no devices working
(serial) stopping manager
(serialcmd) closed /bin/sh
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working
(serialcmd) closed /bin/sh
```

Have I missed something with the RTLSDR setup?

And yes, the rtl_sdr dongle is detected by the host system.
```
root@Synology:/volume2/docker/wmbusmeters# ls -l /dev/rtl*
lrwxrwxrwx 1 root root 15 Feb 24 17:54 /dev/rtlsdr -> bus/usb/002/117
root@Synology:/volume2/docker/wmbusmeters#
```

weetmuts, 2020-02-24:
Wmbusmeters starts a background commandline with rtl_sdr and rtl_wmbus.
These programs are not found in the docker image and/or the default path.

MTrab, 2020-02-24:
Oh - bummer. So I need to see if I can get rtlsdr installed on my Synology box?!

weetmuts, 2020-02-24:
I am not an expert on docker, but surely it should be possible to build a docker image
with rtl_sdr and rtl_wmbus in it? Right @BIBOLV ?

BIBOLV, 2020-02-24:
@MTrab I created test docker image - please try and let us know does it work
'docker run --name=wmbusmeters -v /etc/localtime:/etc/localtime:ro -v /volume2/docker /wmbusmeters/config:/wmbusmeters_data --device=/dev/rtlsdr bibolv/wmbusmeters_test:latest'

MTrab, 2020-02-24:
Got more data in the logfile, but still no-go.

```
(wmbusmeters) version: 0.9.26-23-g33952d8
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) using device: auto
(config) number of meters: 0
(detect) "auto" ""
(detect) driver: auto
(im871a) exists? /dev/im871a
(im871a) exists? /dev/im871a_0
(im871a) exists? /dev/im871a_1
(im871a) exists? /dev/im871a_2
(im871a) exists? /dev/im871a_3
(im871a) exists? /dev/im871a_4
(im871a) exists? /dev/im871a_5
(im871a) exists? /dev/im871a_6
(im871a) exists? /dev/im871a_7
(im871a) exists? /dev/im871a_8
(amb8465) exists? /dev/amb8465
(amb8465) exists? /dev/amb8465_0
(amb8465) exists? /dev/amb8465_1
(amb8465) exists? /dev/amb8465_2
(amb8465) exists? /dev/amb8465_3
(amb8465) exists? /dev/amb8465_4
(amb8465) exists? /dev/amb8465_5
(amb8465) exists? /dev/amb8465_6
(amb8465) exists? /dev/amb8465_7
(amb8465) exists? /dev/amb8465_8
(rfmrx2) exists? /dev/rfmrx2
(rfmrx2) exists? /dev/rfmrx2_0
(rfmrx2) exists? /dev/rfmrx2_1
(rfmrx2) exists? /dev/rfmrx2_2
(rfmrx2) exists? /dev/rfmrx2_3
(rfmrx2) exists? /dev/rfmrx2_4
(rfmrx2) exists? /dev/rfmrx2_5
(rfmrx2) exists? /dev/rfmrx2_6
(rfmrx2) exists? /dev/rfmrx2_7
(rfmrx2) exists? /dev/rfmrx2_8
(cul) exists? /dev/ttyUSB0
(cul) exists? /dev/ttyUSB0_0
(cul) exists? /dev/ttyUSB0_1
(cul) exists? /dev/ttyUSB0_2
(cul) exists? /dev/ttyUSB0_3
(cul) exists? /dev/ttyUSB0_4
(cul) exists? /dev/ttyUSB0_5
(cul) exists? /dev/ttyUSB0_6
(cul) exists? /dev/ttyUSB0_7
(cul) exists? /dev/ttyUSB0_8
(rtlsdr) exists? /dev/rtlsdr
(rtlsdr) checking /dev/rtlsdr
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii /bin/sh: rtl_wmbus: not found<0A>
(rtlwmbus) checkRTLWMBusFrame "/bin/sh: rtl_wmbus: not found<0A>"
(rtlwmbus) only text
(serial) received ascii No supported devices found.<0A>
(rtlwmbus) checkRTLWMBusFrame "No supported devices found.<0A>"
(rtlwmbus) only text
(bgshell) 8 exited with return code 127
(serial) closing non working fd=3
(serial) no devices working
(serial) stopping manager
(serialcmd) closed /bin/sh
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working
(serialcmd) closed /bin/sh
```

BIBOLV, 2020-02-24:
Could you try once more?

MTrab, 2020-02-24:
Gives this, and then the container stops.

```
(wmbusmeters) version: 0.9.26-23-g33952d8
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) using device: auto
(config) number of meters: 0
(detect) "auto" ""
(detect) driver: auto
(im871a) exists? /dev/im871a
(im871a) exists? /dev/im871a_0
(im871a) exists? /dev/im871a_1
(im871a) exists? /dev/im871a_2
(im871a) exists? /dev/im871a_3
(im871a) exists? /dev/im871a_4
(im871a) exists? /dev/im871a_5
(im871a) exists? /dev/im871a_6
(im871a) exists? /dev/im871a_7
(im871a) exists? /dev/im871a_8
(amb8465) exists? /dev/amb8465
(amb8465) exists? /dev/amb8465_0
(amb8465) exists? /dev/amb8465_1
(amb8465) exists? /dev/amb8465_2
(amb8465) exists? /dev/amb8465_3
(amb8465) exists? /dev/amb8465_4
(amb8465) exists? /dev/amb8465_5
(amb8465) exists? /dev/amb8465_6
(amb8465) exists? /dev/amb8465_7
(amb8465) exists? /dev/amb8465_8
(rfmrx2) exists? /dev/rfmrx2
(rfmrx2) exists? /dev/rfmrx2_0
(rfmrx2) exists? /dev/rfmrx2_1
(rfmrx2) exists? /dev/rfmrx2_2
(rfmrx2) exists? /dev/rfmrx2_3
(rfmrx2) exists? /dev/rfmrx2_4
(rfmrx2) exists? /dev/rfmrx2_5
(rfmrx2) exists? /dev/rfmrx2_6
(rfmrx2) exists? /dev/rfmrx2_7
(rfmrx2) exists? /dev/rfmrx2_8
(cul) exists? /dev/ttyUSB0
(cul) exists? /dev/ttyUSB0_0
(cul) exists? /dev/ttyUSB0_1
(cul) exists? /dev/ttyUSB0_2
(cul) exists? /dev/ttyUSB0_3
(cul) exists? /dev/ttyUSB0_4
(cul) exists? /dev/ttyUSB0_5
(cul) exists? /dev/ttyUSB0_6
(cul) exists? /dev/ttyUSB0_7
(cul) exists? /dev/ttyUSB0_8
(rtlsdr) exists? /dev/rtlsdr
(rtlsdr) checking /dev/rtlsdr
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop
```

BIBOLV, 2020-02-25:
Binaries are in image now. Can you check docker console for some clues/errors?
I will need to find my rtl_sdr dongle and test locally, hopefully I will be able to do it in upcoming weeks.

MTrab, 2020-02-25:
This is all:
`(wmbusmeters) using log file /wmbusmeters_data/logs/wmbusmeters.log`

weetmuts, 2020-02-26:
It looks like it has not found any meter configurations.
"No meters configured. Printing id:s of all telegrams heard!"

MTrab, 2020-02-26:
Correct, I haven't set up a specific meter as I'm not sure if it works - so I would just start by listening to all.

weetmuts, 2020-02-26:
Ok, well, when it printed that and did not exit, then I assume it should be working....
you should see things like:
Received telegram from: 12345678
          manufacturer: (KAM) Kamstrup Energi
           device type: Cold water meter
and surrounding debug data.

Have you tested and confirmed reception using a linux/mac computer without
docker?

MTrab, 2020-02-26:
But that's the issue - it DOES exit. The docker container starts and exits immidiatly when last line in the log is written.

I know rtl_sdr is working when manually installed on a Linux box, but I want this to run in a Docker on a Synology NAS.

weetmuts, 2020-02-26:
Ah, it the exit happens with no visible trace in any log file? Tricky.....

BIBOLV, 2020-02-26:
I can confirm that issues is with rtl_sdr within docker image - binary can`t find device. Recompilation from source didn`t help. I will continue to investigate this issue.

BIBOLV, 2020-02-27:
@MTrab please try downloading latest test image `docker run --name=wmbusmeters -v /etc/localtime:/etc/localtime:ro -v /volume2/docker/wmbusmeters/config:/wmbusmeters_data --device=/dev/rtlsdr bibolv/wmbusmeters_test:latest` and change `device=auto` to `device=rtlwmbus` in config

MTrab, 2020-02-27:
Still no luck :(

```
(wmbusmeters) version: 0.9.27
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) using device: rtlwmbus
(config) number of meters: 0
(detect) "rtlwmbus" ""
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop

```

BIBOLV, 2020-02-27:
And container image exited?

MTrab, 2020-02-27:
Yes

BIBOLV, 2020-02-27:
Ok, then your help will be needed for debug. This is what I figured so far - rtl_sdr needs actual path to device to work, so I created "dirty workaround" by creating fake path and symlinked it to /dev/rtlsdr inside docker image. In my case it was /dev/bus/usb/001/004 and it worked. What you could try is running same docker command but this time in `--device` use full path, like `--device=/dev/bus/usb/001/004` and also `device=rtlwmbus` in config. If that does not work then try to run `docker run -ti -v /etc/localtime:/etc/localtime:ro -v /volume2/docker/wmbusmeters/config:/wmbusmeters_data --device=[full_path_to_device] --entrypoint sh wmbusmeters_test` and then you will be inside container. Try running rtl_test and see if device is being detected, if it is then try running `rtl_sdr -f 868.95M -s 1.6e6 -` and if it also works then try running `wmbusmeters --listento=t1 --debug rtlwmbus`

weetmuts, 2020-02-27:
Ping @merbanan do you know who to supply the exact device to rtl_sdr ?
We have trouble getting rtl_sdr to use the device that is supplied to the docker
container. wmbusmeters+rtl_sdr+rtl_wmbus is inside the container.

merbanan, 2020-02-27:
Well I never actually need to do anything. The device is picked up via it's usb id.

merbanan, 2020-02-27:
Ok it looks like docker needs a device or the complete bus. So I think the proper way to do it is to add a udev rule for rtl_sdr devices upon start. And then pass that device (/dev/whatever) via docker.

The actual commands to do this is unknown.

MTrab, 2020-02-27:
Seems like `docker run --name=wmbusmeters -v /etc/localtime:/etc/localtime:ro -v /volume2/docker/wmbusmeters/config:/wmbusmeters_data --device=/dev/bus/usb/002/117 bibolv/wmbusmeters_test:latest` is working - thou no data is received, but that MIGHT be down to range issues (had hoped the RTLSDR antenna was that much better than my Amber device antenna ;)

merbanan, 2020-02-27:
@MTrab try running rtl_433 and see if it picks up anything

merbanan, 2020-02-27:
Can you run rtl_test also and post the output?

MTrab, 2020-02-27:
rtl_433:
`sh: rtl_433: not found`

rtl_test:
```
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...

```

MTrab, 2020-02-27:
`rtl_sdr -f 868.95M -s 1.6e6 -`
[image]

BIBOLV, 2020-02-27:
@MTrab but that is when full bus path is passed to docker. If udev rules are used on host and /dev/rtlsdr is passed to docker then rtl_sdr does not find device in docker

merbanan, 2020-02-27:
@MTrab rtl_433 is a tool you need to compile on your own.

merbanan, 2020-02-27:
@BIBOLV maybe a permission issue? Ive never used docker in that way.

BIBOLV, 2020-02-27:
@merbanan checked permissions - all good. I used udev rules on host, and passed /dev/rtlsdr to docker, running rtl_test did not find device in docker. Then I created /dev/bus/usb/001/ directory in docker and symlinked /dev/rtlsdr to /dev/bus/usb/001/004 and running rtl_test did find device.

weetmuts, 2020-03-08:
Bibo has done a lot of work for getting the docker container running.
Please check https://hub.docker.com/r/weetmuts/wmbusmeters
where the README manual should be up to date with the latest docker containers.

weetmuts, 2020-03-16:
Did you get it to work @MTrab  ?

MTrab, 2020-03-16:
Nope - gave up and digged out a Raspberry Pi and running on that now.

closed 2020-08-02 by weetmuts