wmbusmeters/wmbusmeters #93: Diehl IZAR RC868 I R4

Domochip, 2020-03-08:
Hi, 

I have some diffculties to read my (cold) water meter which is an IZAR RC868 I R4 (P/N : 3031799)
I'm using forked version of rtl-wmbus (weetmuts/rtl-wmbus -branch FixLengthByte) to get correct CRC check

I'm running this command to find counter : 
`wmbusmeters/build_debug/wmbusmeters --t1 rtlwmbus`

And Got : 
```
No meters configured. Printing id:s of all telegrams heard!

(wmbus) Unknown tpl-ci-field a1
Received telegram from: 20eec43b
          manufacturer: (SAP) Sappel
           device type: Oil meter
(wmbus) Unknown tpl-ci-field a1
Received telegram from: 20eec43b
          manufacturer: (SAP) Sappel
           device type: Oil meter
```

Then Run 
`wmbusmeters/build_debug/wmbusmeters --debug --t1 rtlwmbus MonCompt izar 20eec43b NOKEY`

And Got : 
```
(wmbusmeters) version: 0.9.28
(config) using device: rtlwmbus
(config) number of meters: 1
(detect) "rtlwmbus" ""
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) meter izar link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) explicitly listening to: t1
(config) listen link modes calculated to be: t1
(wmbusmeters) configured "MonCompt" "izar" "20eec43b" not-encrypted
(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii T1;1;1;2020-03-08 16:45:31.000;47;127;20EEC43B;0x1944304c3bc4ee209801a121150013c963c6259732060fa88138<0A>
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-03-08 16:45:31.000;47;127;20EEC43B;0x1944304c3bc4ee209801a121150013c963c6259732060fa88138<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 26
(meter) MonCompt: for me? 20eec43b
(meter) MonCompt: yes for me
(meter) MonCompt 20eec43b "1944304C3BC4EE209801A121150013C963C6259732060FA88138"
(wmbus) parseDLL @0 26
(wmbus) parseELL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(wmbus) Unknown tpl-ci-field a1
(wmbus) telegram ignored by all configured meters!

```

I don't understand this error which pop up : **Unknown tpl-ci-field a1**

Thanks on beforehands

weetmuts, 2020-03-10:
Yet another version of the izar... It seems to use another ci-field 0xa1 which
also means manufacturer_specific. I just pushed a small fix to handle 0xa1
and wmbusmeters now print:
382.175 m3	367.256 m3
does it look right to you?

Domochip, 2020-03-10:
Thanks a lot !!!
That's corresponding to my counter value.
It's OK.

closed 2020-03-10 by Domochip