wmbusmeters/wmbusmeters #94: Solve how to properly handle wmbus devices/dongles with/without udev rules.

weetmuts, 2020-03-09:
Currently wmbusmeters rely on hard-coded udev rules to start and stop the daemon.
This works nicely when installing from source. However there are some curiosities
when trying to create an rpm/deb package.

In particular the wmbus dongles often have generic serial chipsets, that 
in the broad sense does not need to be wmbus device at all, these vendor
chipsets are found in other rpm packages like:
    10c4:ea60 - provided by ModemManager,
    0403:6001 - provided by libftdi,
    0bda:2838 - provided by rtl-sdr,
    2047:0863 - currently not provided.

In fact, we already have two wmbus dongles with the same usb vendor version,
which makes it impossible already to create the right udev /dev/dongle link
since we cannot differentiate between the two devices on the udev level.

The only way to differentiate between the dongles is to start talking serial data
with them and detect what is on the other side.

Sooo, it seems like the proper way would be to improve wmbusmeters to find
all serial usb devices, and then probe them, without relying on udev.

However I want wmbusmeters to start automatically if a wmbus dongle is inserted,
so perhaps a broad udev rule (looking for serial devices) is needed to trigger an auto scan for all possible wmbus dongles?

BIBOLV, 2020-03-09:
Looking from package perspective - rpm/deb:
It is more cumbersome, but maybe udev rules can be packed in additional package, that way, if user does not want this feature, he can opt-out. Otherwise it will not be possible in package as it is currently possible with install script. 
Regarding udev rules - IMO automatic udev trigger for all usb devices would work for specific device, but not generic ones like rtlsdr. Like if rtlsdr is plugged in, wmbusmetersd is started, but user wants to use this device for different purposes. How multiple connected devices would be treated? 
Also if udev triggers start of service, would be nice if service would be stopped same way.
So taking in account above and if it is relevant then I would say that udev rules should be managed by user if he want`s it, but it goes against your view....

weetmuts, 2020-03-09:
Yes, a separate package is a nice idea. 

Or perhaps we could move the auto-detect functionality 
into a separate invocation to wmbusmeters, like:
sudo wmbusmeters --detect-and-install
which would scan for devices and then install the proper
udev rules and update the wmbusmeters.conf file.

BIBOLV, 2020-03-09:
Yes, that`s could work

weetmuts, 2020-03-09:
What do you say @dwrobel  ?

dwrobel, 2020-03-10:
>The only way to differentiate between the dongles is to start talking serial data
with them and detect what is on the other side.

There is no guarantee that each device will always contain some unique serialization data.

The alternative is to rely on `udev` [ID_PATH=](https://www.freedesktop.org/software/systemd/man/systemd.link.html#Path=) property which should be persistent, as per documentation, and taking advantage of that it would allows you to differentiate identical devices connected to a different physical ports.

weetmuts, 2020-09-26:
The branch NewDetectionCode with no udev rules,
now seems good enough to have a broader audience test it.

Please join the discussion in #169

closed 2020-09-26 by weetmuts