wmbusmeters/wmbusmeters #95: multical302: ignoring compressed telegram

Epyon01P, 2020-03-09:
I have two multical302's sending encrypted messages. wmbusmeters successfully captures and reconstructs the telegrams, but I always get the _ignoring compressed telegram since format signature hash 0xeb56 is yet unknown_ error, even if I wait many cycles. This happens on both meters.

Below is an excerpt of the debug output. I read in #44 that the multical302 code is a WiP. If it's any help, I can deliver telegrams from at least five different meters.

(wmbusmeters) version: 0.9.27-8-g6f32c28
(config) using device: /dev/ttyUSB0
(config) with: amb8465
(config) number of meters: 1
(detect) "/dev/ttyUSB0" "amb8465"
(detect) is_tty=1 is_stdin=0 is_file=0
(amb8465) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) meter multical302 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) explicitly listening to: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "test" "multical302" "67995929" encrypted
(amb8465) set link mode 0e
(serial /dev/ttyUSB0) sent "FF04010EF4000000"
(serial) received binary FF8401007A
(amb8465) checkAMB8465Frame "FF8401007A"
(amb8465) received full command frame
(amb8465) rssi 122
(amb8465) set link mode completed
(amb8465) set link mode response "0100"
(amb8465) get config
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) received binary FF8A7A00780080
(amb8465) checkAMB8465Frame "FF8A7A00780080"
(amb8465) not enough bytes yet, partial command response 7 126.
(serial) received binary 710200000000FFFFFA00FFFFFFFF
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFF"
(amb8465) not enough bytes yet, partial command response 21 126.
(serial) received binary FFFFFFFFFFFFFFFFFFFFFFFF00320002
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00320002"
(amb8465) not enough bytes yet, partial command response 37 126.
(serial) received binary 1400FFFFFFFFFF010004000000FFFFFF
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF"
(amb8465) not enough bytes yet, partial command response 53 126.
(serial) received binary 01440000000000000000FFFF0B04
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B04"
(amb8465) not enough bytes yet, partial command response 67 126.
(serial) received binary 0100FFFFFFFFFF00030000FFFFFFFFFF
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 83 126.
(serial) received binary FFFF0000FFFFFFFFFFFFFFFFFFFFFFFF
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFF"
(amb8465) not enough bytes yet, partial command response 99 126.
(serial) received binary FFFFFFFFFFFFFFFFFFFFFFFFFFFF
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF$
(amb8465) not enough bytes yet, partial command response 113 126.
(serial) received binary FFFFFFFFFFFFFFFFFFFFFFFF17
(amb8465) checkAMB8465Frame "FF8A7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF$
(amb8465) received full command frame
(amb8465) rssi 23
(amb8465) get config completed
(amb8465) get config response "7A00780080710200000000FFFFFA00FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF003200021400FFFFFFFFFF010004000000FFFFFF01440000000000000000FFFF0B040100FFFFFFFFFF00030000FFFFFFFFFFFFFF0000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF$
(amb8465) config: uart 78
(amb8465) config: radio Channel ff
(amb8465) config: rssi enabled ff
(amb8465) config: mode Preselect 00
(config) listen to link modes: c1
(serial) waiting for stop
(serial) received binary 24442D2C29599967300D8D2085A080
(amb8465) checkAMB8465Frame "24442D2C29599967300D8D2085A080"
(amb8465) not enough bytes yet, partial frame 15 37.
(serial) received binary 092034FE5B3FBB3644C472568678E1
(amb8465) checkAMB8465Frame "24442D2C29599967300D8D2085A080092034FE5B3FBB3644C472568678E1"
(amb8465) not enough bytes yet, partial frame 30 37.
(serial) received binary 11653079C829BD
(amb8465) checkAMB8465Frame "24442D2C29599967300D8D2085A080092034FE5B3FBB3644C472568678E111653079C829BD"
(amb8465) received full frame
(amb8465) rssi 8
(wmbus) parseDLL @0 37
(meter) test: for me? 67995929
(meter) test: probably not for me since media does not match
(meter) test: yes for me
(meter) test 67995929 "24442D2C29599967300D8D2085A080092034FE5B3FBB3644C472568678E111653079C829BD"
(wmbus) parseDLL @0 37
(wmbus) parseELL @10 27
(ELL) decrypting "34FE5B3FBB3644C472568678E111653079C829BD"
(ELL) IV 2D2C29599967300D20A0800920000000
(ELL) decrypted first block "27F47956EB02A917000900000000009D"
(ELL) IV+1 2D2C29599967300D20A0800920000001
(ELL) decrypted second block "22000000"
(ELL) decrypted "34FE5B3FBB3644C472568678E111653079C829BD"
(wmbus) parseAFL @19 18
(wmbus) parseTPL @19 18
(wmbus) ignoring compressed telegram since format signature hash 0xeb56 is yet unknown.
     this is not a problem, since you only need wait for at most 8 telegrams
     (8*16 seconds) until an full length telegram arrives and then we know
     the format giving this hash and start decoding the telegrams properly.
(wmbus) telegram ignored by all configured meters!

weetmuts, 2020-03-10:
Can you run with --debug (or just --logtelegrams) for a while to collect a number of telegrams,
that is the lines that has telegram=|.....|...| on them.

I need both long and short telegrams, to see why the format hash calculation for the long
ones do not match the short ones. You can email me the log if you like.

Epyon01P, 2020-03-10:
Hmm, the above log is with the --debug option. If I run with --logtelegrams the only output is 
(meter) test: probably not for me since media does not match
(wmbus) payload crc error!

If I run with --debug, I get the above error too sometimes, other times just
(meter) test: probably not for me since media does not match
(meter) test: yes for me
(meter) test 67995932 "24442D2C32599967300D8D205670B209203CFC9366250AC792E801F54FCC53C7AEF817DF0A"
(wmbus) parseDLL @0 37
(wmbus) parseELL @10 27
(ELL) decrypting "3CFC9366250AC792E801F54FCC53C7AEF817DF0A"
(ELL) IV 2D2C32599967300D2070B20920000000
(ELL) decrypted first block "D0567956EB084E48001000040003009D"
(ELL) IV+1 2D2C32599967300D2070B20920000001
(ELL) decrypted second block "22000000"
(ELL) decrypted "3CFC9366250AC792E801F54FCC53C7AEF817DF0A"
(wmbus) parseAFL @19 18
(wmbus) parseTPL @19 18
(wmbus) ignoring compressed telegram since format signature hash 0xeb56 is yet unknown.
     this is not a problem, since you only need wait for at most 8 telegrams
     (8*16 seconds) until an full length telegram arrives and then we know
     the format giving this hash and start decoding the telegrams properly.

weetmuts, 2020-03-11:
Hmmm, just collect a looong --debug log and email me. :-)

weetmuts, 2020-03-13:
Found the problem! Silly me,  the AES-CTR only decrypted the two first 16 byte blocks of data. But here was the first telegram longer than two blocks. 

Please test!

Epyon01P, 2020-03-13:
I can confirm wmbusmeters now happily decrypts the C1 telegrams coming from our multical302s. Many thanks for fixing this issue so fast!

closed 2020-03-15 by weetmuts