wmbusmeters/wmbusmeters #98: No signal from Apator at-wimbus-16-2

tomkos, 2020-03-22:
Hi. I'm trying to capture data from my apator at-wimbus-16-2. Everything looks fine, but there is no data capture... does this meter transmits data only periodically?

Below the log from wbmustmeters using debug mode. 

`:~/repo$ wmbusmeters --listento=t1 --debug auto
(wmbusmeters) version: 0.9.28-3-gea27a03
(config) using device: auto
(config) number of meters: 0
(detect) "auto" ""
(detect) driver: auto
(im871a) exists? /dev/im871a
(im871a) exists? /dev/im871a_0
(im871a) exists? /dev/im871a_1
(im871a) exists? /dev/im871a_2
(im871a) exists? /dev/im871a_3
(im871a) exists? /dev/im871a_4
(im871a) exists? /dev/im871a_5
(im871a) exists? /dev/im871a_6
(im871a) exists? /dev/im871a_7
(im871a) exists? /dev/im871a_8
(amb8465) exists? /dev/amb8465
(amb8465) exists? /dev/amb8465_0
(amb8465) exists? /dev/amb8465_1
(amb8465) exists? /dev/amb8465_2
(amb8465) exists? /dev/amb8465_3
(amb8465) exists? /dev/amb8465_4
(amb8465) exists? /dev/amb8465_5
(amb8465) exists? /dev/amb8465_6
(amb8465) exists? /dev/amb8465_7
(amb8465) exists? /dev/amb8465_8
(rfmrx2) exists? /dev/rfmrx2
(rfmrx2) exists? /dev/rfmrx2_0
(rfmrx2) exists? /dev/rfmrx2_1
(rfmrx2) exists? /dev/rfmrx2_2
(rfmrx2) exists? /dev/rfmrx2_3
(rfmrx2) exists? /dev/rfmrx2_4
(rfmrx2) exists? /dev/rfmrx2_5
(rfmrx2) exists? /dev/rfmrx2_6
(rfmrx2) exists? /dev/rfmrx2_7
(rfmrx2) exists? /dev/rfmrx2_8
(cul) exists? /dev/ttyUSB0
(cul) exists? /dev/ttyUSB0_0
(cul) exists? /dev/ttyUSB0_1
(cul) exists? /dev/ttyUSB0_2
(cul) exists? /dev/ttyUSB0_3
(cul) exists? /dev/ttyUSB0_4
(cul) exists? /dev/ttyUSB0_5
(cul) exists? /dev/ttyUSB0_6
(cul) exists? /dev/ttyUSB0_7
(cul) exists? /dev/ttyUSB0_8
(rtlsdr) exists? /dev/rtlsdr
(rtlsdr) exists? /dev/rtlsdr_0
(rtlsdr) exists? /dev/rtlsdr_1
(rtlsdr) exists? /dev/rtlsdr_2
(rtlsdr) checking /dev/rtlsdr_2
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop`

After "waiting for stop" is printed nothing happens... for minutes, I would say even like an hour?

weetmuts, 2020-03-22:
The apator162 can be configured to not transmit during nighttime.
Have you listened to it during the day?

tomkos, 2020-03-22:
Thanks, I did some short test today but I'm not sure that everything was working fine, I will try it tomorrow.. Good night and thank You

tomkos, 2020-04-13:
Still no luck. 
I've tried during a day, but no telegrams are captured. 

I've tried to listen on 898.95 and 898.9. But nothing, there is no errors, as well. 

I've tried to run rtl_test - to check if my packages are not lost, but it looks fine. 
I'm able to receive other signals i.e. airplane ADS-B Signals. 
Below I'm adding detail log. Not sure how to debug it further. 

```
:~/repo$ /usr/bin/rtl_sdr -f 898.9M -s 1.6e6 - | /usr/bin/rtl_wmbus 
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 1600000 S/s.
Tuned to 898900000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

User cancel, exiting...
Reattached kernel driver

~/repo$ /usr/bin/rtl_sdr -f 898.95M -s 1.6e6 - | /usr/bin/rtl_wmbus 
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 1600000 S/s.
Tuned to 898950000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers
^CSignal caught, exiting!

User cancel, exiting...
Reattached kernel driver

osboxes@osboxes:~/repo$ rtl_test -s 1.6e6
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 1600000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 20 bytes
^CSignal caught, exiting!

User cancel, exiting...
Samples per million lost (minimum): 0
Reattached kernel driver
```

tomkos, 2020-04-13:
It works ! I was using wrong frequency. 
It started to work on 868.9M.

closed 2020-04-13 by tomkos