wmbusmeters/wmbusmeters #99: Lorenz water meter CRC issue

makuser, 2020-03-24:
I would like to use wmbusmeters to read Lorenz water meters, which do not seem to be implemented at the moment.

They are configured to use OMS C1 telegrams (OMS v3).

I do however get CRC errors when receiving said telegrams, so I don't know if it's enough to add the meter as explained [here](https://github.com/weetmuts/wmbusmeters/blob/master/HowToAddaNewMeter.txt).

```
(serial) received ascii C1;0;1;2020-03-24 13:24:15.000;121;124;19062913;0x9644fa121329061900077ad7000020046d182e982304135700000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000042018980000000620189800000004404130000000044020980000000020144c0800000310144c0000000020184c00000c0310184c00000002101c4c00000002281c4c000000022040980000000<0A>
(rtlwmbus) checkRTLWMBusFrame "C1;0;1;2020-03-24 13:24:15.000;121;124;19062913;0x9644fa121329061900077ad7000020046d182e982304135700000002fd17000001fd481d426c7f2c44136300000084011362000000c401136300000084021363000000c402130000000042018980000000620189800000004404130000000044020980000000020144c0800000310144c0000000020184c00000c0310184c00000002101c4c00000002281c4c000000022040980000000<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
```

How shall I proceed now?

weetmuts, 2020-03-24:
If they are OMS compliant, then it should be easy to add support for the Lorenz meter.

What you are seeing here is that rtl_wmbus failed to decode the radio telegram correctly.
The telegram is detected and decoded, but some bits are wrong, thus the crc failure.

Can you move closer to the meter and see if you get better reception?

makuser, 2020-03-24:
> If they are OMS compliant, then it should be easy to add support for the Lorenz meter.

Yep, they are. Btw I now found out that the meter is based on the Engelmann (manufacturer code: EFE) Waterstar M https://www.engelmann.de/de/waterstar-m-2/. But this one is also not supported, is it?

> What you are seeing here is that rtl_wmbus failed to decode the radio telegram correctly.
> The telegram is detected and decoded, but some bits are wrong, thus the crc failure.
> 
> Can you move closer to the meter and see if you get better reception?

Well that's the thing. The meter I am trying to read is right next to my antenna. I am receiving about 30 other meters, from nearby houses that must be MUCH FURTHER away than mine, yet yield no CRC errors.

weetmuts, 2020-03-24:
Actually, I have been told that sitting too close to the meter will
sometimes disrupt the reception of rtl_wmbus.
I.e. the signal is too strong and rtl_wmbus is not written to handle that.

Can you try further away? :-) A wall might attenuate the signal nicely.

weetmuts, 2020-05-09:
I am closing this one because I cannot really assist in the rtl_wmbus radio decoding.

closed 2020-05-09 by weetmuts