wmbusmeters/wmbusmeters #101: unexpected meter version 0x05

Dragonidas888, 2020-03-31:
When I try to decrypt the data from the meter I get:

user@raspi4:~ $ echo "T1;1;1;2020-03-31 07:24:23.000;58;65;XXXXXXXX;0x4644b409XXXXXXXX05077aa50006101115f78184ab0f1d1e200000005904103103208047004a4800e73c00193e00453f003e4000e64000e74100f442000144001545005b460000<0A>" | wmbusmeters  --debug --verbose --format=json stdin:rtlwmbus MyWater hydrodigit XXXXXXXX NOKEY
(wmbusmeters) version: 0.9.28-14-gbb1f03f
(config) using device: stdin
(config) with: rtlwmbus
(config) number of meters: 1
(detect) "stdin" "rtlwmbus"
(detect) is_tty=0 is_stdin=1 is_file=0
(serial) override with devicefile: stdin
(serialfile) reading from stdin
(config) meter hydrodigit link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(wmbusmeters) configured "MyWater" "hydrodigit" "XXXXXXXX" not-encrypted
(config) listen to link modes: any
(serial) waiting for stop
(serial) no more data on fd=0
(serial) received binary "54313B313B313B323032302D30332D33312030373A32343A32332E3030303B35383B36353B31353132333033373B3078343634346234303933373330313231353035303737616135303030363130313131356637383138346162306631643165323030303030303035393034313033313033323038303437303034613438303065373363303031393365303034353366303033653430303065363430303065373431303066343432303030313434303031353435303035623436303030303C30413E0A"
(serial) no devices working
(serial) stopping manager
(serialtty) closed stdin -1
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-03-31 07:24:23.000;58;65;XXXXXXXX;0x4644b409XXXXXXXX05077aa50006101115f78184ab0f1d1e200000005904103103208047004a4800e73c00193e00453f003e4000e64000e74100f442000144001545005b460000<3C>0A<3E><0A>"
(rtlwmbus) received full frame
(rtlwmbus) warning: the hex string contains bad characters! Decode stopped partway.
(wmbus) parseDLL @0 71
(meter) MyWater: for me? XXXXXXXX
(meter) MyWater: yes for me
(meter) MyWater XXXXXXXX "4644B409XXXXXXXX05077AA50006101115F78184AB0F1D1E200000005904103103208047004A4800E73C00193E00453F003E4000E64000E74100F442000144001545005B460000"
(wmbus) parseDLL @0 71
(wmbus) parseELL @10 61
(wmbus) parseAFL @10 61
(wmbus) parseTPL @10 61
**(dvparser) cannot handle dif 3F ignoring rest of telegram.
(dvparser) found new format "11158184AB0F1D20000000590431038047004A483C00" with hash efb4, remembering!**
(telegram) DLL L=46 C=44 (from meter SND_NR) M=09b4 (BMT) A=XXXXXXXX VER=05 TYPE=07 (Water meter)
(telegram) TPL CI=7a ACC=a5 STS=00 CFG=1006 (SPECIFIC_16_31)
(hydrodigit) unexpected meter version 0x05 !
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077AA5000610|1115F78184AB0F1D1E200000005904103103208047004A4800E73C00193E00453F003E4000E64000E74100F442000144001545005B460000|+0"
(hydrodigit) 00: 46 length (70 bytes)
(hydrodigit) 01: 44 dll-c (from meter SND_NR)
(hydrodigit) 02: b409 dll-mfct (BMT)
(hydrodigit) 04: XXXXXXXX dll-id (XXXXXXXX)
(hydrodigit) 08: 05 dll-version
(hydrodigit) 09: 07 dll-type (Water meter)
(hydrodigit) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(hydrodigit) 0b: a5 tpl-acc-field
(hydrodigit) 0c: 00 tpl-sts-field
(hydrodigit) 0d: 0610 tpl-cfg 1006 (SPECIFIC_16_31 )
(hydrodigit) 0f: 11 dif (8 Bit Integer/Binary Maximum value)
(hydrodigit) 10: 15 vif (Volume 10⁻¹ m³)
(hydrodigit) 11: * F7 total consumption (24.700000 m3)
(hydrodigit) 12: 81 dif (8 Bit Integer/Binary Instantaneous value)
(hydrodigit) 13: 84 dife (subunit=0 tariff=0 storagenr=8)
(hydrodigit) 14: AB dife (subunit=0 tariff=8 storagenr=360)
(hydrodigit) 15: 0F dife (subunit=0 tariff=8 storagenr=8040)
(hydrodigit) 16: 1D vif (Mass 10² kg)
(hydrodigit) 17: 1E
(hydrodigit) 18: 20 dif (No data Minimum value)
(hydrodigit) 19: 00 vif (Energy mWh)
(hydrodigit) 1a: 00 dif (No data Instantaneous value)
(hydrodigit) 1b: 00 vif (Energy mWh)
(hydrodigit) 1c: 59 dif (2 digit BCD Maximum value storagenr=1)
(hydrodigit) 1d: 04 vif (Energy 10¹ Wh)
(hydrodigit) 1e: 10
(hydrodigit) 1f: 31 dif (8 Bit Integer/Binary Value during error state)
(hydrodigit) 20: 03 vif (Energy Wh)
(hydrodigit) 21: 20
(hydrodigit) 22: 80 dif (No data Instantaneous value)
(hydrodigit) 23: 47 dife (subunit=1 tariff=0 storagenr=14)
(hydrodigit) 24: 00 vif (Energy mWh)
(hydrodigit) 25: 4A dif (4 digit BCD Instantaneous value storagenr=1)
(hydrodigit) 26: 48 vif (Volume flow ext. mm³/s)
(hydrodigit) 27: 00E7
(hydrodigit) 29: 3C dif (8 digit BCD Value during error state)
(hydrodigit) 2a: 00 vif (Energy mWh)
(hydrodigit) 2b: 193E0045
{"media":"water","meter":"hydrodigit","name":"MyWater","id":"XXXXXXXX","total_m3":24.7,"meter_datetime":"","timestamp":"2020-03-31T19:04:35Z"}
(serial) event loop stopped!
(serial) closing 0 devices

Water meter: "BMeter RFM-TX1.1 IP65".
Meter state is over 180 m3.
Can I get the correct data from it?

Thank you very much for all your help.

weetmuts, 2020-04-01:
The decoded bytes are just random noise.
But there are still a lot of zeroes, which indicates that it is not encrypted.

To be OMS compliant the tpl_config should not be tpl-cfg 1006 (SPECIFIC_16_31)
normally it is either 0 or 5 for AES encrypted data.

Can you collect some more telegrams and check if the tpl-cfg are always 1006
(0610 in the rtlwmbus output) and that tpl_acc increments per telegram?

Dragonidas888, 2020-04-01:
All data collected so far (over 6000 records) have a constant value for the tpl-cfg field equal 1006.
The tpl_acc field increases by 1 for the each next telegram. 

(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A70000610|09F57DE6727A0F1D1E200000001425150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+281"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A72000610|CD79BCFFE4EB0F1D1E200000004526150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+372"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A73000610|F11D7DF221650F1D1E200000003027150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+417"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A76000610|C7E59D8E0EEF0F1D1E200000004429150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+551"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A77000610|5C38B435EFD40F1D1E200000003030150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+597"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A78000610|44BB0530797B0F1D1E200000001531150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+642"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A79000610|B726FEFAC4FF0F1D1E200000000032150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+687"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A7A000610|22221B6473240F1D1E200000004432150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+731"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A7B000610|3C8600341ED90F1D1E200000002933150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+776"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A7C000610|F7A541B280630F1D1E200000001434150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+821"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A7D000610|B5E97DD5EF950F1D1E200000005934150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+866"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A7E000610|915DB48031F20F1D1E200000004435150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+911"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A80000610|F9058D16828A0F1D1E200000001437150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1001"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A81000610|D6ED8614BB920F1D1E200000000038150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1047"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A82000610|3D894C0F141B0F1D1E200000004438150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1091"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A83000610|01ED8D02D1950F1D1E200000002939150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1135"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A86000610|37156D7EFE1F0F1D1E200000004541150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1272"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A89000610|47D60E0A340F0F1D1E200000000044150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1407"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A8E000610|61AD4470C1020F1D1E200000004547150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1632"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A8F000610|2706C166BF8A0F1D1E200000003048150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1677"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A91000610|C6FD9604AB820F1D1E200000000050150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1767"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A92000610|2D995C1F040B0F1D1E200000004450150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1811"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A93000610|11FD9D12C1850F1D1E200000003051150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1857"
(hydrodigit) log "telegram=|4644B409XXXXXXXX05077A94000610|2D16E48847FD0F1D1E200000001552150104208047004A4800AC4900193E00453F003E4000E64000E74100F442000144001545005B460000|+1902"

weetmuts, 2020-04-02:
I have been told that when the meter reports a  tpl-cfg 1006,
this means that it is not an OMS-compliant meter and it uses
a proprietary encryption.

If you can reconfigure it to not use encryption, then 
it will revert a format that can be decoded by wmbusmeters.

Dragonidas888, 2020-04-02:
Information from the manufacturer for this meter:
WMBUS EN14757-4 @ 868 MHz radio interface [<10 mW] OMS compatible.

I cant change configuration  for it.

weetmuts, 2020-04-04:
Well, I do not have any information on how to decrypt these
proprietary telegrams. Sorry!

weetmuts, 2020-04-20:
Maybe they sent you the wrong type of meter? When the  tpl-cfg field equals 1006 then it is not an OMS-compliant meter and we do not know how the proprietary encryption works.

Closing this for now. Please re-open if you get more information or get an updated meter, or manage to disable the encryption. Then I can help you with decoding this meter.

bbiegun, 2020-05-27:
I managed to dig into the telegram format. For now there is: overall usage, datetime on meter, and historical data for a year. Need to do some cleanup in the code though.
Would C# or python suits you for incorporating the code? :)

weetmuts, 2020-05-27:
Great work! c# preferred.

bbiegun, 2020-05-27:
Here you have first part! 
telegramByte contains bytes of telegram. It's produce overall usage in m3 and local meter datetime. Extraction of historical data I'll share soon. 

Would you like to receive some training data with expected results?

P.S. maybe it'd be worth to reopen this issue for tracking purpose? I cannot do it unfortunately.

`                    // telegramByte = [0x46, 0x44, 0xB4, 0x09, 0xXX, 0xXX, 0xXX, 0xXX, ...]

                    // if version == 0x5 && tpl-cfg 1006 (SPECIFIC_16_31 )

                    byte[,] decryptVectors = new byte[16, 6] { { 117, 150, 122, 16, 26, 10 }, { 91, 127, 112, 19, 34, 19 }, { 179, 24, 185, 11, 142, 153 }, { 142, 125, 121, 7, 74, 22 }, { 181, 145, 7, 154, 203, 105 }, { 184, 163, 50, 161, 57, 14 }, { 189, 128, 156, 126, 96, 153 }, { 39, 92, 180, 196, 128, 163 }, { 48, 208, 10, 206, 25, 3 }, { 194, 76, 240, 5, 165, 134 }, { 84, 75, 22, 152, 17, 94 }, { 75, 238, 12, 201, 125, 162 }, { 135, 202, 74, 72, 228, 31 }, { 196, 135, 119, 46, 138, 232 }, { 227, 48, 189, 120, 87, 140 }, { 164, 154, 57, 111, 40, 5 } };
                    byte[] decryptData = new byte[6];

                    for (int i = 0; i < 6; ++i)
                        decryptData[i] = (byte)(telegramByte[0xf + i] ^ telegramByte[0xb] ^ decryptVectors[telegramByte[0xb] & 0x0f, i]);

                    string str = "";
                    for (int i = 2; i < decryptData.Length; ++i)
                        str = Convert.ToString(decryptData[i], 16).PadLeft(2, '0') + str;

                    double overallUsage = Double.Parse(str) / 1000; //in m3

                    str = "";
                    for (int i = 0; i < 6; ++i)
                        str = Convert.ToString(telegramByte[0x1c + i], 16).PadLeft(2, '0') + str;

                    DateTime localMeterTime = DateTime.ParseExact(str, "yyMMddHHmmss", null);
`

weetmuts, 2020-05-28:
It seems like there are two kinds of BMeter water meters:

the older version of the meter RFM-TX1 which has a mostly proprietary protocol, but which has been decoded by @bbiegun 

the newer version RFM-TX1.1 which is somewhat OMS compliant, this is the one wmbusmeters currently supports.

So I will rename the current meter in wmbusmeters to rfmtx11 (for the new version) and use rfmtx1 (for the old version.)

bbiegun, 2020-05-28:
I have RFM-TX1.1 mounted/from on 2015 and it's using proprietary protocol. It's marked as V12. All meters here were mounted in the same year so I have no other version to play unfortunately.

weetmuts, 2020-05-28:
Ah, thanks for that info @bbiegun !

bbiegun, 2020-05-29:
The historical bit. It shows usage in particular months and overrides when next month occurring (according to the meter). @weetmuts are you interested in all kinds of errors/frauds or battery level?
@Dragonidas888 - I would parse your first telegram as 188,56m3, local meter time 2020-03-31 10:04:59 with historical data: 
183,04;185,06;155,91;158,97;161,97;164,46;166,14;168,71;171,4;174,09;176,85;180,11. Would it be closer to the truth?
`
                double[] history = new double[12];
                for (int i = 0; i < 12; ++i)
                {
                    history[i] = (telegramByte[0x22 + i * 3] + telegramByte[0x22 + i * 3 + 1] * 0x100 + telegramByte[0x22 + i * 3 + 2] * 0x10000)/100.0;
                }`

weetmuts, 2020-05-30:
@Dragonidas888 @bbiegun 

The rfmtx1 driver is now updated with the code that @bbiegun supplied! :-)

It gives the total consumption of 188.56 m3 for your telegrams @Dragonidas888 ,
is that correct?

Please test!

bbiegun, 2020-05-30:
Thanks a ton! :)
Mine is quiet in the evenings and in  the weekends so I can test it live at Monday. However I ran a simulation and it looks very good! It's consistent with analogue readings I made that day.

bbiegun, 2020-06-01:
I need to say that it's working perfectly :)

`(rfmtx1) log "telegram=|4644B4090102030405077AB7000610|9CF8256021140F1D1E00000000032308010620C53A00DB3B00143D00D13D00033F006E3400253500CE3500A236007E370042380076390000|+134"
{"media":"water","meter":"rfmtx1","name":"a","id":"04030201","total_m3":161.326,"meter_datetime":"2020-06-01 08:23:03","timestamp":"2020-06-01T05:45:12Z"}
(rfmtx1) log "telegram=|4644B4090102030405077AB8000610|847B9565B7BB0F1D1E00000000482308010620C53A00DB3B00143D00D13D00033F006E3400253500CE3500A236007E370042380076390000|+180"
{"media":"water","meter":"rfmtx1","name":"a","id":"04030201","total_m3":161.327,"meter_datetime":"2020-06-01 08:23:48","timestamp":"2020-06-01T05:45:58Z"}
`

Dragonidas888, 2020-06-21:
Great job ! Now it's working perfectly. Thanks a lot for help :)

closed 2020-06-30 by weetmuts