wmbusmeters/wmbusmeters #102: Execute shell only on data change

alchemyx, 2020-04-01:
Hi,

Would it be possible that the wmbusmeters was executing command from "shell" entry only on data change? I am storing data into InfluxDB and I get tons of the same readings into the DB with the same set of values.

Kind regards,
Michał

weetmuts, 2020-04-04:
Yes, we could add a setting --triggeron=total_m3,status
(trigger=total_m3 in conf or meter file)
that would prevent wmbusmeters from logging/shelling
unless total_m3 and/or status has changed since last telegram.

The reason we have to use explicitly set fields, is that some meters send
their own datetime in  the telegram, and this of course changes with 
each telegram.

alchemyx, 2020-04-04:
That sounds like an excellent idea. Happy to test it whenever it's ready.
Thanks for great job.

majekw, 2020-04-04:
I could be also good to have some timeout or time period to limit squeezing values.

For example my water meters broadcasts reading every 15 minutes (or every 1-5 minutes at the beginning of every month for few days), but value changes only once a day.
So, for me limiting executing script for once every hour if value is the same would be optimal - not so many data points, but also I'll get some of them to draw some graph and see if meters are still alive.

weetmuts, 2020-05-09:
I just added --selectfields=...
in preparation of doing --triggeron=...

However when I read your last coment, it looks more like you need
something like --shelltimestamp=always,minute,hour,day

So, the default is always as before, --shelltimestamp=hour
would mean that it would only do one shell invocation per hour.
(Whenever that telegram arrived inside that hour.)

Right?

weetmuts, 2020-05-26:
Ping @majekw ?

weetmuts, 2020-06-30:
Closing this for now. Not a bad idea so it might get implemented in the future.

closed 2020-06-30 by weetmuts