wmbusmeters/wmbusmeters #104: wmbusmeters library and CMake build

momobel, 2020-04-10:
Hi,

We would like to use wmbusmeters as a library to be able to have a custom program on top. The goal is to be able to use a custom WMBus child interface, a custom Meter interface and a custom application.
To achieve this, CMake would be great for configuration, build, installation, as well as all this being available for cross-compilation.

I believe this would be a great addition to wmbusmeters so it is more modular and easy to customize, integrate and use.

You can find initial draft changes in wmbusmeters [here](https://github.com/WattSense/wmbusmeters/pull/1) as well as an example application [here](https://github.com/WattSense/using-libwmbusmeters-example).
It contains how to build and run.

Would you be interested in having this included in wmbusmeters?
If you are, please let me know and I could finalize the changes as well as create a PR to this repository.

Thanks

weetmuts, 2020-04-10:
Thank you for your suggestion. Note that wmbusmeters is licensed under GPLv3 and this is intended, because wmbusmeters is supposed to be the low level daemon running on your computer, decoding incoming wmbus data and transforming it into simpler json structs, that can be consumed by other applications. (other formats like xml or protobuf could also be supported in addition to json and other ways to sending the data, like zeromq, in addition to shell commands, could also be supported.)

Despite wmbus being a standard, there is always some amount of meter specific interpretation that has to be performed to full understand a wmbus-compatible telegram. Sometimes less, sometimes more. I want the knowledge of how to decode the wmbus-compatible meters to be part of wmbusmeters and shared with everyone. If wmbusmeters becomes a library, then the knowledge of how to understand the meter telegrams might be hidden inside an application. Since the GPLv3 is used, if you create a wmbusmeters library and link to your application, then you must license your application under the GPLv3 as well.

So I do not think that a library version is necessary, nor desirable. Also wmbusmeters
has not reached version 1.0 yet, this is because I am still working on the significantly
improved DifVif parser that will cover allmost all of the SND_NR wmbus telegrams possible.
So creating an API now, would be too early anyway.

>>>use a custom WMBus child interface (such as receiving wM-Bus frames over HTTP/TCP/UDP/zeromq or any other)

Yes, I have already thought of adding more ways to input telegrams into wmbusmeters. Adding a tcp/ip/udp port for receving telegrams would be pretty easy to build. Since you can feed wmbusmeters on stdin, any tool that receives network data and outputs it on stdout can be used to feed wmbusmeters, like this for example:

mosquitto_sub -h "broker ip" -t rtl_wmbus/events/ | ./adapt_msg.py  | wmbusmeters  --listento=any stdin:rtlwmbus

>>>use a custom Meter interface (that does DIF/VIF/extraction from a configuration file)

Yes, my intention is that each meter will be more and more like a configuration file. Eventually the difvifparser will be so good that there is a generic meter, that can output a somewhat sensible json structure without any configuration. Then the meter configuration file would be a simple mapping between the generic json data and meter specific json data, that has better names and add textual interpretation of bits in status fields.

Then I can accept that it is possible to offer custom meter configurations that are not part of wmbusmeters, and not necessarily licensed under the GPLv3 since they are configuration files.
(Even though a lot of configurations will be shipped with wmbusmeters under the GPLv3.)

weetmuts, 2020-04-10:
What are the platforms you want to cross-compile to, that is currently not supported?

Does cmake automatically make it possible to cross-compile to those?

momobel, 2020-04-14:
Hi weetmuts,

Thanks for your answer and your time.

First off I would like to say that it is not our intention to hide decoding of wM-Bus meters. In the end they seem to mostly use standard DIF/VIF.

We will most likely add a generic meter that can be decoded through the DIV/VIF settings defined in its configuration.

I see that you may be interested in including an additional interface (zeromq, TCP, UDP, or other) for feeding data to wmbusmeters.
What do you see this interface covering among the following?
- sending wM-Bus frames to wmbusmeters
- receiving meter data from wmbusmeters
- controlling wmbusmeters meters config (dynamically adding and removing meters)

Thanks for the recommendations, we are going to check the way you recommended:
- feeding data to wmbusmeters through stdin
- getting data out with a shell command

<br/>
To answer your last questions, we are currently targetting x86, amd64 and arm.

About CMake, it may be easier than Makefiles to build a project, as well as being platform-independent and build system independent.
Including and linking external libraries is also more practical with CMake.
It also has a standard way to provide compiler and sysroot environment through it which make it easier to integrate in embedded systems generators (buildroot and yocto for example).

closed 2020-05-27 by weetmuts