wmbusmeters/wmbusmeters #107: Apator162 new offset again ?

bdkacz, 2020-04-22:
Hi. 

Again the readings crazy look at atached log file https://drive.google.com/open?id=1Qx5a1ewlAwX8l6EefkdjNkgduEArmAts

There should be 15,992 m3

And second problem with meter id  01608150   log file  https://drive.google.com/open?id=1u5eubWmYKhMrtd8j_sTm8Oi1Dc4UIFYx

Once the value is correct 21,032, but sometimes  it's going crazy for example 672137.216

Both meters start crazy readings at the same time,so probably the source of problem (another offset?)  is the same for both

weetmuts, 2020-04-23:
Another day, another offset. :-/ I will take look.

bdkacz, 2020-04-23:
If  You need more logs I can upload :)

weetmuts, 2020-04-30:
I made another heuristic. After using the bits to guees,  it will look at the value and if it is greater than the 58 million liters. (aka 16 years of 10m3 consumption) then it will assume it is bad and try the other offsets.

It works for the example you sent with 15,992 m3.

closed 2020-04-30 by bdkacz