wmbusmeters/wmbusmeters #108: Add wmbusmeters to the FreeBSD port tree

ZeWaren, 2020-04-26:
I'm currently submitting wmbusmeters to the FreeBSD port tree, so that people can install it using the package collection system (`pkg install wmbusmeters`):

https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=245925

Notable elements:
- The application is built using the FreeBSD build system, using a BSD makefile (the GNU makefile of the project is not used).
- The test targets can be built and run as well.
- The target can be built in debug mode if needed.
- A RC script is added so that the application can be run as a daemon service.

I'll update the README, referencing this issue, when the port is accepted and live.

weetmuts, 2020-04-26:
Feel free to add the necessary build and daemon start files to a ports/bsd directory.
I will probably make some more changes to the config+daemon setup, 
so it helps if I can see the ports code, and perhaps even adjust it.

ZeWaren, 2020-04-27:
Will do!
When the FreeBSD review team finishes reviewing my work, I'll integrate it here.

weetmuts, 2020-05-26:
Closing this for now. Re-open again when the time comes! :-)

closed 2020-05-26 by weetmuts