wmbusmeters/wmbusmeters #109: No comments allowed in wmbusmeters.conf

alchemyx, 2020-05-06:
It would be great if we could use comments in wmbusmeters.conf:

`loglevel=normal
device=rtlwmbus:868.88M
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
#shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "$METER_JSON"
shell=/usr/bin/curl -XPOST http://127.0.0.1:8086/write?db=wodomierz --data-binary "$METER_NAME m3=$METER_TOTAL_M3"`

and it throws

```
May 06 09:31:59 iot systemd[1]: Starting "wmbusmeters service (no udev trigger)"...
May 06 09:31:59 iot wmbusmetersd[13725]: (wmbusmeters) started /run/wmbusmeters/wmbusmeters.pid
May 06 09:31:59 iot systemd[1]: Started "wmbusmeters service (no udev trigger)".
May 06 09:31:59 iot wmbusmetersd[13727]: No such key: #shell
```

weetmuts, 2020-05-06:
Fix pushed to master. Please test!

alchemyx, 2020-05-06:
Wow. That was quick @weetmuts. It does work, thank you!

```
May 06 09:53:53 iot systemd[1]: Starting "wmbusmeters service (no udev trigger)"...
May 06 09:53:53 iot systemd[1]: Started "wmbusmeters service (no udev trigger)".
```

closed 2020-05-06 by alchemyx