wmbusmeters/wmbusmeters #110: Meter key configuration

Afterster, 2020-05-08:
Ok I know it is not a wmbusmeters issue but I have some difficulties for the hardware setup, hopefully it can help others when reaching this issue later...

I own a iM871A USB stick and a B-Meters RFM-TX1.1 meter. Just got it from a local supplier for my home and both are working fine: usb stick is detected on wmbusmeters and it got the B-Meters messages thanks to your recent addition in April (:+1:). Then it failed because of the encryption key.
My issue is that I don't have a clue on how to setup the meter to disable encryption or customize the aes key. How the hell do you do that?! Documentation is missing from manufacturers side whereas I'm pretty sure it is something that could easily be done with a generic tool somewhere and not a restricted & dedicated manufacturers software solution.

Can you drive me to the right direction? Thanks!

weetmuts, 2020-05-09:
From some documentation, it seems that you configure (commission) the meter using B-Meters own software (Windows only) that sends wmbus telegrams to the meter to set its encryption key etc.

There are parts in the wmbus standard that describes how meters can be configured.
If  B-Meters use these standard messages, then it should be possible to write an
admin tool in wmbusmeters to do that.

merbanan, 2020-05-09:
The B-meters device I have actually send configuration as OOK mbus messages via their configuration usb dongle RFM-RX2. You need that device to configure the meter or you someone gives you the aes-key.

weetmuts, 2020-05-09:
But if you know the protocol for the OOK messages, then it ought to be possible to send them using iM871A as well?

Does OOK mean over the air configuration or something?

merbanan, 2020-05-09:
On Off Keying, instead of Frequency Shift Keying

merbanan, 2020-05-09:
It's a weeks work to replicate and figure out the protocol and building the hardware to transmit things. And you need all the hardware also.

weetmuts, 2020-05-09:
Aha, well if it is not the same on the physical radio level, then its not wireless mbus, and its of course impossible to use wmbus-compatible dongles.

So @Afterster , it seems like you need their dongle and their software. Or you can ask someone (where you bought it?) to configure the meter for you.

Afterster, 2020-05-10:
Ok I see. Thanks for the details both of you. Unfortunate to me as I don't have their dongle/software... Closing the issue here for now, thanks again.

closed 2020-05-10 by Afterster