wmbusmeters/wmbusmeters #116: Wmbusmeters on Raspbian (Raspberry Pi 3B+) stops reading data after a few days

MTrab, 2020-05-15:
For some reason wmbusmeters stops reading data from Multical 21 after a few days of running.
A reboot of the Pi will make it work again.

Any ideas on where to start investigating?
(NOT a power issue, it's fitted with a PoE hat ;) )

weetmuts, 2020-05-15:
Which dongle do you use for receiving data?

MTrab, 2020-05-15:
A AMB8465-M USB stick

weetmuts, 2020-05-15:
Is it reproducible? Do you have dislspace to run with - - debug until it  fails?

weetmuts, 2020-05-15:
I found a problem with the imst dongle, where it could get out of sync, some random bytes got into the serial stream, and then it never managed to recover. This is fixed for imst now.

Perhaps its something similar here. Easy to debug if I have the --debug log.

MTrab, 2020-05-15:
I've setup debug logging - I'll let it run till it dies and then post the logs.

weetmuts, 2020-05-18:
If they are long, then you can email me them at oehrstroem@gmail.com

MTrab, 2020-05-20:
5 days and counting - no breakdowns yet.
Going for a small holiday from tomorrow until sunday, bet ya it have gone broken before I get home :D

HenryNe, 2020-05-24:
I have similar problems. After long running many days (3 or more) the reader goes totaly out of sync. here I have output from verbose.

[amber-broken+11573.txt](https://github.com/weetmuts/wmbusmeters/files/4673462/amber-broken%2B11573.txt)

```
"telegram=||F46BE803FD0C02032102FD0B1111ED3E17ECD2810A6413087A540BC025E012FA715E633E30CD99F8789061D71F3F96E2F8F6825B75DC2B11557AC698EC66847F7EC61DB338F3DBF62C59643612A63C54E8DF4EECB35C9A8D601D19CCA02D2CD2E5F12156CA6EA9DDDCF00B0F61C41AEFAE54F4282F6CEE3A24FC80BBF15333461AAD03931FC84F47561977C78A0EE6A870F1653C1B47F3C4A917244D0312EC10788D3DB8F7BE7B26C8DBA53FAEE9D57148020048E56B4D2627FFA0B0FC5BC7116B4EF29B587F9538CFE1B3FB0FEF6F6963592FE94198F0F913D6D04A961163A05E71E04F19071A7A830040A5F56B1CF3A1502E5B87|+11591"
            ?????? 3 bytes unknown
                  ====================== 11 bytes matches the end of Warm Water meter
 62D59087B9303932703FD0C02032102FD0B1111|+11485"
                                        ?????????????? 7 bytes len ok, but wrong data
                                                      ========  ====== 8 bytes matches the Heat Cost Allocator (53/54 and bytes later are all time different)
                             "teleram=||CE44C5140471316413087A530BC02531EA2B0C7AE79F...3295BC|+11485"
```
Problem starts in telegram with timeoffset +11573.
Telegram +11591 starts with F46BE8, then I see the rest "03FD0C02032102FD0B1111" of Warm Water meter, then 7 bytes unknown, then some bytes 6413087A540BC025 from Heat Cost Allocator, but without the 7 start bytes. This is like there are some bytes are overwiten in the stick, before was send.

Sorry with debug never have catched the first broken telegram. But have seen, that the message "not enough bytes yet, partial frame" stays for longer time. If than comes the next meter, first bytes goes into the rest of telegram before.

MTrab, 2020-05-24:
Fresh home from my holiday, I see that wmbusmeters stopped reporting the day I left (exactly as proclaimed :D )

@weetmuts - will mail you the logfile from when it stopped. Seems like it have gone out of sync.

HenryNe, 2020-05-24:
@MTrab have you more than one meter?
I have read, that the transparent data we receive are simple decoded from the air. means no check for proper wmbus telegram. So, if two meters overlaps and the wireless checksum is ok, then the Amber stick will send us the raw data as they are.

I have a PR#117 with a timeout inside telegram. This should re-sync the start of telegram.

MTrab, 2020-05-24:
I have only one meter, but I can see telegrams from a neighbouring meter as well.

weetmuts, 2020-05-24:
Thanks for the logfile @MTrab ! Yes, it looks like it gets out of sync with the parsing.

Many thanks @HenryNe  for the pull request! Great solution. But what is the word shunk? :-)

It should also be possible to solve this by doing better decoding in the driver, to detect
when the raw dongle data gets out of sync. Something similar is done in checkWMBusFrame. 
However it is slightly more complicated since the amber driver has special commands and
more states.

Even if we have better decoding, the timeout is a good last resort, get back in sync solution, 
and should be added to the other dongles as well.

But lets test it first and then we can see how we can change the code so that
it is reusable.

weetmuts, 2020-05-24:
So, lets merge your pull request now @HenryNe .

HenryNe, 2020-05-24:
shunk -> chunk. it's a part of complete data block. Sorry for wrong spelling. :-(

With a true/false from parser you can may be find the next good telegram. And then? Ho to find the real first byte? You would need to skip first byte and check parser again. If it fails skip the next byte, and so on.

I have not seen a checksum for complete telegram. Exist a way to check the complete telegram without parsing it? Exist may be a checksum or a len byte for the encoded part?

An other way would be to read all bytes with a small timeout, for example 500 ms. If the leng byte is identical with the count of bytes, then we can assume the telegram is good.

HenryNe, 2020-05-24:
> 
> 
> So, lets merge your pull request now @HenryNe .

Ok, you can do it. Please please squash and merge it into one commit.

weetmuts, 2020-05-25:
> You would need to skip first byte and check parser again. If it fails skip the next byte, and so on.

Yes, sort of, the re-syncing code checkWMBusFrame looks for a byte followed by 0x44 where the byte indicates the correct length to the end of the buffer. Re-trying parsing could also be possible. 
The question is if it is worth it, some radio telegrams always risk being lost. I like the clear the buffer on timeout approach. It is quite stable.

>I have not seen a checksum for complete telegram. Exist a way to check the complete telegram without parsing it? Exist may be a checksum or a len byte for the encoded part?

In more advanced telegrams there are checksums and wmbusmeters supports checking them. But the most basic telegrams have no such things. Multical21 does have checksums over the data. However if you sucessfully parse the DifVif and get the values you expect, then you probably got the whole telegram.

HenryNe, 2020-05-25:
rx timeout not helps.
Have catched issue with debug.
[amb8465-t+6115.txt](https://github.com/weetmuts/wmbusmeters/files/4677915/amb8465-t%2B6115.txt)

Problem starts with a 00h as len byte (line 324 in the log):
```
(serial) received binary "00CF"
(amb8465) checkAMB8465Frame "00CF"
(amb8465) received full frame
```
Nothing of the "received binary" after detection of "rx long delay" have the typically start sequence "CE44C5" or "B144C5". I was waiting up to timestamp +16698 (4 hours). In this I found 7 bytes of start from one of my meters "B144C514300250". But not the complete telegram. This telegram does not end with "...0B0111".

After stop and start, the first try gots an error:
```
# wmbusmeters --listento=c1,t1 --format=hr --debug --logtelegrams /dev/wmbus
(wmbusmeters) version: 0.9.30 (+patch rx timeout)
(config) using device: /dev/wmbus
(config) number of meters: 0
(detect) "/dev/wmbus" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serialtty) opened /dev/wmbus
(serial) received binary ""
(amb8465) are you there?
(serial /dev/wmbus) sent "FF0B00F4"
(serial) received binary "AA04F34D7EF221EA"
(serialtty) closed /dev/wmbus
(serialtty) opened /dev/wmbus
(serial) received binary ""
(im871a) are you there?
(serial /dev/wmbus) sent "A5010100"
(serial) received binary ""
(serialtty) closed /dev/wmbus
(serialtty) opened /dev/wmbus
(serial /dev/wmbus) sent "2D0D0A"
(serial) received binary ""
(serialtty) closed /dev/wmbus
No wmbus device found! Exiting!
```
The second restart works again.
I feel, needs to close/open serial port to make working it again.

weetmuts, 2020-05-25:
Ok, added a missing detection of obviously broken telegram data.
Like length 0 (in your example 00CF) was parsed as a valid full frame,
which was clearly wrong. 

Please test with the latest master.

HenryNe, 2020-05-26:
[amb8465-t+65620.txt](https://github.com/weetmuts/wmbusmeters/files/4682786/amb8465-t%2B65620.txt)

Was running ~18 hours without problems.
Then out of sync, and not comes back again after 2 hours.

I can see start bytes B1 44 C5, but not at start of buffer:

```
(serial) received binary "FDC1E4BE22657EB1"
(amb8465) checkAMB8465Frame "FDC1E4BE22657EB1"
(amb8465) protocol error in message received!
(amb8465) protocol error "FDC1E4BE22657EB1"
(serial) received binary "44C5143002508203"
(amb8465) checkAMB8465Frame "44C5143002508203"
(amb8465) protocol error in message received!
(amb8465) protocol error "44C5143002508203"
(serial) received binary "067284364416"
```
I will try to step to next 44, and clear only the bytes -1 before.

HenryNe, 2020-05-26:
I found a myst.
From an other console I opened the serial device and send a command with this small script
```
#!/bin/bash
# usage:  send_hex.sh FF040109F3000000 >/dev/wmbus
T="$1"
while [ ${#T} -ge 2 ]
do
    XX=${T:0:2}
    T=${T:2}
    printf "\x$XX"
done
```
First, the wmbusmeters receiced a correct answer, but bytes are out of sync:
>I send FF040109F3000000
wmbusmeters received binary "01007AFF84"
>I send FF040109F3000000
wmbusmeters received binary "01007AFF84"

The exatly answer would be "FF8401007A".
Then comes some meter data. All bad.

I started to sending FF040109F3000000 again and again (manualy).
After every of this send, the wmbusbeter gots exat 5 Bytes from usb dongle. The bytes are data from meters. I found 44 C5 14:

received binary "0B0111CE44"
received binary "C514047131"

Here is a complete log:
[amber-command-injected.txt](https://github.com/weetmuts/wmbusmeters/files/4683085/amber-command-injected.txt)

HenryNe, 2020-05-26:
With device option "/dev/wmbus:amb8465" can not start wmbusmeters again. It always hangs in the 2 bytes offset:
```
# wmbusmeters --listento=c1,t1 --debug --logtelegrams /dev/wmbus:amb8465
^@
(wmbusmeters) version: 0.9.30
(config) using device: /dev/wmbus
(config) with: amb8465
(config) number of meters: 0
(detect) "/dev/wmbus" "amb8465"
(detect) is_tty=1 is_stdin=0 is_file=0
(amb8465) on /dev/wmbus
(serialtty) opened /dev/wmbus
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(amb8465) set link mode 09
(serial /dev/wmbus) sent "FF040109F3000000"
^@(serial) received binary "1400FFFFFF"
(amb8465) checkAMB8465Frame "1400FFFFFF"
(amb8465) protocol error in message received!
(amb8465) protocol error "1400FFFFFF"
^@^C(serial) stopping manager
(amb8465) get config
(serial /dev/wmbus) sent "FF0A02008077"
(config) listen to link modes: c1,t1
(serial) waiting for stop
^@(serial) event loop stopped!
(serial) closing 1 devices
(serial) no devices working
(serial) no devices working
(serialtty) closed /dev/wmbus
```
Tryed multiple times.

Then I started with device "/dev/wmbus" (without "amb8465"). This helps to syncronisire the telegrams after second start:
[amber-re-sync.txt](https://github.com/weetmuts/wmbusmeters/files/4683289/amber-re-sync.txt)

It's working again now.

HenryNe, 2020-05-27:
PR #123 have testet with serial_rawtty_bad.hex and "stdin:amb8465" as device.

weetmuts, 2020-05-27:
Nice pull request @HenryNe , I am waiting eagerly for confirmation from you and @MTrab whether it solves your problems. :-)

MTrab, 2020-05-27:
Can't compile the new version on Raspbian Buster.

```
g++ -Os -fPIC -std=c++11 -Wall -Werror=format-security -Ibuild src/admin.cc -c -E > build/admin.o.src
src/admin.cc:18:9: fatal error: curses.h: Ingen sådan fil eller filkatalog
 #include<curses.h>
         ^~~~~~~~~~

```

weetmuts, 2020-05-27:
Try "sudo apt install ncurses-dev"
to install the dev dependencies for the admin tool, that is being developed.

MTrab, 2020-07-21:
Why is it ALWAYS when I'm not home it stops working?! :D During my summer-vacation it stopped working, and I can now see the log only showing this:

```
(amb8465) rx long delay (15s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (31s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (31s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (15s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (15s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (15s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (47s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (15s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (15s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (15s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (31s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (31s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (15s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (31s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (47s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (47s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (62s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.
(amb8465) rx long delay (15s), drop incomplete telegram
(amb8465) no sensible telegram found, clearing buffer.

```

HenryNe, 2020-07-25:
Hello MTrap, I have seen such also. The error was detectet, but the stick goes not in sync again. To get data again, a RESET command must send to the stick. This does wmbusmeters at start.
As workarround I have count all the errors and exit wmbusmeter. The outer systemd will restart it automatically. That was the only solution, I found so far.

Here is the patch:
[amb8465-retry-count.patch.txt](https://github.com/weetmuts/wmbusmeters/files/4975848/amb8465-retry-count.patch.txt)

weetmuts, 2020-07-26:
Thanks @HenryNe!

Do you think this is a hardware problem with the dongle?

I will think of how to best introduce this fix into wmbusmeters. It seems that it should
be possible to reset the dongle without exiting wmbusmeters.

weetmuts, 2020-07-26:
> Why is it ALWAYS when I'm not home it stops working?! :D During my summer-vacation it stopped working, and I can now see the log only showing this:

Finagle's law. :-) https://en.wikipedia.org/wiki/Finagle%27s_law

I consider these kind of errors very very bad, I want to create an alarm system
that will try reset the dongle and send an alarm if this happens....
It could also monitor silence and not just incomprehensible byte streams.

HenryNe, 2020-07-26:
> Do you think this is a hardware problem with the dongle?

Yes. The same mBus chip, connected via pure serial wires, without serial-USB works perfectly without such problem.

> I will think of how to best introduce this fix into wmbusmeters. It seems that it should
> be possible to reset the dongle without exiting wmbusmeters.

I was trying to send the startup commands from outside of wmbusmeter to the USB serial port. I have seen a positive answer from chip in wmbusmeter debug, for instance config data. But it not helps to go in sync with dara receiving.

I think, the USB needs to close and open.

weetmuts, 2020-08-01:
Ping @bibo @HenryNe @MTrab 
Ok, I have added alarm functionality to wmbusmeter. The branch is AddAlarm please test!

It can detect protocol errors in amber and force a reset of the dongle after 20 errors. (HenryNe MTrab)
It can detect silence from any dongle and force a reset of the dongle. (bibo)
It will log a warning in the logfile/stdout but also invoke an alarmshell if you have defined such a shell.
This is so that you can send the alarm onto your home automation system or something else.

There is a significant rewrite of code to add new timeout functionality and reset functionality.
Aaaand, I do not have my dongles with me, so there might be bugs!

Add to your config file:
alarmshell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters_alarm -m "$ALARM_TYPE $ALARM_MESSAGE"
alarmtimeout=1h
alarmexpectedactivity=mon-sun(00-23)

weetmuts, 2020-08-01:
My pull request is here if you want to study the changes. https://github.com/weetmuts/wmbusmeters/pull/143

BIBOLV, 2020-08-02:
Great stuff, @weetmuts, tnx! Testing...

weetmuts, 2020-08-25:
Add alarm is now merged into master.

weetmuts, 2020-09-26:
The NewDetectionCode can now be tested. 

It contains working alarms which trigger resets, 
regular resets of dongles (once every 24 hours)
and no udev rules.  All for increasing stability of wmbusmeters.

Please join the testing discussion i nissue #169 

I will close this issue.

closed 2020-09-26 by weetmuts