wmbusmeters/wmbusmeters #119: compilation warnings introduced in 0.9.31

dwrobel, 2020-05-25:
While updating to 0.9.31 in Fedora I noticed the new warnings [1]:

```
src/meter_izar.cc:159:49: warning: '%02d' directive output may be truncated writing between 2 and 3 bytes into a region of size between 1 and 3 [-Wformat-truncation=]
  159 |     snprintf(result, sizeof(result), "%04d-%02d-%02d", h0_year, h0_month, h0_day);
      |                                                 ^~~~
src/meter_izar.cc:159:38: note: directive argument in the range [0, 255]
  159 |     snprintf(result, sizeof(result), "%04d-%02d-%02d", h0_year, h0_month, h0_day);
      |                                      ^~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:872,
                 from /usr/include/c++/10/cstdio:42,
                 from /usr/include/c++/10/ext/string_conversions.h:43,
                 from /usr/include/c++/10/bits/basic_string.h:6535,
                 from /usr/include/c++/10/string:55,
                 from src/util.h:23,
                 from src/meters.h:21,
                 from src/meter_izar.cc:18:
/usr/include/bits/stdio2.h:70:35: note: '__builtin___snprintf_chk' output between 11 and 14 bytes into a destination of size 11
   70 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   71 |        __bos (__s), __fmt, __va_arg_pack ());
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/meter_izar.cc: In member function 'std::string MeterIzar::setH0Date()':
src/meter_izar.cc:159:49: warning: '%02d' directive output may be truncated writing between 2 and 3 bytes into a region of size between 1 and 3 [-Wformat-truncation=]
  159 |     snprintf(result, sizeof(result), "%04d-%02d-%02d", h0_year, h0_month, h0_day);
      |                                                 ^~~~
src/meter_izar.cc:159:38: note: directive argument in the range [0, 255]
  159 |     snprintf(result, sizeof(result), "%04d-%02d-%02d", h0_year, h0_month, h0_day);
      |                                      ^~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:872,
                 from /usr/include/c++/10/cstdio:42,
                 from /usr/include/c++/10/ext/string_conversions.h:43,
                 from /usr/include/c++/10/bits/basic_string.h:6535,
                 from /usr/include/c++/10/string:55,
                 from src/util.h:23,
                 from src/meters.h:21,
                 from src/meter_izar.cc:18:
/usr/include/bits/stdio2.h:70:35: note: '__builtin___snprintf_chk' output between 11 and 14 bytes into a destination of size 11
   70 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   71 |        __bos (__s), __fmt, __va_arg_pack ());
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

[1] https://kojipkgs.fedoraproject.org//work/tasks/4451/44964451/build.log

weetmuts, 2020-05-26:
My compiler (g++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0) does not show this warning, so I have temporarily disabled it. Google search results indicate that fixing the warning seems a bit shaky,
even though it should be sort of easy.... %12 for the months, %31 for the days and %9999 for the year.

weetmuts, 2020-05-27:
Fixed.

closed 2020-05-27 by weetmuts