wmbusmeters/wmbusmeters #125: Recommended USB stick

JohnAnderton, 2020-05-30:
What is the most recommended USB stick to be used with this project_

weetmuts, 2020-05-31:
I only have personal experience of the im871a,amb8465 and an rtlsdr compatible dongle.
The commercial dongles (im871a,amb8464) receive well despite having tiny antennas inside the dongle. However the reception range is limited by walls and you must definitely get quite close to the meter if it is mounted underground in a concrete tube.

The rtlsdr/rtl-wmbus solution seems to work for a lot of users, but it does use more cpu-power since it decodes the radio traffic in software. Range seems to be similar to the other dongles, despite
the antenna being much larger.

At least one professional collector use the same dongles, but the versions with an external antenna connector, into which they attach a radio amplifier for the proper frequency, and then a larger antennna. This makes it possible to receive telegrams from meters underground and over larger distances.

weetmuts, 2020-06-30:
Added this answer to the README.

closed 2020-06-30 by weetmuts