wmbusmeters/wmbusmeters #127: SDR rtl2832 is not recognized

x00e, 2020-06-12:
trying to use SDR rtl2832 for reading watermeters around house. 

1 - cloned git repo
2 - make; make test; make install 
3 - rtl_wmbus is installed 

# ls /usr/bin/rtl*
`/usr/bin/rtl_adsb  /usr/bin/rtl_eeprom  /usr/bin/rtl_fm  /usr/bin/rtl_power  /usr/bin/rtl_sdr  /usr/bin/rtl_tcp  /usr/bin/rtl_test  /usr/bin/rtlwmbus  /usr/bin/rtl_wmbus`

# dmesg:
`[ 6074.030416] usb 1-5.2: dvb_usb_v2: rc.query() failed=-71
[ 6074.104919] usb 1-5.2: USB disconnect, device number 8
[ 6074.147516] r820t 10-001a: destroying instance
[ 6074.148050] dvb_usb_v2: 'Realtek RTL2832U reference design:1-5.2' successfully deinitialized and disconnected
[ 6077.274193] usb 1-5.2: new high-speed USB device number 9 using xhci_hcd
[ 6077.388089] usb 1-5.2: New USB device found, idVendor=0bda, idProduct=2838
[ 6077.388106] usb 1-5.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6077.388116] usb 1-5.2: Product: RTL2838UHIDIR
[ 6077.388125] usb 1-5.2: Manufacturer: Realtek
[ 6077.388133] usb 1-5.2: SerialNumber: 00000001
[ 6077.400286] usb 1-5.2: dvb_usb_v2: found a 'Realtek RTL2832U reference design' in warm state
[ 6077.464408] usb 1-5.2: dvb_usb_v2: will pass the complete MPEG2 transport stream to the software demuxer
[ 6077.464459] dvbdev: DVB: registering new adapter (Realtek RTL2832U reference design)
[ 6077.478819] i2c i2c-9: Added multiplexed i2c bus 10
[ 6077.478832] rtl2832 9-0010: Realtek RTL2832 successfully attached
[ 6077.479010] usb 1-5.2: DVB: registering adapter 0 frontend 0 (Realtek RTL2832 (DVB-T))...
[ 6077.479475] r820t 10-001a: creating new instance
[ 6077.487878] r820t 10-001a: Rafael Micro r820t successfully identified
[ 6077.495699] rtl2832_sdr rtl2832_sdr.1.auto: Registered as swradio0
[ 6077.495705] rtl2832_sdr rtl2832_sdr.1.auto: Realtek RTL2832 SDR attached
[ 6077.495709] rtl2832_sdr rtl2832_sdr.1.auto: SDR API is still slightly experimental and functionality changes may follow
[ 6077.506301] Registered IR keymap rc-empty
[ 6077.506521] rc rc0: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.2/rc/rc0
[ 6077.506707] input: Realtek RTL2832U reference design as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5.2/rc/rc0/input21
[ 6077.507832] lirc lirc0: lirc_dev: driver ir-lirc-codec (dvb_usb_rtl28xxu) registered at minor = 0
[ 6077.507848] usb 1-5.2: dvb_usb_v2: schedule remote query interval to 200 msecs
[ 6077.516967] usb 1-5.2: dvb_usb_v2: 'Realtek RTL2832U reference design' successfully initialized and connected`

# lsusb
`Bus 001 Device 009: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T`

# ls /dev/rtl*
`/dev/rtlsdr_2`

# sudo wmbusmeters --debug --t1 /dev/rtlsdr_2:rtlwmbus
`(wmbusmeters) version: 0.9.31-31-g7a55d21
(config) using device: /dev/rtlsdr_2
(config) with: rtlwmbus
(config) number of meters: 0
(detect) "/dev/rtlsdr_2" "rtlwmbus"
(detect) is_tty=1 is_stdin=0 is_file=0
(rtlwmbus) using command: rtlwmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtlwmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop
(bgshell) 9514 exited with return code 2
(serial) closing non working fd=3
(serial) no devices working
(serial) stopping manager
(serialcmd) closed /bin/sh
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working
(serialcmd) closed /bin/sh`

# # sudo wmbusmeters --debug
`(wmbusmeters) version: 0.9.31-31-g7a55d21
(config) using device: auto
(config) number of meters: 0
(detect) "auto" ""
(detect) driver: auto
(im871a) exists? /dev/im871a
(im871a) exists? /dev/im871a_0
(im871a) exists? /dev/im871a_1
(im871a) exists? /dev/im871a_2
(im871a) exists? /dev/im871a_3
(im871a) exists? /dev/im871a_4
(im871a) exists? /dev/im871a_5
(im871a) exists? /dev/im871a_6
(im871a) exists? /dev/im871a_7
(im871a) exists? /dev/im871a_8
(amb8465) exists? /dev/amb8465
(amb8465) exists? /dev/amb8465_0
(amb8465) exists? /dev/amb8465_1
(amb8465) exists? /dev/amb8465_2
(amb8465) exists? /dev/amb8465_3
(amb8465) exists? /dev/amb8465_4
(amb8465) exists? /dev/amb8465_5
(amb8465) exists? /dev/amb8465_6
(amb8465) exists? /dev/amb8465_7
(amb8465) exists? /dev/amb8465_8
(rfmrx2) exists? /dev/rfmrx2
(rfmrx2) exists? /dev/rfmrx2_0
(rfmrx2) exists? /dev/rfmrx2_1
(rfmrx2) exists? /dev/rfmrx2_2
(rfmrx2) exists? /dev/rfmrx2_3
(rfmrx2) exists? /dev/rfmrx2_4
(rfmrx2) exists? /dev/rfmrx2_5
(rfmrx2) exists? /dev/rfmrx2_6
(rfmrx2) exists? /dev/rfmrx2_7
(rfmrx2) exists? /dev/rfmrx2_8
(cul) exists? /dev/ttyUSB0
(cul) exists? /dev/ttyUSB0_0
(cul) exists? /dev/ttyUSB0_1
(cul) exists? /dev/ttyUSB0_2
(cul) exists? /dev/ttyUSB0_3
(cul) exists? /dev/ttyUSB0_4
(cul) exists? /dev/ttyUSB0_5
(cul) exists? /dev/ttyUSB0_6
(cul) exists? /dev/ttyUSB0_7
(cul) exists? /dev/ttyUSB0_8
(rtlsdr) exists? /dev/rtlsdr
(rtlsdr) exists? /dev/rtlsdr_0
(rtlsdr) exists? /dev/rtlsdr_1
(rtlsdr) exists? /dev/rtlsdr_2
You are not in the same group as the device /dev/rtlsdr_2
(serial) stopping manager`

# uname -a
`Linux box 4.15.0-101-generic #102-Ubuntu SMP Mon May 11 10:07:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux`

# lsb_release -a
`No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.4 LTS
Release:        18.04
Codename:       bionic`

x00e, 2020-06-12:
running
~# rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus   gets me:
`T1;0;0;2020-06-12 13:30:41.000;51;61;822306FF;0x1e442423ff06238214637abd00101862b975979be6ae1cda272fc8829d5fb5
T1;0;0;2020-06-12 13:30:45.000;54;138;7618078B;0x1e4424238b071876ff637aff0010d621cf2c4b667becc6cd3e6798d809ff0a
T1;0;0;2020-06-12 13:30:55.000;54;100;763307FF;0x1effffffff07337614637adb00103007ce9b3a60439d1a6bbc9f47e07368bb`

weetmuts, 2020-06-14:
Try this command instead: sudo wmbusmeters --debug --t1 rtlwmbus

weetmuts, 2020-06-14:
You can also tailor your command like this:
 sudo wmbusmeters --debug --t1 'rtlwmbus:/usr/bin/rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus'

x00e, 2020-06-15:
~# sudo wmbusmeters --debug --t1 'rtlwmbus:/usr/bin/rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus'   <-- this one seems to work, but i get o LOT of " telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed" errors. 

bellow is some output from running wmbusmeters for a couple of minutes. hope it helps in any way. 
PS: water meter(s) are IZAR RS 868 

`(wmbusmeters) version: 0.9.31-31-g7a55d21
(config) using device: rtlwmbus
(config) with: /usr/bin/rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(config) number of meters: 0
(detect) "rtlwmbus" "/usr/bin/rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(detect) driver: rtlwmbus
(rtlwmbus) using command: /usr/bin/rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii "T1;1;1;2020-06-15 07:49:49.000;57;132;8223068B;0x1e4424238b06238214637a8a00101862865882a39a89c66718d8ccd1662644<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-06-15 07:49:49.000;57;132;8223068B;0x1e4424238b06238214637a8a00101862865882a39a89c66718d8ccd1662644<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(dvparser) found new format "6286589A89C66718D16626" with hash d626, remembering!
(telegram) DLL L=1e C=44 (from meter SND_NR) M=2324 (HYD) A=8223068b VER=14 TYPE=63 (Unknown)
(telegram) TPL CI=7a ACC=8a STS=00 CFG=1810 (SPECIFIC_16_31)
Received telegram from: 8223068b
          manufacturer: (HYD) Hydrometer
           device type: Unknown
(wmbus) 00: 1e length (30 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 2423 dll-mfct (HYD)
(wmbus) 04: 8b062382 dll-id (8223068b)
(wmbus) 08: 14 dll-version
(wmbus) 09: 63 dll-type (Unknown)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: 8a tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 1018 tpl-cfg 1810 (SPECIFIC_16_31 )
(wmbus) 0f: 62 dif (16 Bit Integer/Binary Minimum value storagenr=1)
(wmbus) 10: 86 vif (Energy kWh)
(wmbus) 11: 58 vife (duration of limit exceed first upper  is 0)
(wmbus) 12: 82A3
(wmbus) 14: 9A dif (4 digit BCD Maximum value)
(wmbus) 15: 89 dife (subunit=0 tariff=0 storagenr=18)
(wmbus) 16: C6 dife (subunit=2 tariff=0 storagenr=210)
(wmbus) 17: 67 dife (subunit=6 tariff=32 storagenr=3794)
(wmbus) 18: 18 vif (Mass g)
(wmbus) 19: D8CC
(wmbus) 1b: D1 dif (8 Bit Integer/Binary Maximum value storagenr=1)
(wmbus) 1c: 66 dife (subunit=1 tariff=2 storagenr=13)
(wmbus) 1d: 26 vif (Operating time hours)
(wmbus) 1e: 44
(wmbus) "telegram=||1E4424238B06238214637A8A00101862865882A39A89C66718D8CCD1662644|+8"
(serial) received ascii "T1;0;0;2020-06-15 07:49:52.000;29;60;FFFFFFFF;0x1e4424ffffffffffffffffff33ffffffffff10ffffffffffffffffffffffff<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-06-15 07:49:52.000;29;60;FFFFFFFF;0x1e4424ffffffffffffffffff33ffffffffff10ffffffffffffffffffffffff<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2020-06-15 07:49:53.000;20;22;FF088CFF;0x1e4424ffff8c08ffffffffff88ffacac88ffffffffffffffffffeeffffffc8<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-06-15 07:49:53.000;20;22;FF088CFF;0x1e4424ffff8c08ffffffffff88ffacac88ffffffffffffffffffeeffffffc8<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2020-06-15 07:49:58.000;15;15;FFFFFF8B;0x1effff238bffffffff637a2200ffff2ea447fffff150d607ffffff17ffda9a<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-06-15 07:49:58.000;15;15;FFFFFF8B;0x1effff238bffffffff637a2200ffff2ea447fffff150d607ffffff17ffda9a<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
[... 630 lines cut ...]
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2020-06-15 07:54:52.000;29;27;7242078B;0x1e4424ff8b074272ff637a170010f8639e9973167c95c3cc0132ffeebdeb2c<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-06-15 07:54:52.000;29;27;7242078B;0x1e4424ff8b074272ff637a170010f8639e9973167c95c3cc0132ffeebdeb2c<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2020-06-15 07:54:53.000;57;128;FF54078B;0x1effffff8b0754ff14637a9eff10ff737849ffff0cff87ffffff7ea8e5ff44<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-06-15 07:54:53.000;57;128;FF54078B;0x1effffff8b0754ff14637a9eff10ff737849ffff0cff87ffffff7ea8e5ff44<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2020-06-15 07:54:54.000;50;129;782506FF;0x1e4424ffff06257814637ab300101b654ea610558bf264d0068d5d8d2f028c<0A>"
[... cut ...]
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-06-15 07:54:56.000;31;24;FFFFFFFF;0x1e44ffffffffffffffffffffffffffffff0062077167ffffffffffffffffff<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) stopping manager
(serial) event loop stopped!
(serial) closing 1 devices
(serial) no devices working
(bgshell) 26899 terminated due to signal 1
(serial) no devices working
(serialcmd) closed /bin/sh
`

weetmuts, 2020-06-15:
The output from rtl_wmbus must start with T1;1;1;.....
if it says T1;0;0;... then the checksums of the received data did not work out,
ie the radio message was garbled. Try being closer to the meter, but not too close. :-)

So wmbusmeters will ignore and warn for T1;0;0 messages.

However you have received a few T1;1;1 messages, when I decode these,
it looks like the telegrams are not OMS-compliant. 

What is the exact type version and manufacturer of this meter that you are receiving?
Can you post a photo?

x00e, 2020-06-15:
i've uploaded a picture of the water meter [here](https://imgur.com/JNikbHM)

i think the water meters is a [Diehl AQUARIUS RS ](https://www.etserv.ro/userfiles/file/fise-tehnice/AQUARIUS%20S-RS-P.pdf) (sorry, specs not in English). 

but definitely the radio module is IZAR RS in 868 MHz

weetmuts, 2020-06-15:
Ah, I tried the wmbusmeters izar driver on your telegrams, but they failed to decode using the default key. Either there is a non-default key on this device, or it has a different protocol altogether.

I am sorry that I will not have time to decode this meter, my focus is on OMS-compliant wmbus meters, which is a big enough task as it is. :-)

There are two people know more about the Izar than I do. (I know nothing!)
@skazi0 who write the driver and @ZeWaren who improved it, maybe they can help?

telegram=|1e4424238b06228214637a5f00103b6302c9bfbf55dfe0524a3e9b0f29ad95|
telegram=|1e4424238b06238214637a8a00101862865882a39a89c66718d8ccd1662644|

ZeWaren, 2020-06-16:
When I was looking at the bytecode of a DIEHL data receiver to understand the logic I saw that there are multiple protocols and payload formats, depending on the various devices.

If you have the same payload format as my device, you can brute force the key (it's only 32 bit).
This is what I did. I took a bunch of frames and I tried to decode them with every key possible until all decoded frames ended up being good data. The code is here: https://github.com/ZeWaren/izar-prios-smart-meter-collector/blob/master/PC/prios_key_cracker.c

On the other hand, if you have a device that uses a different protocol (or most likely a variation of it) than the one which is implement here you'll first need to reverse-engineer it from your frames, or read the byte-code of a receiver system (this is what I did).

iwikus, 2020-09-04:
According to picture of water meter we have same model (same part number - 3004612). I had opened https://github.com/weetmuts/wmbusmeters/issues/155 and now I have find out it is same meter like in this issue. But I have no CRC errors, just unexpected end of data logged.

weetmuts, 2020-09-04:
I added a stub src/meter_izar3.cc and the meter driver izar3.

It cannot decode anything but the driver METER_DETECTION list
knows about one mfct,media,version combo that identifies the izar3.
This will be helpful for new users.

Perhaps there are more combos.

Ok, now lets try to decode it. :)

don-vip, 2021-02-07:
They use one of the two PRIOS default keys. Using the tool I created in https://github.com/weetmuts/wmbusmeters/issues/226#issuecomment-774565756, here are the decoded frames:

> telegram=|1e4424238b06228214637a5f00103b6302c9bfbf55dfe0524a3e9b0f29ad95|

```java
    assertEquals("ReadingData={"
        + "  rawMessage={"
        + "    frame=[1E,44,24,23,8B,06,22,82,14,63,7A,5F,00,10,3B,63,02,C9,BF,BF,55,DF,E0,52,4A,3E,9B,0F,29,AD,95], timestamp=1612657129739.000, moduleId=0"
        + "  },"
        + "  frameDescription={ type=FrameDescMBusRadio,"
        + "    meterId=UHYD8B0663148222000,"
        + "    meterName=IZAR_RS_REALDATA,"
        + "    telegramType=REAL_DATA,"
        + "    withAlarm=false,"
        + "    withError=false,"
        + "    meterDeviceNumber=63148222,"
        + "    mediumByte=0x06,"
        + "    version=139,"
        + "    manufacturer=HYD,"
        + "    errorFlagBytes=0x00,"
        + "    telegramContentType=STANDARD,"
        + "    installFrame=false,"
        + "    synchronizeFrame=false,"
        + "    lField=0x1e,"
        + "    cField=0x44,"
        + "    mField=[24,23],"
        + "    aFieldSecondaryAddress=[8B,06,22,82,14,63],"
        + "    ciField=0x7a,"
        + "    encMode=LFSR },"
        + "  telegramFormatError=false,"
        + "  parsingLevel=FULL_CIPHERED,"
        + "  measurements={"
        + "    UHYD8B0663148222000 -> {"
        + "       0 -> {"
        + "        values=["
        + "          { semantic=volume,  type=QUANTITY,  isValid=true,  value=47.221㎥ }"
        + "        ]"
        + "      },"
        + "      1 -> {"
        + "Date= 2020.5.31,"
        + "        values=["
        + "          { semantic=date,  type=DATE,  isValid=true,  value=2020.5.31 },"
        + "          { semantic=volume,  type=QUANTITY,  isValid=true,  value=46.457㎥ }"
        + "        ]"
        + "      }"
        + "    }"
        + "  }"
        + "}",
    new InterpretMBusRadioImpl().interpretFull(new RawMessage(
        HexString.getByteArray("1e4424238b06228214637a5f00103b6302c9bfbf55dfe0524a3e9b0f29ad95"),
        1612657129739L)).toStringHumanReadable().replace("\r", "").replace("\n", ""));
```

> telegram=|1e4424238b06238214637a8a00101862865882a39a89c66718d8ccd1662644|

```java
    assertEquals("ReadingData={"
        + "  rawMessage={"
        + "    frame=[1E,44,24,23,8B,06,23,82,14,63,7A,8A,00,10,18,62,86,58,82,A3,9A,89,C6,67,18,D8,CC,D1,66,26,44], timestamp=1612657129739.000, moduleId=0"
        + "  },"
        + "  frameDescription={ type=FrameDescMBusRadio,"
        + "    meterId=UHYD8B0663148223000,"
        + "    meterName=IZAR_RS_REALDATA,"
        + "    telegramType=REAL_DATA,"
        + "    withAlarm=false,"
        + "    withError=false,"
        + "    meterDeviceNumber=63148223,"
        + "    mediumByte=0x06,"
        + "    version=139,"
        + "    manufacturer=HYD,"
        + "    errorFlagBytes=0x00,"
        + "    telegramContentType=STANDARD,"
        + "    installFrame=false,"
        + "    synchronizeFrame=false,"
        + "    lField=0x1e,"
        + "    cField=0x44,"
        + "    mField=[24,23],"
        + "    aFieldSecondaryAddress=[8B,06,23,82,14,63],"
        + "    ciField=0x7a,"
        + "    encMode=LFSR },"
        + "  telegramFormatError=false,"
        + "  parsingLevel=FULL_CIPHERED,"
        + "  measurements={"
        + "    UHYD8B0663148223000 -> {"
        + "       0 -> {"
        + "        values=["
        + "          { semantic=volume,  type=QUANTITY,  isValid=true,  value=6.977㎥ }"
        + "        ]"
        + "      },"
        + "      1 -> {"
        + "Date= 2020.5.31,"
        + "        values=["
        + "          { semantic=date,  type=DATE,  isValid=true,  value=2020.5.31 },"
        + "          { semantic=volume,  type=QUANTITY,  isValid=true,  value=6.850㎥ }"
        + "        ]"
        + "      }"
        + "    }"
        + "  }"
        + "}",
    new InterpretMBusRadioImpl().interpretFull(new RawMessage(
        HexString.getByteArray("1e4424238b06238214637a8a00101862865882a39a89c66718d8ccd1662644"),
        1612657129739L)).toStringHumanReadable().replace("\r", "").replace("\n", ""));
```

I think the PR I will write for #226 will fix this one too.

weetmuts, 2021-08-08:
@x00e wmbusmeters can now decode this meter! Yay!
https://github.com/weetmuts/wmbusmeters/issues/284

If you have not already given up @x00e then please test. :-)

closed 2021-08-08 by weetmuts

x00e, 2021-08-08:
@weetmuts  I can confirm that it works:

```
 ~# sudo wmbusmeters --logtelegrams --format=json rtlwmbus:868.9M test_hot auto 63148228 NOKEY
Started config rtlwmbus[00000001] listening on none using fq 868.9M
^Ttelegram=|1E4424238B06288214637A16001013|0413A2340100426CBF274413F7320100|+63
{"media":"warm water","meter":"hydrus","name":"test_hot","id":"63148228","total_m3":79.01,"total_tariff1_m3":0,"total_tariff2_m3":0,"max_flow_m3h":0,"flow_temperature_c":127,"external_temperature_c":0,"current_date":"","total_at_date_m3":78.583,"total_tariff1_at_date_m3":0,"total_tariff2_at_date_m3":0,"at_date":"2021-07-31 00:00","actuality_duration_s":0,"operating_time_h":0,"remaining_battery_life_y":0,"status":"OK","timestamp":"2021-08-08T19:53:01Z","device":"rtlwmbus[00000001]","rssi_dbm":124}
```
thank you and @mchouque for figuring this out  :thumbsup: