wmbusmeters/wmbusmeters #128: Default fields for logfile saving

tatulea, 2020-06-17:
Hi,

I would like to know what are the default fields when I save the telegrams in a logfile.

I have the following output in my logfile:

```
T1;1;1;2020-05-28 22:22:56.000;26;27;93932258;0x294468505822939376f0a0009f27b6280028b62800000e000131006ba1007cb2008dc3009ed4000fe500
```

and I an not sure what each column means.

weetmuts, 2020-06-18:
Hmm, that is the output from rtl_wmbus, not the output from wmbusmeters.

But I can give an answer anyway. :-)

T1;1;1 means that a T1 telegram was successfully received! T1;0;0 means it was received
but the checksums did not match up, ie it was a garbled radio message. wmbusmeters will ignore such telegrams.

The its the timestamp when the computer received the telegram. 
Then there are two numbers 26 and 27, which means packet rssi, and just rssi.
I am not sure exactly the difference.
Then it is the id stored in the telegram.
Lastly it is the raw bytes of the telegram, however with the DLL (data link layer) crc bytes removed.

I checked the telegram and it is from a techem meter.

tatulea, 2020-06-18:
Thanks for your answer! Is there the possibility to have T1;1;0 or T1;0;1? Is there a meaning for the first 1/0 and the second 1/0?

weetmuts, 2020-06-18:
I am not sure but I think that T1;1;0 is also an ok telegram, but the radio signal was garbled
in some way, but it was possible to reconstruct it. 

As for the last 1, see the source, line 660:
https://github.com/xaelsouth/rtl-wmbus/blob/master/t1_c1_packet_decoder.h

Then look for why/when t1_c1_packet_decoder_work.err_3outo is set.
:-)

tatulea, 2020-06-18:
Got it, thanks for your help! I think it comes from "3 out of 6" which should be related to the physical layer. I will look more into it.

tatulea, 2020-06-18:
I have some packets that start with C1;0;1 and wmbusmeters exits when it gets to decide it. The 3 out of 6 is an encoding used just for T, not for C mode. How does wmbusmeters handle the case of C1;0;1?

This is an example of such a telegram:

```
C1;0;1;2020-06-04 10:16:50.000;19;18;03435048;0x374468504850430394087a7a0000000e12e3fba0badcb2098250ccc715697bbf6d37f4cae40529c63b3835d4c7a932622f9ab4b7c62d8274
```

weetmuts, 2020-06-18:
wmbusmeters will ignore those telegrams. it should log a warning in the log if it sees them. at least if you run with --debug.

weetmuts, 2020-06-18:
You must use the latest rtl_wmbus, since it had problems decoding C1 telegrams before, but it is fixed in the most recent version.

closed 2020-06-23 by tatulea