wmbusmeters/wmbusmeters #129: Techem Compact V heat meter

IzeQube, 2020-06-24:
At my home i have a Techem Compact V heat meter. The funny thing is, that I can get decoded values using vario451, but the values are totally wrong.

I also get the warning: (vario451) unexpected meter version 0x45 !

Is there any way to get my heat meter working?

weetmuts, 2020-06-24:
If you run with --debug and the meter name vario451 and then email me the log, then I can take a look.

weetmuts, 2020-06-30:
Ping @IzeQube ?

IzeQube, 2020-07-02:
Sorry for the late reply. After starting it with --debug I could see that the right values were being received, but are then converted what isn't needed for Compact V meter.

I then did a rather hacky fix with a new meter that reads the right values.

I can give you the details when I'm back home later or I can also post a pull request if you want...

weetmuts, 2020-07-02:
Cool! Either way works!

IzeQube, 2020-07-26:
I added a PR, took a while because my dev machine died

weetmuts, 2020-07-30:
Hi @IzeQube ! As you can see I am ready to merge your pull request, but we need an example
telegram for the regression tests. You can email me a telegram if you like and I will anonymize it and add it to the test suite.

weetmuts, 2020-09-05:
Closing this for now. Thanks for contributing the code @IzeQube !

closed 2020-09-05 by weetmuts