wmbusmeters/wmbusmeters #130: How to use IZAR in 433 Mhz

rudsomlima, 2020-06-28:
My DIEHL IZAR water meter is 434 Mhz, how to use? I need to locate the meters first to find mine. What is available here is only 868 Mhz. Thanks.

weetmuts, 2020-06-28:
@merbanan can we receive 433 MHz wmbus using your tool?

merbanan, 2020-06-28:
Yes we can.

merbanan, 2020-06-28:
@rudsomlima get a rtl_sdr device and run rtl_433 -A to see if you can get any signals.

merbanan, 2020-06-28:
But I am doubting that it is transmitting on 433 MHz.

merbanan, 2020-06-28:
https://www.diehl.com/cms/files/DIEHL_IZAR_system_poster_EN_2019.pdf?download=1

"Radio 868/434 MHz" ok, so they use 434MHz. Anyway I need sample signals but I see no issue with adding support.

rudsomlima, 2020-06-28:
> @rudsomlima get a rtl_sdr device and run rtl_433 -A to see if you can get any signals.

rtl_433 -R 104 -s 1200k -f 434470k 
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 154 device decoding protocols [ 104 ]
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1200000 S/s.
Tuner gain set to Auto.
Tuned to 434.470MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:08:54
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206308
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304c0863ba008800a2610d0013512468ff5915ba2d1e184b              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:08:55
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206325
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304c2563ba008800a2610d0013a372fedee01cea6a3fb566              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:08:59
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206320
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304c2063ba008800a2610d0013e86eae7360611ba789566f              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:08:59
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206330
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304c3063ba008800a2510d0013e8c13380d47fd4183da671              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:09:03
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206308
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304c0863ba008800a2710d0013513347b6f23469f4c50149              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:09:04
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206325
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304c2563ba008800a2710d0013a365d1974b3d39b3e4ac64              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:09:07
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206330
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304c3063ba008800a2610d0013e8f8425a291da072508d76              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:09:08
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206320
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304c2063ba008800a2710d0013e879813acb40c87e524f6d              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:09:11
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206308
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-06-28 15:10:40
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206330
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304c3063ba008800a2110d0013e89d8ea678f89b7f51c27b              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000

merbanan, 2020-06-28:
Can you run -R 104 -s 1200k -f 434470k -S known and upload a few samples?

merbanan, 2020-06-28:
I don't see any obvious pattern so I think we need some values from your meter to correlate the readings against.

merbanan, 2020-06-28:
rtl-wmbus should be able to handle this also if you change the frequency.

rudsomlima, 2020-06-28:
> -R 104 -s 1200k -f 434470k -S

[https://drive.google.com/drive/folders/1JHMTQ_7oBoS0uhAldY4okWCOA8dLW3FG?usp=sharing](url)

merbanan, 2020-06-28:
For the device with ID 1206330 we get this:

```
1944304c3063ba008800a2 5 10d0013e8 c13380d47fd4183da6 7 1
1944304c3063ba008800a2 6 10d0013e8 f8425a291da072508d 7 6
1944304c3063ba008800a2 7 10d0013e8 ef6d13823c73ab8b94 7 4
1944304c3063ba008800a2 0 10d0013e8 8aa1efd3d948a68adb 7 9
1944304c3063ba008800a2 1 10d0013e8 9d8ea678f89b7f51c2 7 b
1944304c3063ba008800a2 2 10d0013e8 a4ff7c859aef153ce9 7 c
1944304c3063ba008800a2 3 10d0013e8 b3d0352ebb3ccce7f0 7 e
1944304c3063ba008800a2 4 10d0013e8 d61cc97f5e07c1e6bf 7 3
1944304c3063ba008800a2 5 10d0013e8 c13380d47fd4183da6 7 1
1944304c3063ba008800a2 6 10d0013e8 f8425a291da072508d 7 6
1944304c3063ba008800a2 7 10d0013e8 ef6d13823c73ab8b94 7 4
1944304c3063ba008800a2 0 10d0013e8 8aa1efd3d948a68adb 7 9
1944304c3063ba008800a2 1 10d0013e8 9d8ea678f89b7f51c2 7 b
```

So there is a 3 bit(?) counter. And the data after 10d0013e8 depends on the counter (but most likely more data). So the 10d0013e8 data is the most interesting stuff because if there is a clear text payload the bits here look suspect.

rudsomlima, 2020-06-28:
I'm not really an expert. So I need help

merbanan, 2020-06-28:
Try to figure out what id you meter has. If you have issues then remove the antenna of the rtl-sdr device and place it close to the meter. You should only get your meter then. Now with the id take pictures of the meter before and after a shower and correlate with the payloads. If we are lucky we can locate the bits for the water volume.

merbanan, 2020-06-28:
@rudsomlima I have now promoted you to an expert. Everything is a puzzle and you need to figure out one piece at a time. When I have figured out some proprietary protocols in rtl_433 I have recorded several thousands of messages to get a complete picture. It shouldn't be needed here but the next thing to do now is to figure out what meter id is yours. Then we can start tracking what happens when you consume water and correlate that to the meter readings.

rudsomlima, 2020-06-28:
Is there a way to measure the RSS of the meters? I identify which ones are closest

rudsomlima, 2020-06-28:
I found this article: [https://www.zewaren.net/wmbus-izar-meter.html](url)

merbanan, 2020-06-28:
Looks like it is already supported \o/

merbanan, 2020-06-28:
Just run rtl-wmbus with another frequency.

rudsomlima, 2020-06-29:
> Just run rtl-wmbus with another frequency.

Would it be this?

root@raspberrypi:/home/pi# wmbusmeters --debug --t1 'rtlwmbus:/usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus'
(wmbusmeters) version: _
(config) using device: rtlwmbus
(config) with: /usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(config) number of meters: 0
(detect) "rtlwmbus" "/usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(detect) driver: rtlwmbus
(rtlwmbus) using command: /usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop
(serial) received ascii "/bin/sh: 1: "
(rtlwmbus) checkRTLWMBusFrame "/bin/sh: 1: "
(rtlwmbus) no eol found, partial frame
(serial) received ascii "/usr/bin/rtl_wmbus: not found"
(rtlwmbus) checkRTLWMBusFrame "/bin/sh: 1: /usr/bin/rtl_wmbus: not found"
(rtlwmbus) no eol found, partial frame
(serial) received ascii "<0A>"
(rtlwmbus) checkRTLWMBusFrame "/bin/sh: 1: /usr/bin/rtl_wmbus: not found<0A>"
(rtlwmbus) only text
(bgshell) 2885 exited with return code 127
(serial) closing non working fd=3
(serial) no devices working
(serial) stopping manager
(serialcmd) closed /bin/sh
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working
(serialcmd) closed /bin/sh

merbanan, 2020-06-29:
/usr/bin/rtl_wmbus: not found

You need to install it first but try running rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus manually first so that you know it is working properly.

rudsomlima, 2020-07-01:
I'm having trouble installing rtl_wmbus (debian raspberry)

root@raspberrypi:/home/pi/Downloads/rtl-wmbus-master# make debug
gcc -DDEBUG  -O0 -g3 -ggdb -p -pg -Iinclude -std=gnu99 -Wall -W -Waggregate-return -Wbad-function-cast -Wcast-align -Wcast-qual -Wchar-subscripts -Wcomment -Wfloat-equal -Winline -Wmain -Wmissing-noreturn -Wmissing-prototypes -Wparentheses -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-prototypes -Wswitch -Wunreachable-code -Wunused -Wuninitialized -o "build/rtl_wmbus" rtl_wmbus.c -lm
In file included from rtl_wmbus.c:37:
atan2.h: In function ‘atan2_approximation2’:
atan2.h:48:11: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
     if (x == 0.0f)
           ^~
atan2.h:51:15: warning: comparing floating point with == or != is unsafe [-Wfloat-equal]
         if (y == 0.0f) return 0.0f;
               ^~
In file included from rtl_wmbus.c:39:
t1_c1_packet_decoder.h: At top level:
t1_c1_packet_decoder.h:252:5: warning: no previous prototype for ‘get_mode_a_tlg_length’ [-Wmissing-prototypes]
 int get_mode_a_tlg_length(uint8_t lfield)
     ^~~~~~~~~~~~~~~~~~~~~
t1_c1_packet_decoder.h:257:5: warning: no previous prototype for ‘get_mode_b_tlg_length’ [-Wmissing-prototypes]
 int get_mode_b_tlg_length(uint8_t lfield)
     ^~~~~~~~~~~~~~~~~~~~~
rtl_wmbus.c:47:14: warning: ‘lp_1600kHz_56kHz’ defined but not used [-Wunused-function]
 static float lp_1600kHz_56kHz(int sample, size_t i_or_q)

weetmuts, 2020-07-02:
Please clone my fork of rtl-wbus here: https://github.com/weetmuts/rtl-wmbus
it has fixes for those build errors.

rudsomlima, 2020-07-03:
I can't install rtl-wmbus:

root@raspberrypi:/home/pi/Downloads/rtl-wmbus# make debug
gcc -DDEBUG  -O0 -g3 -ggdb -p -pg -Iinclude -std=gnu99 -Wall -W -Waggregate-return -Wbad-function-cast -Wcast-align -Wcast-qual -Wchar-subscripts -Wcomment -Wno-float-equal -Winline -Wmain -Wmissing-noreturn -Wno-missing-prototypes -Wparentheses -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-prototypes -Wswitch -Wunreachable-code -Wunused -Wuninitialized -o "build/rtl_wmbus" rtl_wmbus.c -lm
root@raspberrypi:/home/pi/Downloads/rtl-wmbus# ls /usr/bin/rtl*
/usr/bin/rtl_adsb  /usr/bin/rtl_eeprom  /usr/bin/rtl_fm  /usr/bin/rtl_power  /usr/bin/rtl_sdr  /usr/bin/rtl_tcp  /usr/bin/rtl_test

weetmuts, 2020-07-03:
The rtl_wmbus source is not that helpful, you need to copy the built binary into its proper location:

sudo cp build/rtl_wmbus /usr/bin

rudsomlima, 2020-07-03:
It worked. It remains to test the night at home to listen to the meters. 
I still inform you today if it works.
Thanks!

root@raspberrypi:/home/pi/Downloads/rtl-wmbus# cp build/rtl_wmbus /usr/bin
root@raspberrypi:/home/pi/Downloads/rtl-wmbus# ls /usr/bin/rtl*
root@raspberrypi:/home/pi/Downloads/rtl-wmbus# wmbusmeters --debug --t1 'rtlwmbus:/usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus'
(wmbusmeters) version: 0.9.33-3-g735f11f
(config) using device: rtlwmbus
(config) with: /usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(config) number of meters: 0
(detect) "rtlwmbus" "/usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(detect) driver: rtlwmbus
(rtlwmbus) using command: /usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop

weetmuts, 2020-07-03:
It should be possible for you to run using this simpler command line:

wmbusmeters --debug --t1 rtlwmbus:434.47M

This will simply replace the frequency 868.95M in the default command line with 434.47M

rudsomlima, 2020-07-03:
root@raspberrypi:/etc# rtl_433 -R 104 -s 1200k -f 434470k
rtl_433 version 20.02-71-gf35becb branch master at 202006272117 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 154 device decoding protocols [ 104 ]
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1200000 S/s.
Tuner gain set to Auto.
Tuned to 434.470MHz.
Allocating 15 zero-copy buffers

time      : 2020-07-03 19:37:38
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 1206365
Version   : 136          Device Type: 0x00         Device Type String: Other Control   : 0x44
Data Length: 26          Data      : 1944304cff62ba008800a2210d001346f3f5d5735dea09de53fe              Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000

root@raspberrypi:/etc# wmbusmeters --format=fields rtlwmbus:434.47M casa izar 1206325 NOKEY
Not a valid id nor a valid meter match expression "1206325"

*/etc/wmbusmeters.conf*
loglevel=normal
device=rtlwmbus
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
meterfilesnaming=name
meterfilestimestamp=day
logfile=/var/log/wmbusmeters/wmbusmeters.log

*/etc/wmbusmeters.d/casa*
name=casa
type=izar
id=*
key=NOKEY

root@raspberrypi:/etc# wmbusmeters --debug --t1 rtlwmbus:434.47M
(wmbusmeters) version: 0.9.33-3-g735f11f
(config) using device: rtlwmbus
(config) with: 434.47M
(config) number of meters: 0
(detect) "rtlwmbus" "434.47M"
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 434.47M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop

[https://docs.google.com/document/d/1zA42CxCJwc1yxUmq4IcGFBHz7_4AHqb5q4JQU0vbJBU/edit?usp=sharing](url)

rudsomlima, 2020-07-04:
I put the code above to say that it is still not generating the logs of the meters.  I already removed and put the rtl 2832 and even then it didn't generate.

weetmuts, 2020-09-04:
@rudsomlima , it seems like you are receiving wmbus telegrams properly! 

Unfortunately the meter is a Sappel IZAR r3 which is not currently supported. 

Please follow up in #127 where I have merged all izar r3 issues.

closed 2020-09-04 by weetmuts