wmbusmeters/wmbusmeters #131: AMB8465 USB stick

WMBus-Sniffer, 2020-06-29:
In the Readme is stated, that "the dongle to be factory reset before use."

How do i can factory reset the AMB8465 USB stick? Which Linux tools I need?

thanks

weetmuts, 2020-06-29:
Ok, it is a bit tricky to stty and echo the bytes to the serial port for each possible baud rate,
so I have now added reset support to wmbusmeters-admin and pushed to master.
Run: 
./build/wmbusmeters-admin --debug

in a second window do: tail -f /var/log/syslog

Now you should see your serial port to the left. (like /dev/ttyUSB0 or something.)
Then try detecting amb8465, which will probably not work, since it is using the wrong buaud, right?
Then try to reset the amb8465. Capture the syslog and email it to me, so I can verify that it looks ok.
:-) I do not have the amb8465 here to test myself.

WMBus-Sniffer, 2020-06-29:
I sent you the logfile. (I am not sure how long I should run the admin app).
The Logfile looks like this:

Jun 29 11:54:52 kali wmbusmeters-admin[4109]: (shell) output: >>><<<
Jun 29 11:54:52 kali wmbusmeters-admin[4109]: (shell) return code 1
Jun 29 11:54:52 kali wmbusmeters-admin[4109]: (shell) 4369 exited
Jun 29 11:54:52 kali wmbusmeters-admin[4109]: (shell) exec (capture output) "/bin/pidof"
Jun 29 11:54:52 kali wmbusmeters-admin[4109]: (shell) arg "wmbusmeters"
Jun 29 11:54:52 kali wmbusmeters-admin[4109]: (shell) output: >>><<<
Jun 29 11:54:52 kali wmbusmeters-admin[4109]: (shell) return code 1
Jun 29 11:54:52 kali wmbusmeters-admin[4109]: (shell) 4370 exited
Jun 29 11:54:54 kali wmbusmeters-admin[4109]: (serial) stopping manager
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) exec (capture output) "/usr/bin/id"
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) arg "-u"
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) output: >>>1000#012<<<
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) return code 0
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) 4379 exited
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) exec (capture output) "/bin/pidof"
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) arg "wmbusmetersd"
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) output: >>><<<
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) return code 1
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) 4380 exited
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) exec (capture output) "/bin/pidof"
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) arg "wmbusmeters"
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) output: >>><<<
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) return code 1
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) 4381 exited
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) exec (capture output) "/bin/pidof"
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) arg "wmbusmetersd"
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) output: >>><<<
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) return code 1
Jun 29 11:55:21 kali wmbusmeters-admin[4378]: (shell) 4382 exited

thanks

weetmuts, 2020-06-29:
Well, did it work? Can you detect it using the same wmbusmeters-admin program?

The syslog should show something like the below, however with some success somewhere.
You can do : tail -f /var/log/syslog | grep wmbusmeters-admin | grep -v shell
to see only the interesting parts.

Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serialtty) opened /dev/ttyUSB0
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial) received binary ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) try factory reset using baud 9600
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial /dev/ttyUSB0) sent "FF1100EE"
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial) received binary ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serialtty) closed /dev/ttyUSB0
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) reset response ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) no response to factory reset using baud 9600
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serialtty) opened /dev/ttyUSB0
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial) received binary ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) try factory reset using baud 19200
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial /dev/ttyUSB0) sent "FF1100EE"
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial) received binary ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serialtty) closed /dev/ttyUSB0
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) reset response ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) no response to factory reset using baud 19200
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serialtty) opened /dev/ttyUSB0
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial) received binary ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) try factory reset using baud 38400
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial /dev/ttyUSB0) sent "FF1100EE"
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial) received binary ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serialtty) closed /dev/ttyUSB0
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) reset response ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) no response to factory reset using baud 38400
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serialtty) opened /dev/ttyUSB0
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial) received binary ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) try factory reset using baud 115200
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial /dev/ttyUSB0) sent "FF1100EE"
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serial) received binary ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (serialtty) closed /dev/ttyUSB0
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) reset response ""
Jun 29 18:19:45 eventb wmbusmeters-admin[17776]: (amb8465) no response to factory reset using baud 115200

WMBus-Sniffer, 2020-06-29:
This is the output:
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (serialtty) opened /dev/ttyUSB0
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (serial) received binary ""
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (amb8465) try factory reset using baud 9600
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (serial /dev/ttyUSB0) sent "FF1100EE"
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (serial) received binary ""
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (serialtty) closed /dev/ttyUSB0
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (amb8465) reset response ""
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (amb8465) no response to factory reset using baud 9600
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (serialtty) opened /dev/ttyUSB0
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (serial) received binary ""
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (amb8465) try factory reset using baud 19200
Jun 29 12:40:28 kali wmbusmeters-admin[5232]: (serial /dev/ttyUSB0) sent "FF1100EE"
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serial) received binary ""
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serialtty) closed /dev/ttyUSB0
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (amb8465) reset response ""
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (amb8465) no response to factory reset using baud 19200
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serialtty) opened /dev/ttyUSB0
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serial) received binary ""
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (amb8465) try factory reset using baud 38400
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serial /dev/ttyUSB0) sent "FF1100EE"
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serial) received binary ""
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serialtty) closed /dev/ttyUSB0
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (amb8465) reset response ""
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (amb8465) no response to factory reset using baud 38400
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serialtty) opened /dev/ttyUSB0
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serial) received binary ""
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (amb8465) try factory reset using baud 115200
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serial /dev/ttyUSB0) sent "FF1100EE"
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serial) received binary ""
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (serialtty) closed /dev/ttyUSB0
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (amb8465) reset response ""
Jun 29 12:40:29 kali wmbusmeters-admin[5232]: (amb8465) no response to factory reset using baud 115200

weetmuts, 2020-06-29:
And what happens when you select the "Detect wmbus receiver" -> amb8465 
what does it say?

Wmbusmeters tries baud rates from 1200 to 115200, but they are all 8N1 
unfortunately it seems that the module supports odd/even parity as well as 1 or 2 stop bits.

Do you know what setting the amb8465 was set to before? 

Also, is there a small button in the amb8465 dongle? Perhaps that one will perform the hard factory reset?

WMBus-Sniffer, 2020-06-29:
The device worked the first time without problem.
I tried to use it after a week paused and there was no receiving data anymore.
I pressed the button many times (from 10 second until 2 Minutes) without any different result (no led indication that there is a reset ongoing)
I don't know which settings was in the beginning.

When I use "Detect wmbus receiver" this ist the result:
Jun 29 13:44:45 kali wmbusmeters-admin[7647]: (serial) stopping manager
Jun 29 13:46:31 kali wmbusmeters-admin[8838]: (amb8465) exists? /dev/ttyUSB0
Jun 29 13:46:31 kali wmbusmeters-admin[8838]: (amb8465) checking /dev/ttyUSB0
Jun 29 13:46:31 kali wmbusmeters-admin[8838]: (serialtty) opened /dev/ttyUSB0
Jun 29 13:46:31 kali wmbusmeters-admin[8838]: (serial) received binary ""
Jun 29 13:46:31 kali wmbusmeters-admin[8838]: (amb8465) are you there?
Jun 29 13:46:31 kali wmbusmeters-admin[8838]: (serial /dev/ttyUSB0) sent "FF0B00F4"
Jun 29 13:46:31 kali wmbusmeters-admin[8838]: (serial) received binary ""
Jun 29 13:46:31 kali wmbusmeters-admin[8838]: (serialtty) closed /dev/ttyUSB0

 "nothing there"

weetmuts, 2020-06-29:
If it worked once with wmbusmeters and then stopped, then I am not sure what the problem is.
But I am certain that a factory reset should not be necessary, since wmbusmeters never change
the values that are restored by the factory reset.

Unplugging it and re-inserting it again does not help either?

WMBus-Sniffer, 2020-06-29:
Strange things
......
sudo build/wmbusmeters --listento=t1 --debug --verbose /dev/ttyUSB0:amb8465
(wmbusmeters) version: 0.9.31-34-g24a42cb
(config) using device: /dev/ttyUSB0
(config) with: amb8465
(config) number of meters: 0
(detect) "/dev/ttyUSB0" "amb8465"
(detect) is_tty=1 is_stdin=0 is_file=0
(amb8465) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(amb8465) set link mode 08
(serial /dev/ttyUSB0) sent "FF040108F2000000"

weetmuts, 2020-06-29:
yeah, but does it work? does the log say receive after inte sent the FF040108.... bytes?

if it works, then it is a classical case of who owns the tty.

WMBus-Sniffer, 2020-06-30:
Sorry no progress .....

root@kali # build/wmbusmeters --listento=t1 --debug --verbose /dev/ttyUSB0:amb8465
(wmbusmeters) version: 0.9.31-34-g24a42cb
(config) using device: /dev/ttyUSB0
(config) with: amb8465
(config) number of meters: 0
(detect) "/dev/ttyUSB0" "amb8465"
(detect) is_tty=1 is_stdin=0 is_file=0
(amb8465) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(amb8465) set link mode 08
(serial /dev/ttyUSB0) sent "FF040108F2000000"

^C(serial) stopping manager
(amb8465) get config
(serial /dev/ttyUSB0) sent "FF0A02008077"
(config) listen to link modes: t1
(serial) waiting for stop
(serial) event loop stopped!
(serial) closing 1 devices
(serial) no devices working
(serial) no devices working
(serialtty) closed /dev/ttyUSB0
root@kali# ls -l /dev/ttyUSB0 
crw-rw---- 1 root root 188, 0 Jun 30 03:20 /dev/ttyUSB0
root@kali:# id
uid=0(root) gid=0(root) groups=0(root)

weetmuts, 2020-06-30:
Have you used a different tool on this dongle? Like the Amber own tool, and changed stuff like baud rate etc?

weetmuts, 2020-06-30:
I have updated the README to explain better, that usually you do >not< have to factory reset your amb8465.
It is only necessary if you have changed the baud rate using a different tool.

WMBus-Sniffer, 2020-06-30:
So, I connected the USB stick with the ACC3 Tool to the device and flashed the firmware.
I also checked the baud rate and the windows tool reports that it is 9600.
After bringing the device back to linux (all virtual) neither the wmbusmeter-admin reset nor the the wmbusmeters listening have change the behavior.

(I know this might be late, but what is the difference between 8465 and 8665?)

weetmuts, 2020-06-30:
Are you running a virtual Linux environment on Windows?

WMBus-Sniffer, 2020-06-30:
Hostsystem: Fedora 32
Guest: Kali Debian
Guest: Windows 10

weetmuts, 2020-06-30:
What happens if you run wmbusmeters on the host fedora system?

WMBus-Sniffer, 2020-06-30:
No different behavior.
I tested this as well on different Linux Distributions (virtual and Host), all shows no communication with the device :(

weetmuts, 2020-06-30:
And it works properly with the Windows software?

WMBus-Sniffer, 2020-06-30:
[image]
I am only using the config App (ACC). But with this software the firmware update and config readout are possible.

weetmuts, 2020-06-30:
This is really weird. We know it worked with wmbusmeters once, but it no longer works. It clearly works with the windows software. The windows software says COM6, do you have other serial devices/ports in the computer?

How do you run the tests from the command line? Have you tried all available ttys?
./build/wmbusmeters --listento=t1,c1 /dev/ttyUSBxx
xx from 0 to whichever there are?

When you run wmbusmeters-admin, which serial ports does it list?

Is the wmbusmeters daemon running already? Has it taken the dongle tty and you get a different one from another
serial->usb device? Has some other software taken the tty? For example if you run a virtual subsytem
and configure it to use the tty, has the virtualbox taken your tty?

WMBus-Sniffer, 2020-06-30:
The COM6 might be regarding virtualization and other reasons. General, the Windows recognize the device and the Linux as well and defined a proper device. 
In Linux this is /dev/ttyUSB0 according from dmesg and there is no other /dev/ttyUSB# device connected.
I am still convinced, that virtualization will bring the device in a perfect way to the guest system.

The wmbusmeters-admin list the ttyUSB0 and reports Jun 30 09:41:03 kali wmbusmeters-admin[2156]: (amb8465) exists? /dev/ttyUSB0

Jun 30 09:41:29 kali wmbusmeters-admin[2156]: (serial) stopping manager
Jun 30 12:01:38 kali wmbusmeters-admin[3754]: (amb8465) exists? /dev/ttyUSB0
Jun 30 12:01:38 kali wmbusmeters-admin[3754]: (amb8465) checking /dev/ttyUSB0
Jun 30 12:01:38 kali wmbusmeters-admin[3754]: (serialtty) opened /dev/ttyUSB0
Jun 30 12:01:38 kali wmbusmeters-admin[3754]: (serial) received binary ""
Jun 30 12:01:38 kali wmbusmeters-admin[3754]: (amb8465) are you there?
Jun 30 12:01:38 kali wmbusmeters-admin[3754]: (serial /dev/ttyUSB0) sent "FF0B00F4"
Jun 30 12:01:38 kali wmbusmeters-admin[3754]: (serial) received binary ""
Jun 30 12:01:38 kali wmbusmeters-admin[3754]: (serialtty) closed /dev/ttyUSB0

[image]

I would suggest to thank you for your effort and try to test the setup with other USB sticks and verify the tools. This will take some time, therefore, maybe other users could confirm the reset function on their setup :)

weetmuts, 2020-07-09:
Ok, please post here and tag me, if you find out the cause for your problem! Closing this for now!

closed 2020-07-09 by weetmuts