wmbusmeters/wmbusmeters #132: Add SAP izar R3 water counter

aboucherka, 2020-06-30:
Hi, 

How can i decode  a frame of "SAP izar R3" water counter, it's  a widespread model in France.
I have 2 in my house. Below a dump produce with rtl_sdr

root@rpi1:~# rtl_sdr -f 868.9M -s 1.6e6 - | rtl_wmbus
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 1600000 S/s.
Tuned to 868900000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers
T1;0;0;2020-06-30 19:45:02.000;29;31;05FFFFFF;0x1944304cffffff05ff00a1613c43ff17ffe44e1e46d1077cffba
T1;1;1;2020-06-30 19:45:02.000;126;115;0013ACD0;0x1c441486d0ac13000f0fa0423505000040030000000005ff0533030500
T1;0;0;2020-06-30 19:45:10.000;27;30;05ABF8FF;0x194430fffff8ab050000a1713c43131739cb07b567ffdea789b8
T1;1;1;2020-06-30 19:45:19.000;125;136;0013AD04;0x1c44148604ad13000f0fa05e9b09000040030000000005ff0593460900
T1;0;0;2020-06-30 19:45:19.000;30;30;FFABF8FF;0x194430fffff8abff0000a1013c4313175c07fbe48239d3ffc6b5
T1;1;1;2020-06-30 19:45:21.000;124;128;0012BE60;0x1c44148660be12000f0fa0b63e0b000040030000000005ff05478e0a00
T1;0;0;2020-06-30 19:45:29.000;27;26;05ABF875;0x1944304c75f8ab050000a1113c4313174b28b24fa3ffff7ddfb7
T1;1;1;2020-06-30 19:45:37.000;129;135;0013AD04;0x1c44148604ad13000f0fa0629b09000040030000000005ff0593460900
T1;0;0;2020-06-30 19:45:38.000;25;30;05ABF8FF;0x1944fffffff8ab050000a1213c4313177259ffffc19e60fff4b0
T1;1;1;2020-06-30 19:45:44.000;125;138;0013A0A2;0x1c441486a2a013000f0fa0d4f611000040030000000005ff05c5291100
T1;0;0;2020-06-30 19:45:46.000;28;27;05ABF8FF;0x194430fffff8ab050000a1313c43ff1765762119e04db9cbedb2
T1;0;0;2020-06-30 19:45:54.000;27;28;FFFFF875;0x1944ff4c75f8ffff0000a1413c43131700ffdd4805ffb4ffffbf
T1;1;1;2020-06-30 19:46:00.000;128;110;0013AB67;0x1c44148667ab13000f0fa030330e000040030000000005ff05358c0d00
T1;0;0;2020-06-30 19:46:02.000;26;27;05ABF875;0x1944304c75f8ab050000a1513c431317179594e324a56d11bbbd
T1;0;0;2020-06-30 19:46:11.000;29;27;05ABF875;0x1944304c75f8ab050000a1ff3c4313172ee44e1effd1077c90ff
T1;1;1;2020-06-30 19:46:18.000;128;129;001333B4;0x1c441486b43313000f0fa0e19404000040030000000005ff0543670400
T1;0;0;2020-06-30 19:46:20.000;25;27;05FFFF75;0x1944304c75ffff050000a1713c43131739cb07ff6702dea789b8
T1;0;0;2020-06-30 19:46:28.000;26;26;05FFF8FF;0x19ff304cfff8ff050000a1013c4313175c07fbe48239d3ffc6b5

Thank for help

aboucherka, 2020-06-30:
also a dump with  "rtl_433"

root@rpi1:~#  rtl_433 -f 868.95MHZ
rtl_433 version 20.02-63-gbea205c branch master at 202006101126 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 125 out of 153 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-153 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.950MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-30 21:25:06
model     : Wireless-MBus Mode      : T            Manufacturer: APT         ID        : 133414
Version   : 15           Device Type: 0x0F         Device Type String: Unknown                         Control   : 0x44          Data Length: 29           Data      : 1c441486b43313000f0fa0e69404000040030000000005ff0543670400
Integrity : CRC
Control Info: 0xA0       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-30 21:25:12
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 6125875
Version   : 0            Device Type: 0x00         Device Type String: Other Control   : 0x44          Data Length: 26           Data      : 1944304c75f8ab050000a1513c431317179594e324a56d11bbbd
Integrity : CRC
Control Info: 0xA1       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-30 21:25:12
model     : Wireless-MBus Mode      : T            Manufacturer: APT         ID        : 141167
Version   : 15           Device Type: 0x0F         Device Type String: Unknown                         Control   : 0x44          Data Length: 29           Data      : 1c44148667ab13000f0fa03a330e000040030000000005ff05358c0d00
Integrity : CRC
Control Info: 0xA0       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-30 21:25:13
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 6125882
Version   : 0            Device Type: 0x00         Device Type String: Other Control   : 0x44          Data Length: 26           Data      : 1944304c7cf8ab050000a161000013ce7369561669edb8995a28
Integrity : CRC
Control Info: 0xA1       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-30 21:25:22
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 6125882
Version   : 0            Device Type: 0x00         Device Type String: Other Control   : 0x44          Data Length: 26           Data      : 1944304c7cf8ab050000a171000013ce64461fbd483e6142432a
Integrity : CRC
Control Info: 0xA1       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-30 21:25:29
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 6125875
Version   : 0            Device Type: 0x00         Device Type String: Other Control   : 0x44          Data Length: 26           Data      : 1944304c75f8ab050000a1713c43131739cb07b56702dea789b8
Integrity : CRC
Control Info: 0xA1       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-30 21:25:30
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 6125882
Version   : 0            Device Type: 0x00         Device Type String: Other Control   : 0x44          Data Length: 26           Data      : 1944304c7cf8ab050000a101000013ce018ae3ecad056c430c27
Integrity : CRC
Control Info: 0xA1       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
[... cut ...]  _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-30 21:25:31
model     : Wireless-MBus Mode      : T            Manufacturer: APT         ID        : 140102
Version   : 15           Device Type: 0x0F         Device Type String: Unknown                         Control   : 0x44          Data Length: 29           Data      : 1c441486a2a013000f0fa0dcf611000040030000000005ff05c5291100
Integrity : CRC
Control Info: 0xA0       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000

weetmuts, 2020-07-01:
Thanks! It seems like you have two different meters transmitting telegrams here?
Again, I am not an izar expert, but the 1c44... telegrams do not properly decode using the izar driver.
The 1944... telegrams do decode, or at least the izar driver does not print fail, however the telegram
is probably not decoded right anyway, it says:

Vatten	05abf875	2931.836 m3	2931.836 m3	2004-12-31	14 y	meter_blocked,underflow,mechanical_fraud	mechanical_fraud	2020-07-01 21:43.40

Which does not look right....

@ZeWaren do you have any opinion?

weetmuts, 2020-07-30:
Closing this issue, since adding izar R3 will he handled in this issue https://github.com/weetmuts/wmbusmeters/issues/127

weetmuts, 2021-08-08:
It seems like the other issue was not really an izar r3 meter but an Diehl/Izar/Hydrus RS meter.
Re-opening this issue.

weetmuts, 2022-01-27:
@aboucherka The izar driver can decode your SAP telegrams now! 

{"media":"water","meter":"izar","name":"Water","id":"05abf87c","prefix":"@28@@","serial_number":"047484","total_m3":1777.253,"last_month_total_m3":1759.958,"last_month_measure_date":"2020-01-01","remaining_battery_life_y":0,"current_alarms":"no_alarm","previous_alarms":"no_alarm","transmit_period_s":8,"manufacture_year":"2028","timestamp":"2022-01-27T19:14:09Z"}

weetmuts, 2022-01-27:
Do you know what the APT telegrams are from what kind of meter?

aboucherka, 2022-01-28:
Thank you ,
I can't check because since 2020 I moved to another apartment.

Regards

closed 2022-01-28 by weetmuts