wmbusmeters/wmbusmeters #137: DWZ Lorenz (Engelmann Waterstar M20) Water Meter

ArtyomBurda, 2020-07-08:
Hi,

Got a water meter (Engelmann Waterstar M20) that is not in the list of the supported devices and need to read it out. 
Have been provided with the following AES-key: _D4A4B37FCE247D0220C072AC783F500B_ 
However, not able to properly decrypt with it. Is it possible that the key is just wrong? Maybe you can help anyhow? 

This is the log I get:
```
wmbusmeters --debug --listento=t1 /dev/ttyUSB0:im871a
(wmbusmeters) version: 0.9.33-3-g735f11f
(config) using device: /dev/ttyUSB0
(config) with: im871a
(config) number of meters: 0
(detect) "/dev/ttyUSB0" "im871a"
(detect) is_tty=1 is_stdin=0 is_file=0
(im871a) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(im871a) set link mode 03
(serial /dev/ttyUSB0) sent "A5010306000203300101"
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) received binary "A5810617FF000300B32581201000010001FF0702320001010000006A70"
(im871a) checkIM871AFrame "A5810617FF000300B32581201000010001FF0702320001010000006A70"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 706a expected 706a
(im871a) received full frame
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 03
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00102081
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) listen to link modes: t1
(serial) waiting for stop
(serial) received binary "A5E2033944FA122162092002067A1E0020257FF86B9582AAB3654AF25D5CC2A7B395E342FE3AF08C72F37ADB33A79D59147C03FD0C08000002FD0B0011000000009218D5"
(im871a) checkIM871AFrame "A5E2033944FA122162092002067A1E0020257FF86B9582AAB3654AF25D5CC2A7B395E342FE3AF08C72F37ADB33A79D59147C03FD0C08000002FD0B0011000000009218D5"
(im871a) has_timestamp=1 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) timestamp 00000000
(im871a) rssi 92
(im871a) got crc16 d518 expected d518
(im871a) received full frame
(wmbus) parseDLL @0 58
(wmbus) parseELL @10 48
(wmbus) parseAFL @10 48
(wmbus) parseTPL @10 48
Received telegram from: 20096221
          manufacturer: (DWZ) Lorenz, Germany
           device type: Warm Water (30°C-90°C) meter
(wmbus) 00: 39 length (57 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: fa12 dll-mfct (DWZ)
(wmbus) 04: 21620920 dll-id (20096221)
(wmbus) 08: 02 dll-version
(wmbus) 09: 06 dll-type (Warm Water (30°C-90°C) meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: 1e tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 2025 tpl-cfg 2520 (AES_CBC_IV synchronous )
(wmbus) "telegram=||3944FA122162092002067A1E0020257FF86B9582AAB3654AF25D5CC2A7B395E342FE3AF08C72F37ADB33A79D59147C03FD0C08000002FD0B0011|+40"
(serial) received binary "A5E2033944FA122162092002067A1F00202595DD62EA95C29B14889F792326075B76BBC213A71BC8DF095F55F9CD25F8772803FD0C08000002FD0B001100000000A08F37"
(im871a) checkIM871AFrame "A5E2033944FA122162092002067A1F00202595DD62EA95C29B14889F792326075B76BBC213A71BC8DF095F55F9CD25F8772803FD0C08000002FD0B001100000000A08F37"
```

Thanks a lot in advance!

weetmuts, 2020-07-08:
Thanks! Can you post/email a log with a couple more telegrams? I tried to decrypt this telegram but it seems it is not quite the same as the other meters way of encrypting using AES_CBC_IV... perhaps it only encrypts part of the payload.

weetmuts, 2020-07-09:
Ok, I have pushed a fix to wmbusmeters to handle partially encrypted telegrams, which is what your waterstarm sends.
Unfortunately I cannot yet decode your telegram because the decrypted part does not decrypt properly with your key.

Are there more than one waterstar meter in your surroundings? Have you checked that the id 20096221 corresponds
to your meter?

ArtyomBurda, 2020-07-09:
Thanks a lot for your support! Here are the specifications of radio transmission from the doc to the meter: 
[image]
The paper from the supplier sais that it has been set up for sending the short telegrams.

Log with couple more telegrams:
```
wmbusmeters --debug --listento=t1 /dev/ttyUSB0:im871a
(wmbusmeters) version: 0.9.33-3-g735f11f
(config) using device: /dev/ttyUSB0
(config) with: im871a
(config) number of meters: 0
(detect) "/dev/ttyUSB0" "im871a"
(detect) is_tty=1 is_stdin=0 is_file=0
(im871a) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(im871a) set link mode 03
(serial /dev/ttyUSB0) sent "A5010306000203300101"
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) received binary "A5810617FF000300B32581201000010001FF0702320001010000006A70"
(im871a) checkIM871AFrame "A5810617FF000300B32581201000010001FF0702320001010000006A70"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 706a expected 706a
(im871a) received full frame
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 03
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00102081
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) listen to link modes: t1
(serial) waiting for stop
(serial) received binary "A5E2033944FA122162092002067A5F00202529D52C112DAC32B507CE6F3A09026AF31577D8270B7E6A6338F7618005BE333D03FD0C08000002FD0B001100000000A2E69C"
(im871a) checkIM871AFrame "A5E2033944FA122162092002067A5F00202529D52C112DAC32B507CE6F3A09026AF31577D8270B7E6A6338F7618005BE333D03FD0C08000002FD0B001100000000A2E69C"
(im871a) has_timestamp=1 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) timestamp 00000000
(im871a) rssi a2
(im871a) got crc16 9ce6 expected 9ce6
(im871a) received full frame
(wmbus) parseDLL @0 58
(wmbus) parseELL @10 48
(wmbus) parseAFL @10 48
(wmbus) parseTPL @10 48
Received telegram from: 20096221
[... 128 lines cut ...]
(im871a) got crc16 9417 expected 9417
(im871a) received full frame
(wmbus) parseDLL @0 58
(wmbus) parseELL @10 48
(wmbus) parseAFL @10 48
(wmbus) parseTPL @10 48
Received telegram from: 20096221
          manufacturer: (DWZ) Lorenz, Germany
           device type: Warm Water (30°C-90°C) meter
(wmbus) 00: 39 length (57 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: fa12 dll-mfct (DWZ)
(wmbus) 04: 21620920 dll-id (20096221)
(wmbus) 08: 02 dll-version
(wmbus) 09: 06 dll-type (Warm Water (30°C-90°C) meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: 64 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 2025 tpl-cfg 2520 (AES_CBC_IV synchronous )
(wmbus) "telegram=||3944FA122162092002067A6400202515084D38D6DAAD1CB753AB39BE30C4A1E9196D7C2E9E2E9BA88874675EC6CD0503FD0C08000002FD0B0011|+348"
```

Thanks again! Yes, this is the only meter I have. For now it is not even connected because I want to test it beforehand. Yes, the number is correct, on the meter itself I can see "DWZ02 2009 6221"

Hope it helps. Anyway, I will try to contact the supplier and get a more information on the encryption they are using.

weetmuts, 2020-07-09:
Thanks! I have double checked the code and data. Your meter telegram reports AES_CBC_IV aka mode 5.
This mode is used by several other meters and seems to be working for them.
According to the standard, the decrypted first block should start with hex 2F2F, this is
how you know that you used the right key. 

If you run with the latest master and the command line:
./build/wmbusmeters --debug /dev/ttyUSB0:im871a MyWaterMeter waterstarm:t1 20096221 D4A4B37FCE247D0220C072AC783F500B

Then you will see output like this:
(TPL) decrypting "15084D38D6DAAD1CB753AB39BE30C4A1E9196D7C2E9E2E9BA88874675EC6CD0503FD0C08000002FD0B0011"
(TPL) num encrypted blocks 2 (32 bytes and remaining unencrypted 11 bytes)
(TPL) IV FA122162092002066464646464646464
(TPL) decrypted  "2AED2F44CE501677912574B4422C504993154F23A20375B3826FBBEAC0C02F03"
(TPL) appended   "2AED2F44CE501677912574B4422C504993154F23A20375B3826FBBEAC0C02F0303FD0C08000002FD0B0011"
(wmbus) decrypted content failed check, did you use the correct decryption key? Ignoring telegram.

The decrypted data should start with 2F2F if you have the right key. Here the decrypted data started with 2AED...

When we got that working, then we can start understanding the content of the telegram, which might
be either compact or full length, fully compliant and supported or something in between.... :-)

The 11 not-encrypted bytes at the end says:
(wmbus) 11: 03 dif (24 Bit Integer/Binary Instantaneous value)
(wmbus) 12: FD vif (Second extension of VIF-codes)
(wmbus) 13: 0C vife (Model/Version)
(wmbus) 14: 080000
(wmbus) 17: 02 dif (16 Bit Integer/Binary Instantaneous value)
(wmbus) 18: FD vif (Second extension of VIF-codes)
(wmbus) 19: 0B vife (Parameter set identification)
(wmbus) 1a: 0011

ArtyomBurda, 2020-07-09:
Thats a lot of info! Thank you! I will order the optical head soon to be able to check/change the AES-key. As soon as I have more info I will get back to you =)

ArtyomBurda, 2020-07-30:
Hi again!

I've finally received the optical head and was able to set up the key myself. Now it is possible to decrypt the telegrams!

Here is a bit of log:

```
./build/wmbusmeters --debug /dev/ttyUSB0:im871a MyWaterMeter waterstarm:t1 20096221 BEDB81B52C29B5C143388CBB0D15A051
(wmbusmeters) version: 0.9.34
(config) using device: /dev/ttyUSB0
(config) with: im871a
(config) number of meters: 1
(detect) "/dev/ttyUSB0" "im871a"
(detect) is_tty=1 is_stdin=0 is_file=0
(im871a) on /dev/ttyUSB0
(serialtty) opened /dev/ttyUSB0
(config) meter waterstarm link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(wmbusmeters) configured "MyWaterMeter" "waterstarm" "20096221" encrypted
(im871a) set link mode 03
(serial /dev/ttyUSB0) sent "A5010306000203300101"
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) received binary "A5810617FF000300B32581201000010001FF070232000101000096D581"
(im871a) checkIM871AFrame "A5810617FF000300B32581201000010001FF070232000101000096D581"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 81d5 expected 81d5
(im871a) received full frame
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 03
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00102081
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) listen to link modes: t1
(serial) waiting for stop
(serial) received binary "A5E203294468507931508076F0A0009F27F8280038F82800001F0201232AF7563BF756008DC3009ED4000FE500000000003AD64E"
(im871a) checkIM871AFrame "A5E203294468507931508076F0A0009F27F8280038F82800001F0201232AF7563BF756008DC3009ED4000FE500000000003AD64E"
(im871a) has_timestamp=1 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) timestamp 00000000
(im871a) rssi 3a
(im871a) got crc16 4ed6 expected 4ed6
(im871a) received full frame
(wmbus) parseDLL @0 42
(meter) MyWaterMeter: for me? 80503179
(meter) MyWaterMeter: not for me: not my id
[... 259 lines cut ...]
(waterstarm) 25: 00000000
(waterstarm) 29: 2F skip
(waterstarm) 2a: 2F skip
(waterstarm) 2b: 2F skip
(waterstarm) 2c: 2F skip
(waterstarm) 2d: 2F skip
(waterstarm) 2e: 2F skip
(waterstarm) 2f: 03 dif (24 Bit Integer/Binary Instantaneous value)
(waterstarm) 30: FD vif (Second extension of VIF-codes)
(waterstarm) 31: 0C vife (Model/Version)
(waterstarm) 32: 080000
(waterstarm) 35: 02 dif (16 Bit Integer/Binary Instantaneous value)
(waterstarm) 36: FD vif (Second extension of VIF-codes)
(waterstarm) 37: 0B vife (Parameter set identification)
(waterstarm) 38: 0011
MyWaterMeter    20096221        0.106 m3        2020-07-30 09:42.07
```
From the log I can see the decrypted data of the total consumption. However, I am a bit confused by some other ids which appear:
```
(meter) MyWaterMeter: for me? 80503179
(meter) MyWaterMeter: not for me: not my id
```
There are no other wmbus devices nearby. Is it possible that the meter provides come more data? Like temperature measurements maybe?

Thanks once more for your support!

weetmuts, 2020-07-30:
Thank you for the telegram! I have now added proper support for the meter. The output in json for your telegram:

{"media":"warm water","meter":"waterstarm","name":"MyWater","id":"20096221","meter_timestamp":"2020-07-30 10:40","total_m3":0.106,"total_backwards_m3":0,"current_status":"OK","meter_version":"000008","parameter_set":"1100","timestamp":"2020-07-30T10:23:23Z"}

The output in fields is:
MyWater	20096221	0.106 m3	0 m3	OK	2020-07-30 12:23.53

The meaing of the status bits are unknown for the moment, so status=0 is printed as OK.
Any other bit is reported as ERROR(xx) where the xx are the bits.
(Typically the bits show if there is a LEAK, BURST, BATTERY LOW or  TAMPER detection.)

I have added your telegram and key to the test suite, so that there will be no regressions in the future.

As for the other meters, you can run wmbusmeters like this:

./build/wmbusmeters  /dev/ttyUSB0:im871a

and it will print all the meters it can hear! It can be your neighbours meters or it can be more
devices in your home that you do not know about. :-)

weetmuts, 2020-07-30:
Pushed to master! Please test!

makuser, 2020-07-31:
@weetmuts Man, I feel bad that I forgot to create a PR, because I implemented the same device family when I started #99. :neutral_face: 

@ArtyomBurda do you have the Software to configure Engelmann meters? Your water meter is of a warm water type, do you happen to have an Engelmann Waterstar M cold water meter as well?

BTW: The model is "Waterstar M", the **[M20]** printed on the device itself is just the year of manufacturing.

I am now again looking into the issue why T1 works fine, but C1 just does not work, no matter what I try.

ArtyomBurda, 2020-07-31:
@weetmuts Thanks! Will be able to test it only the beginning of the next week

@makuser Yes, I have the software and the optical head to configure the meters. Unfortunately, only a warm meter. 

Thanks for the clarification!

Interesting, I will also try with C1 maybe it works for me.

ArtyomBurda, 2020-08-04:
@weetmuts Everything works lika a charm! Thanks a lot! Closing.

P.S. @makuser tried with C1, also works for me.

closed 2020-08-04 by ArtyomBurda

makuser, 2020-08-05:
> P.S. @makuser tried with C1, also works for me.

Alright, thank you for testing.