wmbusmeters/wmbusmeters #138: Multical21 in T1-OMS protocol

dguerrero88, 2020-07-20:
When I try to do a read of a Multical21 with T1-OMS protocol, a message is printed saying that the Multical21 only supports the C1 protocol. I have this water meter and using a optical-usb adapter I configured the meter to T1-OMS protocol.
There are any command to force this mode in the program?

Thanks for all, is a useful library.

Thanks for your Time.

Diego

weetmuts, 2020-07-20:
Aha, I did not know that you could reconfigure your multical21 to transmit using T1 telegrams.

You can try this: in file src/meters.h for the multical21 line replace C1_bit with C1_bit|T1_bit
This should allow you to specify :t1 for the multical21.

If you could at the same time run with --debug and email me the log, then I can check that
the telegrams are the same. If they are not the same, then the log will enable me to add 
any other data available to wmbusmeters understanding of the telegrams.

weetmuts, 2020-07-30:
Ping @dguerrero88 ! Did it work, should I add support for T1 multical21 permanently to wmbusmeters?

eborned, 2020-08-12:
I will be testing the same thing on my Multical 21 in the next few days, so I'll get back to you if this fix works!

dguerrero88, 2020-08-12:
Sorry, I can't test the library and the Multical21 before. With the changes in the library I can support T1 in Multical21 meter, but I can't test with my Multical21, because I have a Multical21 (model 40) and this model only supports C1 protocol.

eborned, 2020-08-13:
Same here unfortunately...

weetmuts, 2020-09-05:
Why do you want to use T1 mode? C1 telegrams come more often and use less battery.

weetmuts, 2020-10-25:
Hi Diego! @dguerrero88  Fix pushed to master!
Please test! Please note that there is also a huge upgrade to version 1.0.0 which includes
the new detection code.

closed 2020-10-31 by weetmuts