wmbusmeters/wmbusmeters #140: M-BUS master hat and kamstrup RF module, wM-Bus dongle.

Nikodemski2002, 2020-07-27:
Hi. I have few questions, (because I'm fresh on this topic)  about connecting wireless kamstrup meters (in my case multical 21) to my RPi with M-BUS master HAT.   @@[LINK for HAT description](https://www.packom.net/m-bus-master-hat/) 
1. Will it even work in this configuration I mentioned.
2. As M-Bus master HAT works by ttyAMA0 serial port, so if it works, I assume that device section will look like this:
`device=/dev/ttyAMA0:(I don't know what should I put here)`

3. What is the 'key' in etc/wmbusmeters.d/MyTapWater? Should I get this from kamstrup, or it is written on Multical or somewhere else?

weetmuts, 2020-07-28:
The multical21 sends wireless mbus data, but the HAT for the RPi is wired mbus.

Most users of wmbusmeters buy a dedicated wmbus usb dongle, check the README for supported dongles.
You can also use a generic radio RTLSDR dongle.

You need your meter key, if you own your meter, then you should have the key.
If it is your water-municipality/company meter, then you have to ask them for the key.
The purpose of the key is only to prevent eaves-dropping of your water consumption,
so there is no harm in you having the key to your own meter. Whether your water-municipality/company
agrees with this statement or not, depends. :-) But a lot of people have received their keys.

In one case, the water-company did not want to provide they key, but offered the user to buy
a preconfigured wmbus->mbus converter with the key inside. You connect this converter 
to your wired RPi mbus master HAT. Then I need to finish the code to speak mbus
and the values can be fetched into wmbusmeter and decoded/delivered as usual.

Nikodemski2002, 2020-07-31:
Thanks! I have ordered im871a-USB, hope everything will work fine.

closed 2020-07-31 by Nikodemski2002