wmbusmeters/wmbusmeters #141: Lansen Pulse Counter

anonymous-one, 2020-07-29:
I apologize for posting here as this is not an issue but much related.

Am currently using (for close to a year now?) wmbusmeters with my Apator water meters. Working flawlessly.

I see now Lansen has optical pulse counters available. This would be perfect for me.

https://www.lansensystems.com/products/wireless-m-bus/sensors-energy/

They are listed as compatible with wmbusmeters.

A) Anyone using these?

B) Is it a simple question of ordering one (...I would like to use 2, one for my power meter and one for my gas meter) throwing a battery in and off it goes?

C) Where to order these? I do not see them available anywhere online.

Thanks, and I am sorry for posting here but I will give my input if / once I get them going with wmbusmeters.

weetmuts, 2020-07-29:
I got the telegrams from @merbanan , are you using this pulse meter? I am thinking of using it myself as well.

I think https://www.piigab.com/ are resellers for Lansen products. Here is a direct linke.
https://www.piigab.com/produkter/hardware/lansen/pulsraknare/
https://www.piigab.com/en/order/

anonymous-one, 2020-07-29:
Thanks!

I'm going to send them a contact and see about ordering one with the probe (LAN-WMBUS-G2-OOP) plus the usb config cable (LAN-CF-CABLE).

This looks like a perfect solution as our fuse box is filled to the max (0 slots empty) and the clamp style power meters are out of the question as there is just no room unless I drill a couple of 20mm holes and mount it on the exterior of the box... Would rather avoid that.

My only concern is if these larsen ones will reach my gateway. About 50M or so to the meter (its located outside of my home, down a short driveway)... I know its hardware + environment specific and all, but do you think its realistic to except the signal to reach me?

I am using a Nooelec rtl-sdr reciever ( https://www.amazon.de/NESDR-Aluminium-RTL2832U-R820T2-Based-Software/dp/B01HA642SW/ref=sr_1_4?dchild=1&keywords=rtl+sdr&qid=1596058793&sr=8-4 ).

I will add that once I fire up wmbusmasters I see a TON of my neighbors apator meters data... Never really snooped as it doesnt interest me, but multiple homes (guessing by the interval the packets come in) so its not just the neighboring home but others as well.

Anyways, willing to experiment with what will hopefully be about 150EUR worth of hardware.

Will post my progress as (if) it all comes together.

Thanks again for the app btw... Really appreciate it!

anonymous-one, 2020-07-30:
These can be ordered directly from Larsen.

I am ordering 2 of the LAN-WMBUS-G2-OOP and the USB config cable.

Cost is very reasonable:

1-25 pieces = 59 EUR / item.

Since I am not in Sweden and have to wire the funds first etc, it will probably be 2 or so weeks before they are in my hands.

Will post my results once they arrive as well as any issues I run into if they arise.

weetmuts, 2020-07-30:
Excellent! I will close this issue for now but please re-open it, or just post in it, to let me know how it goes. 

Only 50 meter distance outside with no heavy (concrete walls) should hopefully be possible to hear! Fingers crossed! :-)
Especially since you hear a lot of other meters further away.

closed 2020-07-30 by weetmuts

anonymous-one, 2020-08-12:
Just a quick update:

I have just made the payment to ship the sensors. The ones with the optical probe are in fact 62 EUR a piece.

Quite a bit of back and forth emailing with Larsen to get them ordered. I do not fault them for it as by the sounds of it they are used to much larger orders. 

A) The configuration cable is 300 EUR. I opted not to order this.
B) The sensors come defaulted to 5 min send interval and encryption turned on.
C) The encryption keys are sent with the order.
D) Shipping EU to EU is 40 EUR.

So I am hoping its just a case of updating my wmbusmeters to latest and adding them to the config.

What I will probably do is fire one up without the optical probe attached (3 pin screw terminal) and it should hopefully start sending a 0 pulses datagram every 5 minutes and we'll see if wmbusmeters is able to read the data.

I have not dug in, but in case any changes need to be made on wmbusmeters, lansen provides a pdf of their packet structure (excuse my terminology)... Need a verified account to download it, but I am fairly sure lansen will be willing to share with a purchaser. (No expert here but...) Although I am fairly certain they are using the correct non-proprietary packet format by the looks of it.

Expecting them by end of next week and will post my progress.

Excited!

weetmuts, 2020-08-12:
Cool! Hopefully,  it should work out of the box!

merbanan, 2020-08-12:
@anonymous-one https://github.com/merbanan/lans_tool can configure the sensors. I use the special configuration cable but I think it is very easy to use a regular usb-ttl converter. I'll try when I get the time.

anonymous-one, 2020-08-16:
If you get a few minutes of free time to check if the usb-ttl works that would be awesome! Pin out would be great too although I should be able to figure it out.
 
The defaults they send them with are acceptable, but it would be nice to lower the send interval to 60 seconds or 120 seconds. I know battery life will suffer but they are rating it about 12 years @ 5 mins... So I should have no trouble getting 2-3 years out of a 120 second or lower interval. Plus I would probably end up turning the encryption off which might save a cpu cycles on each send. I could also then inject my initial pulse count and have it count from present state rather then doing it via software.

---

Such a small thing but I am super excited about this. This is the final piece to my home-auto setup. 5 years in, I have been unable to get my main power meter online as the meter is outside (off property, common area) and the fuse box is just jam-packed so the 'clip' style sensors are out of the question unless I start drilling and placing things outside of the box. As an added bonus I get to bring my gas meter into the fold as well as it has a pulse count led, although I was not too interested in this as our gas bills are fairly reasonable.

anonymous-one, 2020-08-20:
So they arrived today.

Lansen was supposed to send me the keys on Monday but have yet to do so.

I did not email them reminding until last night though once I saw the package had crossed borders into my coutry.

So I am now waiting for the keys. A bit frustrating.

Wmbusmeters is picking something up though:

```
(meter) MeterGas: probably not for me since received media 0x02 does not match
(wmbus) decrypted content failed check, did you use the correct decryption key? Ignoring telegram.
```

@merbanan any idea what the default key could be? I wonder if they are unique to the devices. Each has a qr-code on the side but I scanned it and its just MODEL#.REVISION... Nothing else.

merbanan, 2020-08-20:
I tried yesterday to connect to a sensor via a regular usb-ttl cable but that did not work well. Out of 100 tries I got 1 complete answer from a FTDI clone chip. For some reason the serial link is very flaky. The provider cable works well so my guess is that the device communicates over 1.8V instead of 3.3V. I'll try to verify that later today.

With regard to the key it is unique to each sensor. But for me it was possible to disable the encryption without unlocking the device with the key. So if you have a working cable if might be possible to change settings.

anonymous-one, 2020-08-20:
> I tried yesterday to connect to a sensor via a regular usb-ttl cable but that did not work well. Out of 100 tries I got 1 complete answer from a FTDI clone chip. For some reason the serial link is [...]

@merbanan Ahh, thanks for giving it a shot. I am sure they will send the keys as promised, just have to wait for the gears to turn there... A bit impatient that's all hehe.

merbanan, 2020-08-20:
```
Pinout

--------------------------------------------
Battery compartment
--------------------------------------------
                     1 2 3
                     o o o
O O O O O O          o o o  CONFIG
A B C D E F          4 5 6

A - N/C
B - GND1
C - RX
D - TX
E - VCC (3.3V)
F - GND1

1 - VCC maybe?
2 - RX
3 - GND1
4 - TX
5 - GND2
6 - VCC maybe ?
```

The manufacturer cable is a FTDI TTL-232RG 3.3V cable with a custom connector. The manufacturer connector uses the 3x2 pogo pin header. This connector has 2 grounds (GND1 and GND2) they are connected at GND in the usb connector (this sort of aligns with an application note). If one connect a cable though the A-F pin header the connection is not working. Maybe one must also connect GND2 to ground and not leave it floating. GND1 and GND2 are connected to the GND pins 3 and 5 on the atmega chip.

merbanan, 2020-08-20:
@anonymous-one I got it working, with a needle it should be possible to get a semi stable connection to the sensor.

anonymous-one, 2021-03-08:
Just a small update after using this sensor for quite a few months:

Works great, one caveat:

The lansen spec says at a 5 minute interval, the battery is said to last 10+ years. I set mine to 1 minute intervals thinking 10 years / 5, swapping the battery every 2 years is a-ok by me. Not so... At a 1 minute interval the battery lasted about 3 months. It took me forever to update the setting to 5 minutes since it involves quite a bit of monkey work (read posts above)...

So, I would strongly suggest setting these to 5 minute intervals or more...

Maybe something specific to my case... Who knows.

weetmuts, 2021-03-08:
Interesting! Thanks for the info!