wmbusmeters/wmbusmeters #142: Kamstrup multical 602 - no output from rtl_wmbus

janchrillesen, 2020-07-29:
First of all - I am not sure if this issue should be filed here, or for rtl_wmbus... I have a Kamstrup multical 602 heat meter.

I can receive telegrams using my RTL-SDR dongle and rtl_433:

# rtl_433 -f 868.95M -G 4
rtl_433 version 20.02-108-ge64d3e5 branch master at 202007291509 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

	Use -G for testing only. Enable protocols with -R if you really need them.

Registered 156 out of 162 device decoding protocols [ 1-4 6-8 10-17 19-26 29-64 67-162 ]
rtl_433: warning: 106 "Wireless M-Bus, Mode R, 4.8kbps (-f 868330000)" does not support CSV output
rtl_433: warning: 107 "Wireless M-Bus, Mode F, 2.4kbps" does not support CSV output
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.950MHz.

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-29 23:17:48
model     : Wireless-MBus Mode      : C            Manufacturer: KAM         ID        : 78343597
Version   : 28           Device Type: 0x0C         Device Type String: Heat  Control   : 0x44          Data Length: 62           Data      : 41442d2c973534781c0c8d207964b1b0214f98f4a811a260ab07b4a593294c69c89e277b0d7668b1846eb06863ba9496c4a250a8f49a50eca639a56477da
Integrity : CRC

As I would like to decode the telegrams with wmbusmeters I tried to run:

# ./wmbusmeters --debug --t1  rtlwmbus
(wmbusmeters) version: 0.9.34
(config) using device: rtlwmbus
(config) number of meters: 0
(detect) "rtlwmbus" ""
(detect) driver: rtlwmbus
(rtlwmbus) using command: rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(config) listen to link modes: any
(serial) waiting for stop

After waiting for more than 10 minutes, nothing happens. rtl_433 receives telegrams every 4-5 minutes.

Any ideas on what might be the issue here?

weetmuts, 2020-07-30:
Hmm, what happens if you run rtl_wmbus stand-alone like this:
rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus

Can you use my fork of rtl_wmbus? https://github.com/weetmuts/rtl-wmbus

janchrillesen, 2020-07-30:
Absolutely nothing happens. I am already using your fork, since it fixes compile time warnings

`# rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | rtl-wmbus/build/rtl_wmbus
`
This has been running for an hour

weetmuts, 2020-07-30:
Can you try adjusting the frequency 868.9M 868.85M 869.0M 869.5M etc...

janchrillesen, 2020-07-30:
That definately made a difference!

```
# rtl_sdr -f 868.9M -s 1.6e6 - 2>/dev/null | rtl-wmbus/build/rtl_wmbus
C1;1;1;2020-07-30 11:07:52.000;127;122;78343597;0x3f442d2c973534781c0c8d208030e5b021e884fa772e33abf41f0eaba9fbd8b2e4de8217214fafe00f565bbfa97a9ebc5186740907eb4a3404b0a39fa21fcfd3
C1;1;1;2020-07-30 11:09:25.000;127;110;78343597;0x3f442d2c973534781c0c8d208154e5b0210116c4dde441d6f9b8f6ff4f75fd1d5ad375f6fd14b7e806d2932307893f9a149da48c39f16a4e047a7d20b0e844e9
C1;1;1;2020-07-30 11:10:58.000;140;145;78343597;0x3f442d2c973534781c0c8d208268e5b02142ebfe25fccbdb990c6d21f06a261a2eefc36f0ff8e9cfce619f88a325880fdc51bc25236a8740a2c10b47f14874be
C1;1;1;2020-07-30 11:12:32.000;137;144;78343597;0x3f442d2c973534781c0c8d20838ce5b021e1ab4df39e80aaca0b26f9a38ddb9e132464dc9e370fa34bd4ed03457bbcc46d662bad07d3a2e897a352161781376d
C1;1;1;2020-07-30 11:14:05.000;131;134;78343597;0x3f442d2c973534781c0c8d208490e5b0219d32f49c27da2788983c967296a8308ebc9eab2346a6c3cd080c13953c3d049d0978ee7c618cb08337f026cc443d4f
C1;1;1;2020-07-30 11:15:38.000;142;113;78343597;0x5c442d2c973534781c0c8d2085b4e5b021d129a4ffdde2d7706c642ee518bfbed147cd352064075162591f099f01c6a8fb89fa8e6073ce4715ffcf7c8f87747fc93699f6f61133f9878e6fd111dc46f22396b5feb8331849232822a37e
C1;1;1;2020-07-30 11:17:11.000;128;132;78343597;0x3f442d2c973534781c0c8d2086c8e5b0214e4eedd7b4af67c3f5812eccd08448ec055dae3faa34eed57461e045b94ae1cfd531fede6c1ea1e93b7e400f52b06f
C1;1;1;2020-07-30 11:18:44.000;131;150;78343597;0x3f442d2c973534781c0c8d2087ece5b0217f46035d1fb40ffe8f176fb13bad654dfc623eb42861e4dd9e3962a4853674cc5c994e986131bb23b953734222ca4a
```

Next step would be to try to decode the telegrams. I assume the Multical 302 is somewhat similar to my 602. How do I find out if I need a key, or if there's an issue parsing the telegram? The serial number above does match what's printed on my meter

```
# ./wmbusmeters rtlwmbus:868.9M Fjernvarme multical302:c1 78343597 NOKEY
(meter) Fjernvarme: probably not for me since received media 0x0c does not match
(wmbus) payload crc error!
(meter) Fjernvarme: probably not for me since received media 0x0c does not match
(wmbus) payload crc error!

```

weetmuts, 2020-07-30:
Excellent! Do you have the key for your meter? We need it to get the content of the meter so that we can add proper support for it.

weetmuts, 2020-07-30:
You can see that the telegram is encrypted when you run with --debug, 
the line (wmbus) 0d: 30e5b021 sn (AES_CTR) 
and the crc error indicates that it has not succesfully decoded the telegram.

weetmuts, 2020-07-30:
Use 

./wmbusmeters --debug --t1  rtlwmbus:868.9M

to adjust the frequencey that wmbusmeters will use when invoking rtl_wmbus.

janchrillesen, 2020-08-01:
Thank you very much for your support. As my original question has been answered I am closing this issue. I have requested the KEM file from my utility company. If/when I get it, and the multical302 decoder does not support Multical 602 I'll open a new issue

closed 2020-08-01 by janchrillesen