wmbusmeters/wmbusmeters #144: Bash, wmbusmeters output to variable always empty.

tstepnia, 2020-08-02:
#!/bin/bash
mysql_insert=$(wmbusmeters --selectfields=total_m3 --oneshot --t1 rtlwmbus mytap izar 40f43f82  NOKEY)
echo $mysql_insert 

When run as script from shell - returns nothing.

Command in () works fine when run directly from shell.

Any clue?

BIBOLV, 2020-08-02:
I can confirm that your script should work:
`~# bash test.sh
3.843
~# cat test.sh
#!/bin/bash
mysql_insert=$(/usr/bin/wmbusmeters --selectfields=total_m3 --oneshot /root/wmbusmeters/simulations/simulation_apas.txt sim apator162 20202020 NOKEY)
echo $mysql_insert
~#
`
Try adding full path to wmbusmeters binary, maybe it will help.

tstepnia, 2020-08-08:
This is exactly what I get with or without full path for version - wmbusmeters-0.9.32-1.fc32.x86_64

[user@host tmp]$ cat w
#!/bin/bash
        echo "before insert"    
        mysql_insert=$(/usr/bin/wmbusmeters --selectfields=total_m3 --oneshot --t1 rtlwmbus mytap izar 40f43f82  NOKEY)
        echo $mysql_insert

[user@host tmp]$ ./w
before insert

[user@host tmp]$

weetmuts, 2020-08-08:
Can you add --debug to your wmbusmeters command and run the script again?

weetmuts, 2020-08-09:
Add both - - debug --usestderr

weetmuts, 2020-08-09:
Ah, found the problem! Tricky one.

wmbusmeters always returned a non-zero value (ie a failure) when used with rtlwmbus and --oneshot.
It worked fine with the other dongles....ie then wmbusmeters returned 0.
The failure caused the bash script to terminate.

The problem was that I terminated the rtlmbus subshell with SIGINT to pid-group 0. 
This was a temporary fix, according to my comment.
Since I did not understand how to terminate the subshell properly.
However, this caused wmbusmeters to exit with a failure since the whole process terminated
as if you had pressed ctrl-c.

The right fix was a one-liner, instead of
-    int rc = kill(0, SIGINT);
+    int rc = kill(pid, SIGTERM);

Send SIGTERM to the subshell pid. Should have figured that one out earlier. :-)

Fix pushed to master. Please test!

weetmuts, 2020-08-10:
Aha, I knew it wasn't that easy. The SIGTERM terminate the subshell, but not the rtl_sdr|rtl_wmbus processes.
If you tried to start wmbusmeters again it will fail.

I have now pushed the proper fix. When spawning the subshell, use setpgid(0,0) to make it its own process group leader.
Now we can send SIGTERM to -pid (ie negative subshell pid) and it will send SIGTERM to all members of the process group.

Seems to work properly. Please test!

weetmuts, 2020-09-04:
It works for me now! Closing this!

closed 2020-09-04 by weetmuts