wmbusmeters/wmbusmeters #145: Getting meter key from preconfigured wmbus->mbus converter.

Nikodemski2002, 2020-08-10:
Hi. Is there any way I can get the meter key from Kamstrup RF module wM-BUS to M-BUS converter, and M-BUS master hat for RPi or the im871a-USB.

I have written an email to my provider with a question to send me these keys, but I haven't received any answer for four days

I mentioned this problem in the #140 case.

Regards.

weetmuts, 2020-08-10:
Yes, I need to add wired mbus support to wmbusmeters. Its on my todo-list.
Do you have an mbus -> serial -> usb adapter in mind?

Nikodemski2002, 2020-08-11:
No. I mean the wireless M-BUS to wired M-BUS Kamstrup donglt. It looks like this:
[image]

And I have two options to read data from it the M-BUS master HAT for RPi or im871a-usb.

weetmuts, 2020-08-12:
Cool, I have also ordered an M-BUS master HAT for RPi, to add mbus support to wmbusmeters.

weetmuts, 2020-08-12:
But I will need your help to debug the mbus communication with your particular wmbus-dongle.

weetmuts, 2020-08-12:
Aha, now I understand your question! I do not know if it is possible to extract the key from the dongle......
I somehow doubt that it is possible....but who knows.

merbanan, 2020-08-12:
If you open the hardware there might be some clues on how to extract it.

Nikodemski2002, 2020-08-12:
I read that you can read the AES key from this dongle, but only with Kamstrup USB-tool. But I already received the KEM file and decrypted it with [python script](https://gist.github.com/mbursa/caa654a01b9e804ad44d1f00208a2490).

Thanks for your interest.
Best regards.

closed 2020-08-12 by Nikodemski2002

merbanan, 2020-08-12:
@Nikodemski2002 so how did you get them to send you the KEM file? Is the procedure that you need to buy this adapter and then request the KEM file that you then use with the Kamstrup USB-tool?

Nikodemski2002, 2020-08-12:
@merbanan I just mailed them that I will not use the provided converter and asked them to send me the KEM file for certain water meters. Next time I will just register every water meter on Kastrup customer panel so I will have simple access to encryption keys.