wmbusmeters/wmbusmeters #146: Push data to influxdb

Nikodemski2002, 2020-08-12:
Hi. I have launched an influxdb with the grafana visualizer, on my RPI with wmbusmeters. I have created an empty database called "multical". As I read through the [influxdb docs](https://docs.influxdata.com/influxdb/v1.3/guides/writing_data/) for pushing data to it, I have set up this command in the wmbusmeters.conf:

`shell=echo curl -i -XPOST 'http://localhost:8086/write?db=' --data-binary 'total_m3'`

But when I check the database it doesn't show anything (Grafana also empty).

So here is my question, what am I doing wrong. I assume that I misunderstood the method of sending, but I been at it for 6 hours and still no results. I tried also using the tutorial that uses the prometheus as a database and remake it for my purpose, but also didn't work.

Regards.

weetmuts, 2020-08-13:
The data to be sent is passed to the shell command using environment variables.
You have to specify $METER_TOTAL_M3 somewhere on the command line.
This environment variable contains your total cubic meters.

If you have a Multical21 metre and you type:
wmbusmeters --listenvs simulation_dummy.txt Water multical21 12345678 NOKEY
it will list the available env variables. METER_JSON contains all the below
in a json structure.

METER_JSON
METER_TYPE
METER_NAME
METER_ID
METER_TOTAL_M3
METER_TARGET_M3
METER_MAX_FLOW_M3H
METER_FLOW_TEMPERATURE_C
METER_EXTERNAL_TEMPERATURE_C
METER_CURRENT_STATUS
METER_TIME_DRY
METER_TIME_REVERSED
METER_TIME_LEAKING
METER_TIME_BURSTING
METER_TIMESTAMP

BIBOLV, 2020-08-15:
First of, do not use echo as it will just echo to STDOUT what you have provided, as shown in example below:
```bash
~]# wmbusmeters --shell='echo /bin/cat /etc/passwd | wc -l' /root/wmbusmeters/simulations/simulation_t1_and_c1.txt sim multical21 76348799 NOKEY
1
```
vs
```bash
~]# wmbusmeters --shell='/bin/cat /etc/passwd | wc -l' /root/wmbusmeters/simulations/simulation_t1_and_c1.txt sim multical21 76348799 NOKEY
33
```

Second thing is that I would sugest to try to write data from console, before putting cmd in wmbusmeters, just to be sure that it works, so if you plan to insert just consumtion and meterid in multical db then data consists of the measurement m3, the tag keys meter_id with the tag values 76348799, the field key value with a field value of 6.001:
```bash
~]# curl -i -XPOST 'http://localhost:8086/write?db=multical' --data-binary 'm3,meter_id=76348799 value=6.001'
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: baa10904-df26-11ea-8f28-fcaa14ebd93c
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.7.10
X-Request-Id: baa10904-df26-11ea-8f28-fcaa14ebd93c
Date: Sat, 15 Aug 2020 18:40:02 GMT
```
and verify that data has been inserted
```bash
~]# curl -G 'http://localhost:8086/query?pretty=true' --data-urlencode "db=multical" --data-urlencode "q=SELECT * from m3"
{
    "results": [
        {
            "statement_id": 0,
            "series": [
                {
                    "name": "m3",
                    "columns": [
                        "time",
                        "meter_id",
                        "value"
                    ],
                    "values": [
                        [
                            "2020-08-15T18:40:02.108908072Z",
                            "76348799",
                            6.001
                        ]
                    ]
                }
            ]
        }
    ]
}
```
then you are ready to use it in wmbusmeters and you need add variables as @weetmuts pointed out
```bash
~]# wmbusmeters --shell="/usr/bin/curl -i -XPOST 'http://localhost:8086/write?db=multical' --data-binary 'm3,meter_id='\$METER_ID' value='\$METER_TOTAL_M3''" /root/wmbusmeters/simulations/simulation_t1_and_c1.txt sim multical21 76348799 NOKEY
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: 11535921-df28-11ea-8f6b-fcaa14ebd93c
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.7.10
X-Request-Id: 11535921-df28-11ea-8f6b-fcaa14ebd93c
Date: Sat, 15 Aug 2020 18:49:37 GMT
```
and verify that data has been inserted
```bash
~]# curl -G 'http://localhost:8086/query?pretty=true' --data-urlencode "db=multical" --data-urlencode "q=SELECT * from m3"
{
    "results": [
        {
            "statement_id": 0,
            "series": [
                {
                    "name": "m3",
                    "columns": [
                        "time",
                        "meter_id",
                        "value"
                    ],
                    "values": [
                        [
                            "2020-08-15T18:40:02.108908072Z",
                            "76348799",
                            6.001
                        ],
                        [
                            "2020-08-15T18:49:37.058308484Z",
                            "76348799",
                            6.408
                        ]
                    ]
                }
            ]
        }
    ]
}
```

so in the end you would end up with shell line in config file something similar to this
```
shell=/usr/bin/curl -i -XPOST 'http://localhost:8086/write?db=multical' --data-binary 'm3,meter_id='\$METER_ID' value='\$METER_TOTAL_M3''
```

weetmuts, 2020-09-04:
Ping @Nikodemski2002 !

Did you solve it?

weetmuts, 2020-09-11:
Closing this for now.

closed 2020-09-11 by weetmuts