wmbusmeters/wmbusmeters #149: 4a3105e amb8465 hangs after "get config"

HenryNe, 2020-08-19:
amb8465 is not working. Start command and output:
```
# killall wmbusmeters; sleep 1; wmbusmeters --listento=c1,t1 --logtelegrams --debug --verbose /dev/ttyS2
(wmbusmeters) version: 0.9.34-8232
(config) using device: /dev/ttyS2
(config) number of meters: 0
(detect) "/dev/ttyS2" ""
(detect) is_tty=1 is_stdin=0 is_file=0
(serial) before config: /dev/ttyS2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CD|RING|DSR)
(serial) after config:  /dev/ttyS2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CD|RING|DSR)
(serialtty) opened /dev/ttyS2
(serial) received binary ""
(amb8465) are you there?
(serial /dev/ttyS2) sent "FF0B00F4"
(serial) received binary "FF8B042701CE25BD"
(serialtty) closed /dev/ttyS2
(amb8465) on /dev/ttyS2
(serial) registered regular callback 0 DEVICE_AMB8465 every 60 seconds
(serial) before config: /dev/ttyS2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CD|RING|DSR)
(serial) after config:  /dev/ttyS2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CD|RING|DSR)
(serialtty) opened /dev/ttyS2
(amb8465) get config
(serial /dev/ttyS2) sent "FF0A02008077"
```
Than it hangs. I missing the answer from config command.
After CTRL-C some more output is printed:
```
^C
(serial) stopping manager
(config) all possible link modes that the meters might transmit on: none
No meters configured. Printing id:s of all telegrams heard!

(wmbus) set timeout DEVICE_AMB8465 to "0" with expected activity ""
(amb8465) set link mode 09
(serial /dev/ttyS2) sent "FF040109F3000000"
(config) listen to link modes: c1,t1
(serial) waiting for stop
(serial) event loop stopped!
(serial) closing 1 devices
(serial) no devices working emergency exit!
(serial) no devices working emergency exit!
(serialtty) closed /dev/ttyS2
```
Same with specific device:
```
# wmbusmeters --listento=c1,t1 --logtelegrams --debug --verbose /dev/ttyS2:amb8465
(wmbusmeters) version: 0.9.34-8232
(config) using device: /dev/ttyS2
(config) with: amb8465
(config) number of meters: 0
(detect) "/dev/ttyS2" "amb8465"
(detect) is_tty=1 is_stdin=0 is_file=0
(amb8465) on /dev/ttyS2
(serial) registered regular callback 0 DEVICE_AMB8465 every 60 seconds
(serial) before config: /dev/ttyS2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CD|RING|DSR)
(serial) after config:  /dev/ttyS2 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CD|RING|DSR)
(serialtty) opened /dev/ttyS2
(amb8465) get config
(serial /dev/ttyS2) sent "FF0A02008077"
```
It hangs at same point.

Version:
```
# wmbusmeters --version
wmbusmeters: 0.9.34-8232
aa24eac099dd67f320d0a157c6fb61e5950604d4
```

weetmuts, 2020-08-19:
Fix pushed. Please test.

weetmuts, 2020-08-19:
A better fix pushed, even though the previous fix, fixed your problem.

There is ongoing work with the init/reset code. I want to share the code between the initial boot of the dongle
and the reset that is triggered when an alarm goes off. It turns out that getConfiguration works nicely in the
reset but not in the init. (Since the eventLoop has not started running yet.)

Starting the eventLoop early worked for the serial dongles, but caused all sorts of problem
when feeding wmbusmeters from stdin. So I chose to remove the unnecessary getConfiguration
in the amber deviceReset function instead.

HenryNe, 2020-08-20:
Thanks! Serial device is working again.

wmbusmeters: 0.9.34-35-g78a353d
78a353de5967585edbcad1d3d10c60d4f6fa295f

closed 2020-08-20 by HenryNe