wmbusmeters/wmbusmeters #155: Support for IZAR RS 868

iwikus, 2020-08-25:
Hi,
I'm trying to decode messages from 'IZAR RS 868'. I can see output from rtl_433 
<pre>
time      : 2020-08-25 23:21:19
model     : Wireless-MBus Mode      : T            Manufacturer: HYD         ID        : 65530791
Version   : 56           Device Type: 0x63         Device Type String:       Control   : 0x44          Data Length: 31           Data      : 1e4424238b07536538637aac0010f4909dabf1599f676c7418544e4d6b2268
Integrity : CRC
Control Info: 0x7A       Access number: 0xAC       Device Type: 0x00         Configuration Word: 0xF410 Payload Encrypted: 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-25 23:21:24
model     : Wireless-MBus Mode      : T            Manufacturer: HYD         ID        : 65280791
Version   : 56           Device Type: 0x63         Device Type String:       Control   : 0x44          Data Length: 31           Data      : 1e4424238b07286538637a2c0010f0fde86cf9a2645ee8bdec7dde178a1419
Integrity : CRC
Control Info: 0x7A       Access number: 0x2C       Device Type: 0x00         Configuration Word: 0xF010
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-25 23:21:37
model     : Wireless-MBus Mode      : T            Manufacturer: HYD         ID        : 65510791
Version   : 56           Device Type: 0x63         Device Type String:       Control   : 0x44          Data Length: 31           Data      : 1e4424238b07516538637a94001078924a4c8d155781d1979f84ab90f6e687
Integrity : CRC
Control Info: 0x7A       Access number: 0x94       Device Type: 0x00         Configuration Word: 0x7810
</pre>

It reports manufacturer is "HYD"

Running `./wmbusmeters --debug rtl433  --listento=t1 izar '*' NOKEY` is reporting unexpected end of data and failed decoding:

<pre>
(wmbus) parseDLL @0 31
(meter) --listento=t1: for me? 6493078b
(meter) --listento=t1: probably not for me since manufacturer differs
(meter) --listento=t1: probably not for me since received media 0x63 does not match
(meter) --listento=t1: yes for me
(meter) --listento=t1 6493078b "1E4424238B07936438637A3100101975FA00E49E391ED5409CD5E13037035E"
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(dvparser) warning: unexpected end of data
(dvparser) warning: unexpected end of data (vif expected)
(dvparser) found new format "75FA00D5409CD5E1305E" with hash e1f2, remembering!
(telegram) DLL L=1e C=44 (from meter SND_NR) M=2324 (HYD) A=6493078b VER=38 TYPE=63 (Unknown)
(telegram) TPL CI=7a ACC=31 STS=00 CFG=1910 (SPECIFIC_16_31)
(izar) log "telegram=|1E4424238B07936438637A31001019|75FA00E49E391ED5409CD5E13037035E|+4"
(izar) Decoded PRIOS data: 
(izar) Decoding PRIOS data failed. Ignoring telegram.
(izar) 00: 1e length (30 bytes)
(izar) 01: 44 dll-c (from meter SND_NR)
(izar) 02: 2423 dll-mfct (HYD)
(izar) 04: 8b079364 dll-id (6493078b)
(izar) 08: 38 dll-version
(izar) 09: 63 dll-type (Unknown)
(izar) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(izar) 0b: 31 tpl-acc-field
(izar) 0c: 00 tpl-sts-field
(izar) 0d: 1019 tpl-cfg 1910 (SPECIFIC_16_31 )
(izar) 0f: 75 dif (32 Bit Real Value during error state storagenr=1)
(izar) 10: FA vif (?)
(izar) 11: 00 vife (?)
(izar) 12: E49E391E
(izar) 16: D5 dif (32 Bit Real Maximum value storagenr=1)
(izar) 17: 40 dife (subunit=1 tariff=0 storagenr=1)
(izar) 18: 9C vif (Mass 10¹ kg)
(izar) 19: D5 vife (duration of limit exceed last lower  is 1)
(izar) 1a: E1 vife (duration of a,b first  is 1)
(izar) 1b: 30 vife (per kWh)
(izar) 1c: 3703
(izar) 1e: 5E dif (12 digit BCD Maximum value storagenr=1)
--listento=t1   6493078b        0 m3    0 m3    0-00-00 0 y     general_alarm   leakage,sensor_fraud    2020-08-25 23:26.57
</pre>

Is this some different protocol or key issue?

iwikus, 2020-08-25:
I'm thinking if HYD could mean I should use hydrus, I have unexpected end of data again and unexpected meter version:
<pre>
rtl433) checkRTL433Frame "2020-08-25 23:44:11,,,Wireless-MBus,,65510791,,,,CRC,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,1e4424238b07516538637a9e00107892433ec0f5a35246417025a45c5b5702,,,,,99,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,T,56,,122,158,0,30736,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,<0A>"
(wmbus) parseDLL @0 31
(meter) --listento=t1: for me? 6551078b
(meter) --listento=t1: probably not for me since manufacturer differs
(meter) --listento=t1: probably not for me since received media 0x63 does not match
(meter) --listento=t1: yes for me
(meter) --listento=t1 6551078b "1E4424238B07516538637A9E00107892433EC0F5A35246417025A45C5B5702"
(wmbus) parseDLL @0 31
(wmbus) parseELL @10 21
(wmbus) parseAFL @10 21
(wmbus) parseTPL @10 21
(dvparser) warning: unexpected end of data
(dvparser) warning: unexpected end of data (vif expected)
(dvparser) found new format "92433EA35246A45C5B02" with hash 787e, remembering!
(telegram) DLL L=1e C=44 (from meter SND_NR) M=2324 (HYD) A=6551078b VER=38 TYPE=63 (Unknown)
(telegram) TPL CI=7a ACC=9e STS=00 CFG=7810 (accessibility synchronous SPECIFIC_16_31)
(hydrus) unexpected meter version 0x38 !
(hydrus) log "telegram=|1E4424238B07516538637A9E001078|92433EC0F5A35246417025A45C5B5702|+278"
(hydrus) 00: 1e length (30 bytes)
(hydrus) 01: 44 dll-c (from meter SND_NR)
(hydrus) 02: 2423 dll-mfct (HYD)
(hydrus) 04: 8b075165 dll-id (6551078b)
(hydrus) 08: 38 dll-version
(hydrus) 09: 63 dll-type (Unknown)
(hydrus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(hydrus) 0b: 9e tpl-acc-field
(hydrus) 0c: 00 tpl-sts-field
(hydrus) 0d: 1078 tpl-cfg 7810 (accessibility synchronous SPECIFIC_16_31 )
(hydrus) 0f: 92 dif (16 Bit Integer/Binary Maximum value)
(hydrus) 10: 43 dife (subunit=1 tariff=0 storagenr=6)
(hydrus) 11: 3E vif (Volume flow m³/h)
(hydrus) 12: C0F5
(hydrus) 14: A3 dif (24 Bit Integer/Binary Minimum value)
(hydrus) 15: 52 dife (subunit=1 tariff=1 storagenr=4)
(hydrus) 16: 46 vif (Volume flow ext. 10⁻¹ m³/min)
(hydrus) 17: 417025
(hydrus) 1a: A4 dif (32 Bit Integer/Binary Minimum value)
(hydrus) 1b: 5C dife (subunit=1 tariff=1 storagenr=24)
(hydrus) 1c: 5B vif (Flow temperature °C)
(hydrus) 1d: 57
(hydrus) 1e: 02 dif (16 Bit Integer/Binary Instantaneous value)
--listento=t1   6551078b        0 m3    0 m3/h  2020-08-25 23:44.11
</pre>

weetmuts, 2020-08-26:
Is it an older meter? Is it an R3 or R4 version? Can you post a picture?
Note that after the wmbus device name (rtl433 in your case) there must be no other options.

So instead of:
wmbusmeters --debug rtl433 --listento=t1 izar '*' NOKEY
you should write:
wmbusmeters --debug --listento=t1 rtl433 izar '*' NOKEY

not that it helps right now though.

iwikus, 2020-08-26:
I think new, on the box is 2019. I don't known revision. Maybe figure it out from part number? See pictures below.
[image]
[image]
[image]

weetmuts, 2020-09-04:
Closing this issue, since adding izar R3 will he handled in this issue #127

closed 2020-09-04 by weetmuts