wmbusmeters/wmbusmeters #159: FlowIQ3100 - (wmbus) payload crc error!

jinx17, 2020-09-03:
Hi,
I try to read FlowIQ3100. Command started successfully, and response received. But I think there is a problem with CRC checking. Can you help me and point to the right direction. I use RPi4 and im871a USB on DietPi OS (stripped down raspbian).

**command sent:** wmbusmeters --debug /dev/ttyUSB0:im871a MyTapWater flowiq3100:c1 68776655 **CORRECTKEYHERE...**
**debug output:**
```
(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0
(config) with: im871a
(config) number of meters: 1
(detect) "/dev/ttyUSB0" "im871a"
(detect) is_tty=1 is_stdin=0 is_file=0
(im871a) on /dev/ttyUSB0
(serial) registered regular callback 0 DEVICE_IM871A every 60 seconds
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0
(config) meter flowiq3100 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "MyTapWater" "flowiq3100" "68776655" encrypted
(wmbus) set timeout DEVICE_IM871A to "0" with expected activity ""
(im871a) set link mode 06
(serial /dev/ttyUSB0) sent "A5010306000206300101"
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) received binary "A5810617FF000600B32547201000010001FF070232000101000025B7E8"
(im871a) checkIM871AFrame "A5810617FF000600B32547201000010001FF070232000101000025B7E8"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 e8b7 expected e8b7
(im871a) received full frame
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 06
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00102047
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) listen to link modes: c1
(serial) waiting for stop
(serial) received binary "A5E20322442D2C556677681D168D281E833819214FD34C3EEF47F8C969526A48232A5BAD8CA378C60000001163"
(im871a) checkIM871AFrame "A5E20322442D2C556677681D168D281E833819214FD34C3EEF47F8C969526A48232A5BAD8CA378C60000001163"
(im871a) has_timestamp=1 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) timestamp 0000c678
(im871a) rssi 00
(im871a) got crc16 6311 expected 6311
(im871a) received full frame
(wmbus) parseDLL @0 35
(meter) MyTapWater: for me? 68776655
(meter) MyTapWater: yes for me
(meter) MyTapWater 68776655 "22442D2C556677681D168D281E833819214FD34C3EEF47F8C969526A48232A5BAD8CA3"
(wmbus) parseDLL @0 35
(wmbus) parseELL @10 25
(ELL) decrypting "4FD34C3EEF47F8C969526A48232A5BAD8CA3"
(ELL) IV 2D2C556677681D162883381921000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "F4FDF957BECEB907DC70654D660547C3"
(ELL) block 1 block_size 2 offset 16
(ELL) decrypted "9ACA"
(ELL) decrypted "F4FDF957BECEB907DC70654D660547C39ACA"
(wmbus) payload crc error!
(wmbus) telegram ignored by all configured **meters!**
```

weetmuts, 2020-09-04:
Usually payload crc error means that you do not have the right key.
See if you can verify the key you have. Are you listening to the right meter?

If it still fails and you really really think you have the right key,
then you can email me the key and I can check if I can debug the
decoding of the telegram you provided here.

jinx17, 2020-10-12:
Yes. You are correct. It was a problem with the correct key.
Thank you.

closed 2020-10-12 by jinx17