wmbusmeters/wmbusmeters #160: Expanded support for Diehl Hydrus

caydenm, 2020-09-04:
I have a Diehl Hydrus hot water meter and I can't seem to get it working with either HYDRUS of IZAR types. 

An example packet captured using "rtl_433 -R 104 -s 1200k -f 434570k" 

time      : 2020-09-04 15:46:22
model     : Wireless-MBus
Mode      : T
Manufacturer: HYD
ID        : 51283361
Version   : 36
Device Type: 0x07
Device Type String: Water
Control   : 0x44
Data Length: 79
Data      : 4e4424236133285124077a4f004005238a80f3efcf368039e28fe83bc9e914e3f0793f6e98b6c5cbf310721c77999fde12e3398e4a4a5b20eb823ddd1c1fed787dd930c19a1cb344d1fedc3a6cf11a
Integrity : CRC
Control Info: 0x7A
Access number: 0x4F
Device Type: 0x00
Configuration Word: 0x0540
Payload Encrypted: 1

A default key of "HYDRUS_25" is used for these models, however I get an error that this is an invalid key when attempting to use it. 

Any help would be appreciated!

weetmuts, 2020-09-04:
Do you have a photo of this kind of meter? I am curious if it is the Sappel/Diehl IZAR r3, which
we cannot currently decode. There are pictures of the izar r3 in this issue #155 (now closed and merged into #127)

If it is not the same meter, then we have to identify exactly what kind of meter this is.

caydenm, 2020-09-04:
It looks to be this model, 434 + L-Bus variant:
https://www.diehl.com/metering/en/portfolio/water-metering/water-metering-product/hydrus/63697/

weetmuts, 2020-09-04:
Thanks, it is definitely not an IZAR clipon/meter.

weetmuts, 2020-09-04:
The key for a wmbus meter is a hex string like: 00112233445566778899AABBCCDDEEFF
representing 128 bits.

If you can figure out how they expect you to translate HYDRUS_25 into such a hex string,
then we can get further. Right now I cannot do much since the telegram is encrypted with AES
and no further progress can be done without the proper key.

weetmuts, 2020-09-04:
The telegram has the exact same length as the other hydrus telegram. The headers
are also very similar, only the manufacturer and the version differs. 
(The older meters probably had the manufacturer HYD (Hydrometer) where
the newer meters have the manufacturer DME (DIEHL Metering).

I think it is likely that the hydrus driver will work when you have the right key.

caydenm, 2020-09-05:
@c-soft and @zalewma it looks like you managed to get meters supported by the IZAR@NET tools working with the default keys. Any insights on how you went from https://github.com/weetmuts/wmbusmeters/issues/20#issuecomment-487257660 to a working implementation?

caydenm, 2020-09-05:
After doing a bit more research it appears the default key is the same as the second default key in the IZAR file (51728910E66D83F8). This key appears to get repeated twice, however either the default key isn't used or there is some incompatibility between it and the current hydrus implementation.

caydenm, 2020-09-06:
Closing this out. It appears I had some trouble with the default rtl433 command line. Changing it to rtl433:"rtl_433 -s 1200k -f 434570k -F csv" allowed me to test this and verify that a non-default encryption key is used. 

Thanks for your help @weetmuts and for bulding an excellent tool.

closed 2020-09-06 by caydenm

klsystem1, 2024-11-15:
Is the diehl hydrus now integrated in rtl433?

klsystem1, 2026-01-30:
I do have rtl_433 installed on my raspberrypi, wmbusmeters can't cope with rtl sdr stick, with wmbusmeters auto I only get a 100% CPU problem on raspberrypi 3.

But I do get the data package with rtl_433, and it is not encrypted (encrypted : 0). So can anyone tell me how to decode the values from this?

oot:>rtl_433 -f 868.95M -F json | head

{"time" : "2026-01-30 07:23:52", "model" : "Wireless-MBus", "mode" : "T", "M" : "DME", "id" : 10105053,
"version" : 118, "type" : 7, "type_string" : "Water", "C" : 68, "data_length" : 63,
"data" : "3c44a5115350101076077aa7003005ade7e683204ab5b138ab57205b060b2c53ae60c937457330091f31461928764ba817c1a1a1b8d178d0fc5215fdbcfb3f4fc8",
"mic" : "CRC", "CI" : 122, "AC" : 167, "ST" : 0, "CW" : 1328, "payload_encrypted" : 0}

Thanks a lot.
Klaus