wmbusmeters/wmbusmeters #161: FLOWIQ2200 - got data but not sure whats to do next..help?

cicci44, 2020-09-04:
the goal is to push data to OPC UA server on a raspberry pi or domoticz.. 

i think i suceeded to read floqiq2200 using terminal
 sudo wmbusmeters --debug rtlwmbus:868.9M MyTapWater flowiq3100 10036452 NOKEY

But i feel kinda stupid, because i think i need a decryption key but cant figure out how to find it...
[image]

cicci44, 2020-09-05:
ok, i logged in into My Kamstrup, and requested EKS using serialnumber and device code but didnt suceed. i Guess i cant access the data.

weetmuts, 2020-09-08:
yes, you need a key.

weetmuts, 2020-09-08:
If its your own bought meter, the seller should be able to help you.
If it is the water utility company's meter, then you will have to ask them for the key.
Sometimes they do give it too you, but not always.

cicci44, 2020-09-08:
> 
> 
> If its your own bought meter, the seller should be able to help you.
> If it is the water utility company's meter, then you will have to ask them for the key.
> Sometimes they do give it too you, but not always.

i asked, got the kem.zip file and password, used utils/kem-import.py  
python kem-import.py -o outputfile 20200908_1317_selected_DownloadMeters.zip.kem 32143242332
i think i got the key, going to make a test soon 👍

weetmuts, 2020-09-26:
Great! Do you think you could send me a telegram, you add --logtelegrams to the command line
or add logtelegrams=true to the conf file.

With that telegram I can double check that the content of the FlowIQ2200 if it contains more than FlowIQ3100
and I can add an alias driver name flowiq2200 that points to flowiq3100.

cicci44, 2020-10-11:
> Great! Do you think you could send me a telegram, you add --logtelegrams to the command line
> or add logtelegrams=true to the conf file.
>
> With that telegram I can double check that the content of the [...]

sorry for late reply, here u go 
https://pastebin.com/wj5Zm2ks

weetmuts, 2020-10-25:
Thanks @cicci44 !

It seems like this telegram does contain several differences compared to flowiq3100.
I have now added proper support for flowiq2200 in wmbusmeters.
Please rerun your previous tests with the master branch and the flowiq2200 meter instead.
You will see a json that looks like this:
{"media":"cold water","meter":"flowiq2200","name":"Vatten","id":"10036452","total_m3":44.206,"target_m3":43.108,"target_datetime":"2020-10-01 00:00","current_flow_m3h":0,"max_flow_m3h":0.495,"min_flow_m3h":0,"min_flow_temperature_c":12,"max_flow_temperature_c":20,"external_temperature_c":19,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-10-25T18:49:36Z"}

weetmuts, 2020-11-15:
@cicci44 Does it work as expected?

cicci44, 2020-11-16:
> 
> 
> @cicci44 Does it work as expected?

yes, it with 3100 for 2-3weeks, after that it stopped working. (rpi fail) 

i think i got it working, i have to come back with info tomorrow. Måste sova också ;)

cicci44, 2020-11-17:
Seems to work fine 👍 
> (shell) arg "HOME=/home/pi mosquitto_pub -u USER -P PASSWORD -h IP -p PORT -t water -m "$METER_JSON""
> (shell) env "METER_JSON={"media":"cold water","meter":"flowiq2200","name":"MyTapWater","id":"10036452","total_m3":47.232,"target_m3":45.585,"target_datetime":"2020-11-01 00:00","current_flow_m3h":0,"max_flow_m3h":0.276,"min_flow_m3h":0,"min_flow_temperature_c":10,"max_flow_temperature_c":19,"external_temperature_c":19,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-11-17T19:02:53Z","device":"rtlwmbus[1234]","rssi_dbm":95}"
> (shell) env "METER_TYPE=flowiq2200"
> (shell) env "METER_NAME=MyTapWater"
> (shell) env "METER_ID=10036452"
> (shell) env "METER_TOTAL_M3=47.232"
> (shell) env "METER_TARGET_M3=45.585"
> (shell) env "METER_TARGET_DATETIME=2020-11-01 00:00"
> (shell) env "METER_CURRENT_FLOW_M3H=0"
> (shell) env "METER_MAX_FLOW_M3H=0.276"
> (shell) env "METER_MIN_FLOW_M3H=0"
> (shell) env "METER_MIN_FLOW_TEMPERATURE_C=10"
> (shell) env "METER_MAX_FLOW_TEMPERATURE_C=19"
> (shell) env "METER_EXTERNAL_TEMPERATURE_C=19"
> (shell) env "METER_CURRENT_STATUS="
> (shell) env "METER_TIME_DRY="
> (shell) env "METER_TIME_REVERSED="
> (shell) env "METER_TIME_LEAKING="
> (shell) env "METER_TIME_BURSTING="
> (shell) env "METER_TIMESTAMP=2020-11-17T19:02:53Z"
> (shell) waiting for child 22261 to complete.
> (shell) /bin/sh: return code 0
>

closed 2020-12-01 by weetmuts