wmbusmeters/wmbusmeters #163: install.sh does not set up user:group properly

pswiatki, 2020-09-14:
When running `make install ` the following is the output:

```
[pswiatki@sdr-linux wmbusmeters]$ sudo make install
Building 0.9.36
binaries: installed /usr/bin/wmbusmeters and /usr/sbin/wmbusmetersd
man page: installed /usr/share/man/man1/wmbusmeters.1.gz
useradd: group 'dialout' does not exist
user: added wmbusmeters
usermod: group 'wmbusmeters' does not exist
user: added pswiatki to group wmbusmeters

```
It is assumed the `dialout ` group exists and also that `wmbusmeters` is created outside of the build process (manually?).
OS is Arch Linux.

weetmuts, 2020-09-14:
Thanks for the information! 
The dialout group is used in Ubuntu for serial devices like /dev/ttyUSB0 etc.
These are assumed to be modems and the modem manager will first try to talk to them.
The wmbus dongles are serial devices but do not respond as modems and the modemmanager
will then ignore them.

If you belong to the dialout group, then you can always accces these serial devices.

The current wmbusmeter.master uses udev rules to reassign the ownership for the plugged in
device to the wmbusmeters group.  Unfortunately this does not work that well
since the wmbus dongles cannot be reliably detected based the usb vendor numbers. 
From the udev perspective the wmbus dongles (except one) are just recognized as a generic 
serial devices.

So in the branch NewDetectionCode no udev rules are used and both wmbusmeters and yourself must belong
to the dialout group instead. Thus wmbusmeters (irrespective if its running as a daemon or as your own user)
can connect to the serial devices and probe them directly, skipping the udev step.

From this post: https://bbs.archlinux.org/viewtopic.php?id=178552
it seems like the Arch linux equivalent to dialout is called uucp group.

It seems like in ubuntu, when you create a new user, there is a default group with the same name created.
But this might not be the case in Arch. Anyway the install.sh in NewDetectionCode does handle the group
creation before the user creation. 

If you can adjust the install manually for now, then I can ask you to try the NewDetectionCode when 
it is a bit more stable and I have had time to integrate the arch linux uucp instead of dialout fix.

pswiatki, 2020-09-14:
OK, thank you so much for your comment. I was more concerned about the user explicitly stated in the service file for systemd, rather than udev rules.  As it is stated (in the same `install.sh` script):

```
[Service]
Type=forking
PrivateTmp=yes
**User=wmbusmeters
Group=wmbusmeters**
Restart=always
RestartSec=1

```

weetmuts, 2020-09-14:
Yes, I want the daemon to run as a user with limited acces rights.

The wmbusmeters user/group should only be allowed to write to the /var/log/wmbusmeters directory
and talk to the wmbus dongles and execute commands, that send data using for example mqtt.

It is also convenient to use the wmbusmeters group to allow your own user
to read the wmbusmeters log files without having to sudo all the time.

Clearly the install script does not work for Arch I wil fix that.
If you have other suggestions and improvements please let me know.

pswiatki, 2020-09-14:
Sure, I will do just that when I find something.

weetmuts, 2020-09-26:
If you can test the branch NewDetectionCode, it has an updated install script, among many other things.
See if it works for you! See issue #169

weetmuts, 2020-10-25:
New install script in master (along with the rest of the NewDetectionCode, read CHANGES) 
Please test!

weetmuts, 2021-01-11:
Closing this for now.

closed 2021-01-11 by weetmuts