wmbusmeters/wmbusmeters #164: iM871A-USB error reading device

matze-dev, 2020-09-17:
Hi all,
I plug in an USB device: iM871A-USB to read out values from Multical 21.
Unfortunateley the USB device seems to have a problem. 

Are there any indications to find out what the error message means?

(serial) no working devices, stopping before entering select.
(serial) stopping manager

 wmbusmeters --debug --verbose /dev/ttyUSB0:im871a MyTapWater multical21:c1 74802810 XXXXXXcryptXXXXX

out:
(wmbusmeters) version: HEAD_
(config) using device: /dev/ttyUSB0
(config) with: im871a
(config) number of meters: 1
(detect) "/dev/ttyUSB0" "im871a"
(detect) is_tty=1 is_stdin=0 is_file=0
(im871a) on /dev/ttyUSB0
(serial) registered regular callback 0 DEVICE_IM871A every 60 seconds
(config) meter multical21 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "MyTapWater" "multical21" "74802810" encrypted
(wmbus) set timeout DEVICE_IM871A to "0" with expected activity ""
(im871a) set link mode 06
(im871a) get config
(config) listen to link modes: c1
(serial) waiting for stop
(serial) no working devices, stopping before entering select.
(serial) stopping manager
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working emergency exit!
(serialtty) closed /dev/ttyUSB0

Output from journalctl -xa:
-- The job identifier is 1299 and the job result is done.
Sep 17 19:05:00 debian-nb kernel: usb 2-2.1: new full-speed USB device number 6 using uhci_hcd
Sep 17 19:05:00 debian-nb kernel: usb 2-2.1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
Sep 17 19:05:00 debian-nb kernel: usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 17 19:05:00 debian-nb kernel: usb 2-2.1: Product: WiMOD iM871A-usb
Sep 17 19:05:00 debian-nb kernel: usb 2-2.1: Manufacturer: Silicon Labs
Sep 17 19:05:00 debian-nb kernel: usb 2-2.1: SerialNumber: 00B93EA5
Sep 17 19:05:00 debian-nb kernel: cp210x 2-2.1:1.0: cp210x converter detected
Sep 17 19:05:00 debian-nb kernel: usb 2-2.1: cp210x converter now attached to ttyUSB0
Sep 17 19:05:00 debian-nb systemd[1]: Invalid unit name "wmbusmeters@/dev/im871a_0.service" was escaped as "wmbusmeters@-dev-im871a_0.service" (maybe you should use systemd-escape?)
Sep 17 19:05:00 debian-nb systemd[1]: Starting "wmbusmeters service (udev triggered by /dev/im871a_0)"...
-- Subject: A start job for unit wmbusmeters@-dev-im871a_0.service has begun execution
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit wmbusmeters@-dev-im871a_0.service has begun execution.
-- 
-- The job identifier is 1300.
Sep 17 19:05:00 debian-nb wmbusmetersd[3587]: (wmbusmeters) started /run/wmbusmeters/wmbusmeters--dev-im871a_0.pid
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "device" "/dev/ttyUSB0:im871a"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "logtelegrams" "true"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "format" "json"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "meterfiles" "/var/log/wmbusmeters/meter_readings"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "meterfilesaction" "append"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "meterfilesnaming" "name"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "meterfilestimestamp" "day"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "logfile" "/var/log/wmbusmeters/wmbusmeters.log"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "shell" "/usr/local/bin/mosquitto_pub -d -h localhost -t wmbusmeters/$METER_ID -m "$METER_JSON""
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "alarmshell" "/usr/local/bin/mosquitto_pub -d -h localhost -t wmbusmeters_alarm -m "$ALARM_TYPE $ALARM_MESSAGE""
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "alarmtimeout" "1h"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "alarmexpectedactivity" "mon-sun(00-23)"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) "" ""
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) loading meter file /etc/wmbusmeters.d/MyTapWater
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) name=MyTapWater
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) type=multical21
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) id=74802810
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) key=<notprinted>
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (config) overriding device with "/dev/im871a_0"
Sep 17 19:05:00 debian-nb wmbusmetersd[3590]: (wmbusmeters) using log file /var/log/wmbusmeters/wmbusmeters.log
Sep 17 19:05:00 debian-nb systemd[1]: Started "wmbusmeters service (udev triggered by /dev/im871a_0)".
-- Subject: A start job for unit wmbusmeters@-dev-im871a_0.service has finished successfully
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- A start job for unit wmbusmeters@-dev-im871a_0.service has finished successfully.

weetmuts, 2020-09-18:
It looks like the daemon is running. Thus you cannot access the wmbus device from the command line because
the daemon is hogging it.

Stop the daemon:
sudo systemctl stop wmbusmeters@-dev-im871a_0.service

then try again from the command line.

matze-dev, 2020-09-18:
Dear Fredrik,
Thanx, but the problem is still present. Now I can't read out with outher failures.

wmbusmeters --debug --verbose /dev/ttyUSB0:im871a MyTapWater multical21:c1 74802810 xxxxxcryptxxxxxxxx
(wmbusmeters) version: HEAD_
(config) using device: /dev/ttyUSB0
(config) with: im871a
(config) number of meters: 1
(detect) "/dev/ttyUSB0" "im871a"
(detect) is_tty=1 is_stdin=0 is_file=0
(im871a) on /dev/ttyUSB0
(serial) registered regular callback 0 DEVICE_IM871A every 60 seconds
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0
(config) meter multical21 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "MyTapWater" "multical21" "74802810" encrypted
(wmbus) set timeout DEVICE_IM871A to "0" with expected activity ""
(im871a) set link mode 06
(serial /dev/ttyUSB0) sent "A5010306000206300101"
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) received binary "A5810617FF000600B32582201000010001FF070232000101000000A7AA"
(im871a) checkIM871AFrame "A5810617FF000600B32582201000010001FF070232000101000000A7AA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 aaa7 expected aaa7
(im871a) received full frame
(im871a) get config completed
(im871a) config: device mode 00
(im871a) config: link mode 06
(im871a) config: wmbus c-field 00
(im871a) config: wmbus mfg id 25b3 (IMS)
(im871a) config: wmbus device id 00102082
(im871a) config: wmbus version 01
(im871a) config: wmbus device type 00
(im871a) config: radio channel 01
(im871a) config: radio power level 07
(im871a) config: radio data rate 02
(im871a) config: radio rx window 32
(im871a) config: auto power saving 00
(im871a) config: auto RSSI attachment 01
(im871a) config: auto rx timestamp attachment 01
(im871a) config: led control 00
(im871a) config: rtc control 00
(config) listen to link modes: c1
(serial) waiting for stop
(wmbus) Hit timeout(0 s) but no expected activity!
(wmbus) Hit timeout(0 s) but no expected activity!
(wmbus) Hit timeout(0 s) but no expected activity!
(wmbus) Hit timeout(0 s) but no expected activity!
(wmbus) Hit timeout(0 s) but no expected activity!

weetmuts, 2020-09-18:
The log looks good. Is the meter transmitting on C1 or T1?

matze-dev, 2020-09-18:
Hi Fredrik,
I assume that this is C1, on T1 it gives an error:
wmbusmeters --debug --verbose /dev/ttyUSB0:im871a MyTapWater multical21:t1 74802810 xxxcryptxxx
(cmdline) cannot set link modes to: t1 because meter multical21 only transmits on: c1

Best regards
   Matthias.

matze-dev, 2020-09-18:
Hi Frederik,
the Type of the meter is 021 41 B0 - this must be an t1 communication - but the reading is only for c1 allowed?

Best regards Matthias.

weetmuts, 2020-09-18:
It is very easy to make wmbusmeters accept t1 telegrams for multical21. Just change the LIST_OF_METERS 
table in meters.h Please test.

Is type "021 41 B0" the config numbers printed on the meter? How do you know it is T1?

matze-dev, 2020-09-18:
Hi Frederik,
thanx, https://products.kamstrup.com/download.php?uid=515d4ab700278 - see here the definition at page 16 and http://products.kamstrup.com/ajax/downloadFile.php?uid=5bcecc825f706&display=1 page 2
My type is 021 41 xxx - 41 means Wireless M-Bus. 868 MHz. mode T1 - OMS

Best regards Matthias.

matze-dev, 2020-09-18:
Hi Frederik,
setting changed - it works!!!!!!
Can you fix it in the official sources the there is a choice?

Best regards

      Matthias.

matze-dev, 2020-09-18:
You can close the issue.

matze-dev, 2020-10-07:
Hi all,
are there a future solution for T1 datagrams in multical 21? When will you expect this?
Best regards

 Matthias.

weetmuts, 2020-10-08:
Yes, I have the fix. It will unfortunately mean a regression, since people running 
with just multical21 will be forced to add :c1, like multical21:c1 

In a way it is inevitable, since we discovered that multical21 can indeed transmit on t1 or c1 and
you should let wmbuseters know so that it can check if you can actually hear the meter.

Previously multical21 could only use c1, therefore the meter defaulted to c1.

I think perhaps I should make wmbusmeters always require the linkmode.
This way there will be no future regressions if meters are discovered to be
transmitting on new linkmodes....

What do you think?

matze-dev, 2020-10-09:
Hi all,
yes I fixed the problem while I changed the type in meters.h to the fix value t1 bit. Recompilation and it works.
I think it's a good way to be more flexible to select the mode in command line (append the :c1 or :t1). There are other meters which have multiple modes, you can unify the behaviour.
Please regard to the documentation.

Best greetings
  Matthias.

weetmuts, 2020-11-15:
The fix is in master. There is a slight backward incompatibility for those who use the im871a.
There is no problem for rtlwmbus or amb8465 users, since these dongles can listen to c1 and t1 
at the same time.

closed 2020-11-15 by weetmuts