wmbusmeters/wmbusmeters #165: Novice user trying to implement multical603

elbaek89, 2020-09-22:
Hi Weetmuts

As the title suggests i've been trying to implement the newer multical 603 to your software, but my perhaps limited experienced in software development has me stuck. 

As the help file suggests, i've identified the meter i expect to be most similar to the 603 heat meter (the multical403 heat meter). 
I have then taken the following steps: 

1: Made a copy of the meter_multical403.cc file, and replaced all instances of 403 with 603. 
2: Modified the meters.h and included a line for the 603 following the 403 in both LIST_OF_METERS and METER_DETECTION

LIST_OF_METERS: 
    X(multical403,C1_bit, Heat,        MULTICAL403, Multical403)  \
    X(multical603,C1_bit, Heat,        MULTICAL603, Multical603)  \

METER_DETECTION: 
    X(MULTICAL403,MANUFACTURER_KAM, 0x0d,  0x34) \
    X(MULTICAL603,MANUFACTURER_KAM, 0x04,  0x35) \

The entries in meter detection i found by running wmbusmeters listento=c1 command, where i got answers from two meters: 
Received telegram from: 80363608
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
           device type: Heat meter (0x04)
            device ver: 0x35
         device driver: unknown!
Received telegram from: 68542068
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
           device type: Cold water meter (0x16)
            device ver: 0x1d
         device driver: flowiq3100

The first one is the Multical603, while the second one is another meter that i'm not particularly interested in at the moment. Nonetheless the flowiq3100 helped me find out which entries i should put enter in METER_DETECTION. 

3: Finally, since i have the kem-file and password, i edited the kem-import.py script:

    # meter model identification
    # CONTRIBUTING NOTE: additional meter types supported by wmbusmeters can be put here
    # if their identification in KEM file is known
    if (meterName == 'MC302') and (meterModel.startswith('302T')):
        wmbusmeters_driver = 'multical302'
    elif (meterName == 'MC403') and (meterModel.startswith('403')):
        wmbusmeters_driver = 'multical403'
    elif (meterName == 'MC21') and (meterModel.startswith('021')):
        wmbusmeters_driver = 'multical21'
    elif (meterName == 'MC603') and (meterModel.startswith('603')):
        wmbusmeters_driver = 'multical603'
    else:
        wmbusmeters_driver = None

@@@@@@@

At first, i had an error in the kem-import.py file, where it would leave the wmbusmeters_driver = 'multical21'. Before i realized this error, apparently something went (kinda) right, because i got the following in the meterfiles. However, i have not been able to get new loggings - and of course most of the registers are missing due to the wrong format. 

{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.65,"target_m3":0,"max_flow_m3h":0.018,"flow_temperature_c":50.67,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T12:46:23Z"}
{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.65,"target_m3":0,"max_flow_m3h":0.018,"flow_temperature_c":50.61,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T12:48:02Z"}
{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.65,"target_m3":0,"max_flow_m3h":0.018,"flow_temperature_c":50.62,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T12:49:41Z"}
{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.65,"target_m3":0,"max_flow_m3h":0.017,"flow_temperature_c":50.45,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T12:51:20Z"}
{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.65,"target_m3":0,"max_flow_m3h":0.016,"flow_temperature_c":50.35,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T12:52:59Z"}
{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.65,"target_m3":0,"max_flow_m3h":0.018,"flow_temperature_c":50.16,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T12:54:37Z"}
{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.65,"target_m3":0,"max_flow_m3h":0.015,"flow_temperature_c":50.05,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T12:56:16Z"}
{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.65,"target_m3":0,"max_flow_m3h":0.013,"flow_temperature_c":49.75,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T12:59:34Z"}
{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.66,"target_m3":0,"max_flow_m3h":0.012,"flow_temperature_c":48.84,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T13:12:41Z"}
{"media":"heat","meter":"multical21","name":"80363608","id":"80363608","total_m3":4.66,"target_m3":0,"max_flow_m3h":0.013,"flow_temperature_c":48.74,"external_temperature_c":127,"current_status":"","time_dry":"","time_reversed":"","time_leaking":"","time_bursting":"","timestamp":"2020-09-22T13:14:19Z"}

@@@@@@@@@@

Now when i change the driver in the meter file to 'multical603', i get the message below in the log file: 
(config) No meters supplied. You must supply which link modes to listen to. Eg. --listento=<modes>

Changing it back to something that is recognized ('multical21') gives me: 
(meter) 80363608: meter detection did not match the selected driver multical21! correct driver is: unknown!

I had some trouble installing wmbusmeters by cloning the git, but that may be because it was already installed. I found that i had a running installation which was located in the /home/pi/libmbus-0.8.0 directory. Sorry if these issues make troubleshooting difficult. I'll gladly try to make a fresh install if you can instruct me how to remove the current installation. 

I hope I have supplied all that is needed! The fact that i have seen some measurements makes me feel like i am close to a breakthrough - but who knows :) 

Thank you for your work on the package! 

Kind regards
elbaek

weetmuts, 2020-09-22:
Looks like you are on the right track!

If you can create a fork on github. Clone your fork, apply your changes and then
git push your fork, the I can check your code!

Also we need an example telegram from the regression tests
and to check 

Can you post/email me such a telegram from the --debug log,
the lines look like this:
(wmbus) "telegram=||3E44A.......................|+0"
You put:
telegram=|3E44A.......................|
into a file named for example:
simulation_xxyy.txt 
and run 
wmbusmeters simulation_xxyy.txt

then you can replay and test over and over again.

elbaek89, 2020-09-22:
Hi weetmuts

When i try to run the wmbusmeters command using the meterfile in which i have specified the multical603 meter type that i think i have set up, i get the following: 
pi@raspberrypi:/etc/wmbusmeters.d $ wmbusmeters --debug /dev/ttyUSB0:im871a 80363608 multical603 80363608 FAFEE528771B56D1781F0F269CA7808F
Not a valid meter type "multical603"

Specifying the same id and key but using a multical403 template instead gives me the result below: 
pi@raspberrypi:~/libmbus-0.8.0/wmbusmeters $ wmbusmeters --debug /dev/ttyUSB0:im871a test403 multical403 80363608 FAFEE528771B56D1781F0F269CA7808F
(wmbusmeters) version: 0.9.36
(config) using device: /dev/ttyUSB0
(config) with: im871a
(config) number of meters: 1
(detect) "/dev/ttyUSB0" "im871a"
(detect) is_tty=1 is_stdin=0 is_file=0
(im871a) on /dev/ttyUSB0
(serial) registered regular callback 0 DEVICE_IM871A every 60 seconds
(config) meter multical403 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "test403" "multical403" "80363608" encrypted
(wmbus) set timeout DEVICE_IM871A to "0" with expected activity ""
(im871a) set link mode 06
(im871a) get config
(config) listen to link modes: c1
(serial) waiting for stop
(serial) no working devices, stopping before entering select.
(serial) stopping manager
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working emergency exit!
(serialtty) closed /dev/ttyUSB0

I'm not sure what the cause of the above is, because it seems like i am still getting readings when applying the multical403 driver - even though the log file states that it's the wrong file: 

{"media":"heat","meter":"multical403","name":"testMeter403","id":"80363608","total_energy_consumption_kwh":0,"total_volume_m3":4.95,"volume_flow_m3h":0.016,"t1_temperature_c":50.18,"t2_temperature_c":23.26,"at$$26,"at_date":"","current_status":"","timestamp":"2020-09-22T20:23:30Z"}

That reading is from just now - seems like the measurement time is a couple of hours off local time though - probably just a combination of UTC vs my timezone + summertime. 

I haven't been using git for this little project yet, and i fear that all the issues i've had during installation would make that process messy. Since i would like to go about this properly and to be able to share if i get it working, is there a foolproof way for me to completely remove wmbusmeters to allow for a fresh install? (sorry, did say novice)

Kind regards
elbaek

elbaek89, 2020-09-23:
Oh - and btw. I don't know if this is any use to you, but I have a full telegram from the same logger but from a **wired** mbus module. Don't know if that could be of use in adjusting the multical603.cc file, but i'll leave it here just in case: 

`<?xml version="1.0" encoding="ISO-8859-1"?>
<MBusData>

    <SlaveInformation>
        <Id>80133685</Id>
        <Manufacturer>KAM</Manufacturer>
        <Version>53</Version>
        <ProductName></ProductName>
        <Medium>Heat: Inlet</Medium>
        <AccessNumber>1</AccessNumber>
        <Status>00</Status>
        <Signature>0000</Signature>
    </SlaveInformation>

    <DataRecord id="0">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Energy (kWh)</Unit>
        <Value>8953</Value>
        <Timestamp>2020-08-26T12:12:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>12801</Value>
        <Timestamp>2020-08-26T12:12:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>4915</Value>
        <Timestamp>2020-08-26T12:12:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="3">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Volume (1e-2  m^3)</Unit>
        <Value>18237</Value>
        <Timestamp>2020-08-26T12:12:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="4">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>0</Tariff>
        <Device>1</Device>
        <Unit>Volume (1e-2  m^3)</Unit>
        <Value>0</Value>
        <Timestamp>2020-08-26T12:12:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="5">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Tariff>0</Tariff>
        <Device>2</Device>
        <Unit>Volume (1e-2  m^3)</Unit>
        <Value>0</Value>
        <Timestamp>2020-08-26T12:12:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="6">
        <Function>Instantaneous value</Function>
[... 173 lines cut ...]
        <Unit>Fabrication number</Unit>
        <Value>80133685</Value>
        <Timestamp>2020-08-26T12:12:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="28">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>2000101</Value>
        <Timestamp>2020-08-26T12:12:01Z</Timestamp>
    </DataRecord>

    <DataRecord id="29">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>11851101</Value>
        <Timestamp>2020-08-26T12:12:01Z</Timestamp>
    </DataRecord>

</MBusData>
`

/elbaek

weetmuts, 2020-09-23:
I think your daemon is running. Stop it with sudo systemctl stop wmbusmeters@..... something.
You can also do:
make uninstall

Then reboot.
Rebuild your source but do not run make install.
Now you can do ./build/wmbusmeters .....
without any problems with the daemon.

elbaek89, 2020-09-23:
I've removed everything and started fresh but running "make uninstall" as per your suggestion, then rebooted and then removed the mess of directories in which i accidentally installed wmbusmeters at first. 

Now, with a fresh install, i can't seem to get the daemon to load properly when installing the stick. 
`● wmbusmeters@-dev-im871a_0.service                                                                                                                             loaded failed failed    "wmbusmeters service (udev triggered by /dev/im871a_0)"
`
I'm getting the following entries in syslog: 

```
Sep 23 08:53:02 raspberrypi systemd[1]: Starting "wmbusmeters service (udev triggered by /dev/im871a_0)"...
Sep 23 08:53:02 raspberrypi systemd[2414]: wmbusmeters@-dev-im871a_0.service: Failed to determine user credentials: No such process
Sep 23 08:53:02 raspberrypi systemd[2414]: wmbusmeters@-dev-im871a_0.service: Failed at step USER spawning /bin/mkdir: No such process
Sep 23 08:53:02 raspberrypi systemd[2415]: wmbusmeters@-dev-im871a_0.service: Failed to determine user credentials: No such process
Sep 23 08:53:02 raspberrypi systemd[2415]: wmbusmeters@-dev-im871a_0.service: Failed at step USER spawning /bin/chown: No such process
Sep 23 08:53:02 raspberrypi systemd[1]: wmbusmeters@-dev-im871a_0.service: Control process exited, code=exited, status=217/USER
Sep 23 08:53:02 raspberrypi systemd[1]: wmbusmeters@-dev-im871a_0.service: Failed with result 'exit-code'.
```
Did i mess anything up during the install?

elbaek89, 2020-09-23:
Still having issues with getting the daemon running - but with this install i managed to get further than last time! 
I think I just received the telegram: 

(I have included a part of the output from the --debug log - the full-length (non-compressed?) telegram is towards the end). 
```
(wmbus) ignoring compressed telegram since format signature hash 0x7869 is yet unknown.
     this is not a problem, since you only need wait for at most 8 telegrams
     (8*16 seconds) until an full length telegram arrives and then we know
     the format giving this hash and start decoding the telegrams properly.
(wmbus) telegram ignored by all configured meters!
(serial) received binary "A5E20323442D2C682054681D168D20D45018E820F234F4862A77210083F1688643FAC4EDFBC03D05C60000003E20"
(im871a) checkIM871AFrame "A5E20323442D2C682054681D168D20D45018E820F234F4862A77210083F1688643FAC4EDFBC03D05C60000003E20"
(im871a) has_timestamp=1 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) timestamp 0000c605
(im871a) rssi 00
(im871a) got crc16 203e expected 203e
(im871a) received full frame
(wmbus) parseDLL @0 36
(meter) 80363608: for me? 68542068
(meter) 80363608: not for me: not my id
(wmbus) telegram ignored by all configured meters!
(wmbus) Hit timeout(0 s) but no expected activity!
(serial) received binary "A5E20309472D2C08363680350400000000B61129"
(im871a) checkIM871AFrame "A5E20309472D2C08363680350400000000B61129"
(im871a) has_timestamp=1 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) timestamp 00000000
(im871a) rssi b6
(im871a) got crc16 2911 expected 2911
(im871a) received full frame
(wmbus) parseDLL @0 10
(meter) 80363608: for me? 80363608
(meter) 80363608: yes for me
(meter) 80363608 80363608 "09472D2C083636803504"
(wmbus) parseDLL @0 10
(telegram) DLL L=09 C=47 (from meter ACC_NR) M=2c2d (KAM) A=80363608 VER=35 TYPE=04 (Heat meter) (driver multical603)
(wmbus) telegram ignored by all configured meters!
(serial) received binary "A5E20342442D2C0836368035048D20E18025B620D5068BAB662E9A238F8F057134E9F170B957B5C6DED106363C32EDAD105A1D0F3995528363625036986356C5F9FBEEB49187A0BB000000B558"
(im871a) checkIM871AFrame "A5E20342442D2C0836368035048D20E18025B620D5068BAB662E9A238F8F057134E9F170B957B5C6DED106363C32EDAD105A1D0F3995528363625036986356C5F9FBEEB49187A0BB000000B558"
(im871a) has_timestamp=1 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) timestamp 0000bba0
(im871a) rssi 00
(im871a) got crc16 58b5 expected 58b5
(im871a) received full frame
(wmbus) parseDLL @0 67
(meter) 80363608: for me? 80363608
(meter) 80363608: yes for me
(meter) 80363608 80363608 "42442D2C0836368035048D20E18025B620D5068BAB662E9A238F8F057134E9F170B957B5C6DED106363C32EDAD105A1D0F3995528363625036986356C5F9FBEEB49187"
(wmbus) parseDLL @0 67
(telegram) DLL L=42 C=44 (from meter SND_NR) M=2c2d (KAM) A=80363608 VER=35 TYPE=04 (Heat meter) (driver multical603)
(wmbus) parseELL @10 57
(ELL) decrypting "D5068BAB662E9A238F8F057134E9F170B957B5C6DED106363C32EDAD105A1D0F3995528363625036986356C5F9FBEEB49187"
(ELL) IV 2D2C083636803504208025B620000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "87D0780406A500000004FF072B010000"
(ELL) block 1 block_size 16 offset 16
(ELL) decrypted "04FF089C000000041421020000043B12"
(ELL) block 2 block_size 16 offset 32
(ELL) decrypted "0000000259D014025D000904FF220000"
(ELL) block 3 block_size 2 offset 48
(ELL) decrypted "0000"
(ELL) decrypted "87D0780406A500000004FF072B01000004FF089C000000041421020000043B120000000259D014025D000904FF2200000000"
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=e1 SN=8025b620 (AES_CTR session=0 time=746072) CRC=87d0
(wmbus) parseAFL @19 48
(wmbus) parseTPL @19 48
(dvparser) found new format "040604FF0704FF080414043B0259025D04FF22" with hash 7869, remembering!
(telegram) TPL CI=78
(multical603) log "telegram=|42442D2C0836368035048D20E18025B62087D078|0406A500000004FF072B01000004FF089C000000041421020000043B120000000259D014025D000904FF2200000000|+296"
(multical603) 00: 42 length (66 bytes)
(multical603) 01: 44 dll-c (from meter SND_NR)
(multical603) 02: 2d2c dll-mfct (KAM)
(multical603) 04: 08363680 dll-id (80363608)
(multical603) 08: 35 dll-version
(multical603) 09: 04 dll-type (Heat meter)
(multical603) 0a: 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(multical603) 0b: 20 ell-cc (slow_resp sync)
(multical603) 0c: e1 ell-acc
(multical603) 0d: 8025b620 sn (AES_CTR)
(multical603) 11: 87d0 payload crc (calculated 87d0 OK)
(multical603) 13: 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(multical603) 14: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 15: 06 vif (Energy kWh)
(multical603) 16: A5000000
(multical603) 1a: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 1b: FF vif (Vendor extension)
(multical603) 1c: 07 vife (?)
(multical603) 1d: 2B010000
(multical603) 21: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 22: FF vif (Vendor extension)
(multical603) 23: 08 vife (?)
(multical603) 24: 9C000000
(multical603) 28: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 29: 14 vif (Volume 10⁻² m³)
(multical603) 2a: * 21020000 total volume (5.450000 m3)
(multical603) 2e: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 2f: 3B vif (Volume flow l/h)
(multical603) 30: * 12000000 volume flow (0.018000 m3/h)
(multical603) 34: 02 dif (16 Bit Integer/Binary Instantaneous value)

```

weetmuts, 2020-09-23:
Very nice!

You can see in your debug log which fields from the meter that you have properly handled in the driver.
This one for example is fine:
(multical603) 2a: * 21020000 total volume (5.450000 m3)
whereas this one for total kwh is not handled:
(multical603) 16: A5000000

Try to use findKey to find the kwh (look in meter_multical302.cc for example.)
Do the same for the rest of the unhandled fields.

Then there are the vendor extensions, they have to be looked up explicitly 
using: extractDVuint32(&t->values, "04FF07", &offset, &something_);
(You need to add extractDVuint32, but it should be trivial, just copy extractDVuint24.)
Then we need to figure out what those bits really mean!

Then you need to make sure the addPrints are correct of course, to get 
the handled data out in the json.

elbaek89, 2020-09-23:
Alright! I'll look into that later today. In the mean time, can you give me some pointers about how to troubleshoot the daemon not loading? 

`Sep 23 08:53:02 raspberrypi systemd[1]: Starting "wmbusmeters service (udev triggered by /dev/im871a_0)"...
Sep 23 08:53:02 raspberrypi systemd[2414]: wmbusmeters@-dev-im871a_0.service: Failed to determine user credentials: No such process
Sep 23 08:53:02 raspberrypi systemd[2414]: wmbusmeters@-dev-im871a_0.service: Failed at step USER spawning /bin/mkdir: No such process
Sep 23 08:53:02 raspberrypi systemd[2415]: wmbusmeters@-dev-im871a_0.service: Failed to determine user credentials: No such process
Sep 23 08:53:02 raspberrypi systemd[2415]: wmbusmeters@-dev-im871a_0.service: Failed at step USER spawning /bin/chown: No such process
Sep 23 08:53:02 raspberrypi systemd[1]: wmbusmeters@-dev-im871a_0.service: Control process exited, code=exited, status=217/USER
Sep 23 08:53:02 raspberrypi systemd[1]: wmbusmeters@-dev-im871a_0.service: Failed with result 'exit-code'.`

I've tried snooping around, but haven't found anything yet. The "Failed to determine user credentials" made me run "sudo make install" again, since this is when the wmbusmeters user is created (couldn't figure out if it existed or not). But this did not resolve it.

/elbaek

Bluhme1, 2020-09-23:
Hello Elbaek
Following your efforts with great interest. I expect to have my heater meter replaced by a Multical603 within the next few months.
Right now I am using Weetmuts's wonderful software to read my Multical 21 water meter. (After I succeeded in  getting an encryption key from the authorities): Will the Multical603 require an encryption key, also?
/Lars (Denmark)

elbaek89, 2020-09-23:
Hi Lars

Yeah i think many Danes are getting the 603 these days. And yes, i can confirm that you need an encryption key. I've also heard that it varies how much trouble you can expect obtaining it - I'm lucky to have a district heating supplier that appreciates tinkering. We will be using it for research activities, and so far i'm not planning on trying out other software implementations as i hope we can implement it here without too much trouble - especially given the fact that i've already seen temperature measurements which line up with the ones i received from the wired mbus. 

/elbaek

elbaek89, 2020-09-25:
> Sep 23 08:53:02 raspberrypi systemd[1]: Starting "wmbusmeters service (udev triggered by /dev/im871a_0)"...
> Sep 23 08:53:02 raspberrypi systemd[2414]: wmbusmeters@-dev-im871a_0.service: Failed to [...]

FYI for anyone experiencing the above: 
Ok - so i managed to fix the above issue with starting the service. Turned out the wmbusmeters user was not created during my run of the install since the group wmbusmeters already existed from a previous install, which resulted in the following being printed when running the install file: 

useradd: group wmbusmeters exists - if you want to add this user to that group, use -g.

In my case, running the following line fixed the issue: 
sudo useradd --system --shell /usr/sbin/nologin -g dialout wmbusmeters

Back to looking at the actual implementation now ...

elbaek89, 2020-09-25:
```
pi@raspberrypi:~ $ wmbusmeters --debug /dev/ttyUSB0:im871a 80363608 multical603 80363608 KEY
(wmbusmeters) version: 0.9.36-5-g30623d3 with local changes
(config) using device: /dev/ttyUSB0
(config) with: im871a
(config) number of meters: 1
(detect) "/dev/ttyUSB0" "im871a"
(detect) is_tty=1 is_stdin=0 is_file=0
(im871a) on /dev/ttyUSB0
(serial) registered regular callback 0 DEVICE_IM871A every 60 seconds
(config) meter multical603 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "80363608" "multical603" "80363608" encrypted
(wmbus) set timeout DEVICE_IM871A to "0" with expected activity ""
(im871a) set link mode 06
(im871a) get config
(config) listen to link modes: c1
(serial) waiting for stop
(serial) no working devices, stopping before entering select.
(serial) stopping manager
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working emergency exit!
(serialtty) closed /dev/ttyUSB0
pi@raspberrypi:~ $ systemctl |grep wmbusmeters
wmbusmeters@-dev-im871a_0.service                                                                                                                             loaded active running   "wmbusmeters service (udev triggered by /dev/im871a_0)"
system-wmbusmeters.slice                                                                                                                                      loaded active active    system-wmbusmeters.slice
```

What does the following part refer to? Is it the wmbus dongle or something else?
(serial) no working devices, stopping before entering select.
(serial) stopping manager
(serial) event loop stopped!
(serial) closing 0 devices
(serial) no devices working emergency exit!
(serialtty) closed /dev/ttyUSB0

I'm not sure what has changed since it suddently won't allow me to listen in on that specific meter, since i did that plenty yesterday. Any hints appreciated.

I've also tried running some simulations - and i just now realized that the simulation file literally has to be named "simulation_" for it to work. Here i seem to be getting responses similar to what i was seeing back when i could listen to the meter. 
```

pi@raspberrypi:~/wmbusmeters/simulations $ wmbusmeters --debug ~/wmbusmeters/simulations/simulation_unknown2.txt 80363608 multical603 80363608 NOKEY
(wmbusmeters) version: 0.9.36-5-g30623d3 with local changes
(config) using device: /home/pi/wmbusmeters/simulations/simulation_unknown2.txt
(config) number of meters: 1
(detect) "/home/pi/wmbusmeters/simulations/simulation_unknown2.txt" ""
(detect) driver: simulation file
(simulator) in /home/pi/wmbusmeters/simulations/simulation_unknown2.txt
(serial) registered regular callback 0 DEVICE_SIMULATOR every 60 seconds
(config) meter multical603 link mode(s): c1
(config) all possible link modes that the meters might transmit on: c1
(config) listen link modes calculated to be: c1
(wmbusmeters) configured "80363608" "multical603" "80363608" not-encrypted
(wmbus) set timeout DEVICE_SIMULATOR to "0" with expected activity ""
(simulator) set link mode c1
(simulator) set link mode completed
(simulator) get link mode
(simulator) config: link mode c1
(config) listen to link modes: c1
(simulator) from file "2e4433300502010007ff7ab66800002f2f02fd1b550002fd971d01000efd3a2300000000008e40fd3a000000000000"
(wmbus) parseDLL @0 47
(meter) 80363608: for me? 00010205
(meter) 80363608: not for me: not my id
(wmbus) telegram ignored by all configured meters!
(simulator) from file "42442D2C0836368035048D20E18025B62087D0780406A500000004FF072B01000004FF089C000000041421020000043B120000000259D014025D000904FF2200000000"
(wmbus) parseDLL @0 67
(meter) 80363608: for me? 80363608
(meter) 80363608: yes for me
[... 32 lines cut ...]
(multical603) 24: 9C000000
(multical603) 28: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 29: 14 vif (Volume 10⁻² m³)
(multical603) 2a: * 21020000 total volume (5.450000 m3)
(multical603) 2e: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 2f: 3B vif (Volume flow l/h)
(multical603) 30: * 12000000 volume flow (0.018000 m3/h)
(multical603) 34: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical603) 35: 59 vif (Flow temperature 10⁻² °C)
(multical603) 36: * D014 T1 flow temperature (53.280000 °C)
(multical603) 38: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical603) 39: 5D vif (Return temperature 10⁻² °C)
(multical603) 3a: * 0009 T2 flow temperature (23.040000 °C)
(multical603) 3c: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 3d: FF vif (Vendor extension)
(multical603) 3e: 22 vife (per hour)
(multical603) 3f: * 00000000 info codes ()
80363608        80363608        0 kWh   5.45 m3 0.018 m3/h      53.28 °C        23.04 °C                2020-09-25 11:09.37
(serial) stopping manager
(serial) waiting for stop
(serial) event loop stopped!
(serial) closing 0 devices
```

One question though, does the above indicate that we are only receiving 5 measurements from the meter? I was really hoping that there would be an "Instantaneous Power" hiding somewhere in there - i.e. in kW and not kWh.

weetmuts, 2020-09-26:
I have merged your code and brought it into the NewDetectionCode branch. 
Please try this branch. Do a "sudo make install" then use auto as the device selector as much as you can.
For example:
./build_debug/wmbusmeters --format=json auto Heat multical603 80363608 __KEY__

If you run on your simulation file:
./build_debug/wmbusmeters --format=json --addconversions=MJ --debug simulations/simulation_multical603.txt Heat multical603 80363608 ""

You will get this output:

{"media":"heat","meter":"multical603","name":"Heat","id":"80363608","total_energy_consumption_kwh":165,"total_energy_consumption_mj":594,"total_volume_m3":5.45,"volume_flow_m3h":0.018,"t1_temperature_c":53.28,"t2_temperature_c":23.04,"at_date":"","current_status":"","something_a":"299","something_b":"156","timestamp":"2020-09-26T07:15:59Z"}

I fixed a minor typo in your code, with the kwh vs MJ units. So now you get proper total_energy_consumption always in kwh,
but if you add addconversions=MJ it will add a mj value as well.

The values something_a and something_b are encoded in the telegram but as vendor extensions, ie we do not
know what the represent. See if you can figure out what the mean! :-)

Thank you very much for your pull request and work on adding this meter!

weetmuts, 2020-09-26:
If you have problems with the NewDetectionCode branch please comment in this issue #169

elbaek89, 2020-09-28:
Hi Weetmuts

Did some digging, and have found which registers something_a and b refer to. 

The response from wmbusmeters: 
```

(multical603) 00: 42 length (66 bytes)
(multical603) 01: 44 dll-c (from meter SND_NR)
(multical603) 02: 2d2c dll-mfct (KAM)
(multical603) 04: 08363680 dll-id (80363608)
(multical603) 08: 35 dll-version
(multical603) 09: 04 dll-type (Heat meter)
(multical603) 0a: 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(multical603) 0b: 20 ell-cc (slow_resp sync)
(multical603) 0c: a9 ell-acc
(multical603) 0d: 90ecb720 sn (AES_CTR)
(multical603) 11: 7b1e payload crc (calculated 7b1e OK)
(multical603) 13: 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(multical603) 14: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 15: 06 vif (Energy kWh)
(multical603) 16: * 19010000 total energy consumption (281.000000 kWh)
(multical603) 1a: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 1b: FF vif (Vendor extension)
(multical603) 1c: 07 vife (?)
(multical603) 1d: * F7010000 something A (503)
(multical603) 21: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 22: FF vif (Vendor extension)
(multical603) 23: 08 vife (?)
(multical603) 24: * 03010000 something B (259)
(multical603) 28: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 29: 14 vif (Volume 10⁻² m³)
(multical603) 2a: * A1030000 total volume (9.290000 m3)
(multical603) 2e: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 2f: 3B vif (Volume flow l/h)
(multical603) 30: * 15000000 volume flow (0.021000 m3/h)
(multical603) 34: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical603) 35: 59 vif (Flow temperature 10⁻² °C)
(multical603) 36: * 9B13 T1 flow temperature (50.190000 °C)
(multical603) 38: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical603) 39: 5D vif (Return temperature 10⁻² °C)
(multical603) 3a: * 2009 T2 flow temperature (23.360000 °C)
(multical603) 3c: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical603) 3d: FF vif (Vendor extension)
(multical603) 3e: 22 vife (per hour)
(multical603) 3f: * 00000000 info codes ()
```
Which i  could then match against a telegram from the wired mbus module of the same meter:

```
    <DataRecord id="1">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>503</Value>
        <Timestamp>2020-09-28T09:47:15Z</Timestamp>
    </DataRecord>

    <DataRecord id="2">
        <Function>Instantaneous value</Function>
        <StorageNumber>0</StorageNumber>
        <Unit>Manufacturer specific</Unit>
        <Value>259</Value>
        <Timestamp>2020-09-28T09:47:15Z</Timestamp>
    </DataRecord>
```
 
Knowing this we can conclude that the meter is sending the following datagram, where something_a and b are the "Energy forward" and "Energy returned" (Energy E8 & Energy E9 in Kamstrup documentation - see picture 2).

[image]

[image]

I was hoping that there would be the register called "Energy (actual)", since this has a higher resolution that than accumulating Energy total measurement - but a workaround for me is then to rely on the Flow (actual) in combination with the temperature differences instead. I am still a bit weary about how to actually implement the above findings in the code (never did c++ before) - so i don't know if its possible for you to plug the info in at the right places whenever you get the time? :) Otherwise then let me know, and i'll give it a shot.

Bluhme1, 2020-10-24:
Hi elbaek89. I am told the my meter will be changed to a multical603 soon. I have my system set up to get data from my water meter Multical21 already using a "MyTapWater" configuration file. How does your configuration file for you Multical603 look like?
Lars (Denmark)

weetmuts, 2020-10-25:
Hi Eric! @elbaek89 I have pushed a fix to master (including the NewDetectionCode)
that renames something_a to energy_forward_kwh and something_b to energy_returned_kwh

Can you check that it works ok for you and that the units make sense?
Normally the unit is encoded into the wmbus protocol but not for the vender extensions.

weetmuts, 2020-10-25:
@Bluhme1 You simply add a new file in /etc/wmbusmeters.d/MyHeatMeter
Contents:
name=MyHeatMeter
type=multical603
id=12345678
key=.....

How it then is being report to your collection system is outside of my knowledge.
But you can add a unique shell=... 
command per meter. If you need to make very different invocations.

But if you are using mosquitto_pub then you can also use ENV variables to 
differentiate the post between the meters, apart from the contents in the json
which of course is different.

weetmuts, 2020-12-01:
Support for multical603 is now wmbusmeters.

closed 2020-12-01 by weetmuts

YOYONL89, 2020-12-04:
Hello weetmuts,

is it possible to add support of the Multical603 to the docker image?
Or is it allready included in the last image? (created 5 days ago)

weetmuts, 2020-12-05:
Sure @YOYONL89 !

It should be in the docker image. If you can shell into the docker image,
then you can run:

wmbusmeters --listmeters

and it will print all supported meters.
or you can do

wmbusmeters --listmeters=multi

and it will print any meter with a name or purpose that contains the string multi.

YOYONL89, 2020-12-07:
@weetmuts 
First of all many thanks for the update.

hmm i cant get in with shell (not internal and not external) which is gonna make diagnoses a bit more complex.
Docker image comes back with the following: Not a valid meter type "multical603"
Another issue that rises is that you're update might have broken the passing through of devices to the docker.
if its not suitable to resolve this in this topic i don't mind to open a new issue.

kind regards,

Johan

weetmuts, 2020-12-07:
Yes, please open a new issue!