wmbusmeters/wmbusmeters #166: Expand ElectricityMeter to include apparent and reactive power

Skeen, 2020-09-24:
Hello,

I'm currently writing the device code for the [EM24 3-phase power meter](https://www.gavazzionline.com/pdf/EM24DINDS.pdf), and it reports reactive power in `kVARh` alongside true power in `kWh`. From these quantities apparent power can be calculated in `kVAh`, however the interface for the `ElectricityMeter` does not currently expose apparent or reactive power.

Thus the question:

How would you feel about adding apparant and reactive power to the `ElectricityMeter` interface?

weetmuts, 2020-09-24:
Please add it! The interfaces and c++ inheritance are not really used in any way, yet.
But eventually it might make sense to put common code for the ElectricityMeters somehwere,
and then the interfaces might be useful.

I looked at your pull request and it is excellent. We only need a test case,
ie a telegram in simulations_c1.txt and another meter in test_c1_meters.sh
 
You can email me a telegram if you like (from the --debug log) and I can anonymize it and
add it to the test suite.

Skeen, 2020-09-24:
I have emailed you a telegram.