wmbusmeters/wmbusmeters #169: rtlsdr device not found when code from NewDetectionCode branch used

pswiatki, 2020-09-25:
When I tried **`NewDetectionCode`** branch I could not get the daemon (systemd service started manually) to find my dongle.
I could find this in **`/var/log/wmbusmeters.log`**:

```
(wmbusmeters) logging started 2020-09-25 14:01:32
No wmbus device detected, waiting for a device to be plugged in.
(wmbusmeters) waiting for telegrams
(wmbusmeters) shutting down
(wmbusmeters) logging started 2020-09-25 15:31:10
No wmbus device detected, waiting for a device to be plugged in.
(wmbusmeters) waiting for telegrams
(wmbusmeters) shutting down
(wmbusmeters) logging started 2020-09-25 15:59:07
No wmbus device detected, waiting for a device to be plugged in.
```
There is neither **`/dev/rtlsdr`** device file nor the symlink pointing to the proper location in **`/dev/bus/usb/......`**  hierarchy.
What needs to be done to get it to work? I switched back or the **`master`** branch and everything went back to normal.

weetmuts, 2020-09-26:
Thank you for testing the NewDetectionCode branch!

When you run "sudo make install" for this branch, it will remove the udev rules.
Since the new wmbusmeters daemon should stick around forever, waiting
for devices to be plugged in  or be removed.

It seems for me, that in the rtlsdr case, the wmbusmeters user must 
be a member of the plugdev group(Ubuntu) to be able to enumerate and find the rtlsdr devices.
The install script fixes this. But maybe this is not the case for you Linux distribution?
Or maybe I got something else wrong. :-)

If you run with loglevel=debug (aka --debug), what does it say?

pswiatki, 2020-09-26:
Just checked the memberships:
```
$ id wmbusmeters 
uid=985(wmbusmeters) gid=985(wmbusmeters) groups=985(wmbusmeters),14(uucp),1001(dialout),1002(plugdev)
```
So yes, **`wmbusmeters`** is a member of **`plugdev`**. However, the only udev rule relevant after removal of wmbusmeters udev rules are those that come with **`rtl-sdr`** project and I do not see any symlinking spec in the rule file in **`/etc/udev/rules.d`**:
```
# RTL2832U OEM vid/pid, e.g. ezcap EzTV668 (E4000), Newsky TV28T (E4000/R820T) etc.
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0bda", ATTRS{idProduct}=="2838", ENV{ID_SOFTWARE_RADIO}="1", MODE="0660", GROUP="plugdev"
```
This is what I get in the log file when loglevel is elevated to debug:

```
(wmbusmeters) logging started 2020-09-27 00:44:31
(wmbusmeters) version: NewDetectionCode_0.9.36-59-gd443fd4
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) number of meters: 1
(main) configured "P15_Woda" "izar" "XXXXXXXXXX" not-encrypted
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>>23930
<<<
(shell) return code 0
(shell) 23933 exited
(shell) exec (capture output) "/bin/pidof"
(serial) EVENT thread interrupted
(shell) arg "wmbusmeters"
(shell) output: >>>23930
<<<
(shell) return code 0
(shell) 23934 exited
(serial) EVENT thread interrupted
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) TIMER thread interrupted

```
Looks like it gets stuck and does not get hold of the dongle at all (that is: can not find it for some reason).

weetmuts, 2020-09-27:
Thanks, wmbusmeters looks for the /dev/swradio0 device. As you say, there seems to be no symlink
generated in your udev rule for the rtl_sdr? (The symlink is created in Ubuntu 18)

This would explain why wmbusmeters does not see the rtl_sdr device.
The actualy link /dev/swradio0 is not really used for the moment, it is used only to detect the 
rtl_sdr dongle and then start rtl_sdr which does the real stuff.

Perhaps there is a better way to detect the rtl_sdr dongle?

pswiatki, 2020-09-27:
I guess there has to be the appropriate udev rule that would create that symlink in Ubuntu 18.
I have the RTL-SDR tools installed from sources and although the udev rules are part of the project, they do not create any symlinks. I guess there must be some other way they use to detect the presence of the stick (traversing the USB device tree and looking for specific VID/PID combos perhaps?).
Optionally, **`wmbuwmeters`** could re-introduce the udev rule to set a well-known symlink in the **`/dev`** tree

weetmuts, 2020-09-27:
Well, the udev rule for the rtl_sdr is probably quite stable. The problem is the wmbus dongles, they usually
have a generic usb2serial chip, so you cannot differentiate between two completely different wmbus dongles,
because on the usb level, they look like the same serial chip.

But, yes, if I can avoid udev rules that is better. So the answer is probably somewhere in the rtl_sdr sources. Since that program can do it, wmbusmeters should be able to do it! If you can find the code in question in rtl_sdr, that would speed up the process!

pswiatki, 2020-09-27:
Well.... I just checked:
the following array is defined in **`librtlsdr.c`**:
```
static rtlsdr_dongle_t known_devices[] = {
        { 0x0bda, 0x2832, "Generic RTL2832U" },
        { 0x0bda, 0x2838, "Generic RTL2832U OEM" },
(....)
```
Also, in **`convenience/convenience.c`** one can find a definition of **`verbose_device_search()`** function which is later used in **`rtl_sdr.c`** main utility application. Depending on what is used as the command line parameter **`verbose_device_search()`** tries to find a suitable match. So their approach looks pretty simplistic.

pswiatki, 2020-09-27:
I think the issue of multiple wmbus dongles using the same USB-UART bridge should be addressed by some middle layer in the software so that the dongle can be uniquely identified by some polling mechanism and independent of the underlying communication interface.  **`rtl-sdr`** is somewhat easier in this regard, as it uses a separate process in the middle (**`rtl-wmbus`**)

weetmuts, 2020-09-27:
Yeah, that is what the new detection cod is doing. :-)
It finds all serial interface and probes them to detect which dongles are attached. As a bonus you can now have multiple dongles at the same time.
You will be able to configure one for c1 at 868Mhz another for t1 at 463Mhz etc.

weetmuts, 2020-09-27:
I changed wmbusmeters to link to librtlsdr directly and use its functions to check if there is an rtl_sdr dongle attached.

You need to: sudo apt install librtlsdr-dev
Run ./configure and it will tell you if you have it properly.

Please test!

pswiatki, 2020-09-27:
**`apt install`** is a no-go on my system (Arch Linux). I compiled **`rtl-sdr`** project (which includes lib) from sources. But the end result should be the same. When you say **`./configure`** you mean the one that is in **`wmbusmeters`**.

pswiatki, 2020-09-27:
Just tried  **`NewDetectionCode`** branch again. Much better result. It looks to be successful:

```
(serial) no devices working emergency exit!
(wmbusmeters) shutting down
(shell) sending SIGTERM to process group 36709
(bgshell) sent sigterm, now waiting for child 36709 to exit.
(bgshell) 36709 terminated due to signal 15
(serial) no devices working emergency exit!
(serialcmd) closed /bin/sh pid=36709 fd=4
(wmbusmeters) logging started 2020-09-28 01:10:45
(wmbusmeters) version: NewDetectionCode_0.9.36-67-gf777f71
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) number of meters: 1
(main) configured "P15_Woda" "izar" "4122bcb2" not-encrypted
(main) rtlsdr device 0_Realtek_RTL2838UHIDIR_0000616 not currently used.
Started DEVICE_RTLWMBUS on 0_Realtek_RTL2838UHIDIR_0000616 (auto)
(rtlwmbus) using command: /usr/bin/rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(serial) EVENT thread interrupted
(serialcmd) opened /bin/sh pid 38110 fd 5 (rtlwmbus)
(config) meter izar link mode(s): t1
(config) all possible link modes that the meters might transmit on: t1
(config) listen link modes calculated to be: t1
(main) regular reset of DEVICE_RTLWMBUS 0_Realtek_RTL2838UHIDIR_0000616 will happen every 82800 seconds
(config) listen to link modes: any
(wmbus) no alarm (expected activity) for DEVICE_RTLWMBUS
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
(serial) waiting for stop
(serial) received ascii "T1;1;1;2020-09-27 23:10:47.000;13;20;4113E570;0x1944304c70e51341cc01a2311a00136959202ee511395b17fb40<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-09-27 23:10:47.000;13;20;4113E570;0x1944304c70e51341cc01a2311a00136959202ee511395b17fb40<0A>"
(rtlwmbus) received full frame
```
By the way: is there anything in the log (at debug level) I could use to gauge the signal level (or signal to noise ratio) when the frame of interest is received?

elbaek89, 2020-10-01:
Hi Weetmuts 

I've commented on the branch i pushed with some information about the last two registers. 

I see that the udev rules have been disabled for this branch. Is there a way for me to start the daemon and have it log in the samme manner as the automatic logging that happened on the master branch? I.e. saving to meter_readings with filenames according to day etc? I've tried using the wmbusmetersd -useconfig=/ command, but i don't see any logs show up. 

Kind regards
elbaek

weetmuts, 2020-10-04:
Big changes. Several bugs introduced (to be fixed of course), but new major functionality written.

Try the NewDetectionCode branch like this
with your im871a dongle:

wmbusmeters auto:c1

This will print: (your serial nr in [...] will be different)
Started auto im871a[00101199] on /dev/ttyUSB0 listening on c1

So it reads out the internal id of the dongle, check if the number
is the same as printed on the dongle, it is for me. :-)

wmbusmeters im871a:t1
This will also print:
Started auto im871a[00101199] on /dev/ttyUSB0 listening on c1

And finally:
wmbusmeters im871a[00101199]:c1

So if you have two im871a and you want to listen to c1 and t1,
and they have different antennas etc, you could say:
wmbusmeters im871a[12345678]:c1 im871a[11223344]:t1

I.e. you can pinpoint each dongle to set the appropriate linkmode.

You can also do it with:
wmbusmeters /dev/ttyUSB0:im871a:c1 /dev/ttyUSB1:im871a:c1
but it might be less stable of course.

There is a lot of bugs in NewDetectionCode right now.
But I felt I had to push my work so far.

Listening to meters works like before:
wmbusmeters auto:c1 Water multical21:c1 76348799 KEY...

I kind of like the obvious visual that c1 receiver and c1 transmitter.

weetmuts, 2020-10-04:
>I've commented on the branch i pushed with some information about the last two registers.
@elbaek89 where did you push this?

>I see that the udev rules have been disabled for this branch. Is there a way for me to start the daemon and have it log in the samme manner as the automatic logging that happened on the master branch? 

Yes you do "sudo systemctl enable wmbusmeters.service" to enable the daemon at boot.
"sudo systemctrl start wmbusmeters.service" to start it rebooting.

Right now however the branch is very buggy, so please wait before you test this.

pswiatki, 2020-10-04:
> "sudo systemctrl start wmbusmeters.service" to start it rebooting.
You probably mean to start the service (run the daemon), right?

By the way: does it mean the branch could be unstable for all other meters, too? (I use izar)
I am currently @ f777f71d51f1500e9fa36cf8d8b1b3e655d6fda1 and it seems to work well.
Should I hold short of performing `git fetch `operation for now?

weetmuts, 2020-10-05:
>>"sudo systemctrl start wmbusmeters.service" to start it rebooting.
>You probably mean to start the service (run the daemon), right?

Yes, you are right, I meant that you start the service, without rebooting.

The rtlwmbus is not working, only im871a and amb8465. I will ping you
when the rtlwmbus is working again.

pswiatki, 2020-10-05:
I will stay at the commit I currently have the HEAD pointing to until further notice. Thanks!

elbaek89, 2020-10-06:
> > I've commented on the branch i pushed with some information about the last two registers.
> > @elbaek89 where did you push this?

I wasn't sure where to post my findings, so i posted it in the issue i created: 
https://github.com/weetmuts/wmbusmeters/issues/165#issuecomment-699910983

weetmuts, 2020-10-07:
Ok, now I got it working. Specify as device auto:c1 either in conf file and start daemon
or run from commandline: wmbusmeters auto:c1

You can plug in multiple rtlwmbus dongles and then remove them.
You can plug in multiple im871a/amb8465 and remove them.

wmbusmeters should stay up running. Printing a warning if no wmbus devices are found at all.
See if you can get it to crash! :-) @pswiatki Please test!

If you have multiple dongles, it will print multiple copies of the received telegrams. I will fix this.

If you have multiple im871a that can only listen to one linkmode, then you can do:
wmbusmeters im871a:c1 im871a:t1
to capture both c1 and t1 telegrams.

With amb8465 and rtlwmbus this is not necessary:
wmbusmeters auto:c1,t1
or 
wmbusmeters amb8465:c1,t1

pswiatki, 2020-10-07:
Thanks a lot, @weetmuts! I have plenty of RTL-SDR sticks. Will let you know how the deamon behaves. Hopefully I will be able to stress test it over the weekend. Will keep you posted.

pswiatki, 2020-10-12:
Quick question: is there any way to refer to a particular `rtlwmbus` instance (and rtl_sdr device) in the setup of `wmbusmeters`?

weetmuts, 2020-10-12:
@pswiatki Yes, and no. The im871a/amb8465 and probably the other hardware wmbus dongles can be
queried for which serial they will use when transmitting. This is the identify I indent im871a[12345678] to use.

rtlsdr dongles on the other hand have no unique identity.

The udev rule in Ubuntu that creates /dev/swradio0 /dev/swradio1 is somewhat sticky,
which means that if plug in rtlsdr dongle A first then B, you get /dev/swradio0 and /dev/swradio1
unplug A and /dev/swradio1 is still there referring to B. Only when you reboot your linux will it renumerate
and potentially give new numbers to the devices.

Unfortunately the rtlsdr api is not as clever. You can ask for a list of rtlsdr dongles using the api
and you get them ordere 0,1,2 etc unfortunately these numbers are not sticky,
which that when you unplug an rtlsdr device no identity is valid anymore.
My current code solves this by closing all rtlwmbus devices and reopening them again
if wmbusmeters detect that one rtlwmbus device was lost.

So, I do have added support for libusb, and theoretically we could implement ourself the equivalent
of the Ubuntu udev rule within wmbusmeters, this would give us sticky identies like 0,1,2.
Or we add the Ubuntu udev rule on other Linux versions like Arch, when we install wmbusmeters.

Or I just add support for rtlwmbus[0] and rtlwmsbus[1] and you will get whatever happens
to land on 0 and 1, but it is very unclear how stable this is. Perhaps it is stable enough
assuming that you will reboot after each time you  plugin/unplug your rtlsdr dongles.
But I think it will be so confusing that it is not really usable.

What do you think?

pswiatki, 2020-10-12:
I believe `rtlsdr` library API allows the user to query e.g. the serial string in a given device. I have to check to make sure. If this is the case, `wmbusmeters` could allow some sort of key/value pair to be used to look up the match on the list of `rtlsdr` devices to use for a given meter. E.g.  key: "serial", value: "serial number 23" (which is a string). Question is whether the list of keys is stable in the library for `rtlsdr`, I would have to check that as well. What do you think?

weetmuts, 2020-10-12:
Yes, please test, the code is in rtlsdr.cc 
the function rtlsdr_get_device_usb_strings(i, mfct, product, serial);
but serial seems to always be 00000001, at least for me.

pswiatki, 2020-10-12:
That's OK, it can be always the same string (on devices as shipped - never modified in any way). Therefore, this key/value pair should only be used as a possible option in the meter setup, but not mandatory. If not specified - `wmbusmeters` could take the very first rtlsdr device as returned by the `rtlsdr` lib, but if present in the meter setup - it would be up to the user to ensure there is a match (or no readings would be accepted, as the specified interface would be not present). One can use a tool from the `rtlsdr` package (`rtl-eeprom`) that can be used to change e.g. the serial string of a device. I did just that on my devices and now I can easily recognize them in several applications I use those sticks with.

weetmuts, 2020-10-12:
Yes!!! rtl_eeprom set serial works perfectly on my device. 
Then for sure I can make wmbusmeters use the serial to distinguish between devices.

pswiatki, 2020-10-12:
Perfect! Glad I could be of some help. By the way: is there any way to somehow forward the information on the signal quality from **`rtlwmbus`** to **`wmbusmeters`**? You see: for some reason I can read plenty of water meters from the entire neighbourhood without any issues (some of them have really strong signal with good SNR, so easily readable from inside the house without any special equipment), but my own water meter produces only a weak signal and even with a 10-element yagi-uda antenna pointed directly at the spot in the garden where the meter is buried under ground (at least 2.5 meters deep!) I get only marginal signal from it. So, your **`wmbusmeters`** gets telegrams nicely from it with the hardware I have, but for whatever reason (weather, or what not) sometimes the traffic (just from this one meter) stops completely for hours at the time and no readings are seen/registered (meanwhile, I can see plenty of telegrams from other meters, which are rejected - **`"not for me"`**)

weetmuts, 2020-10-12:
Ok, I have now pushed support for using the rtlsdr device serial. I have set my two devices to 4422 and 4711.
So I can now write:
wmbusmeters rtlwmbus[4422]:c1,t1 rtlwmbus[4711]:434M:t1 
You can plug in the 434M antenna into 4711 and the 868M antenna into 4422.
(There is at least one user you has listened to wmbus telegrams on 434M using rtlwmbus/wmbusmeters.)

weetmuts, 2020-10-12:
Yes, I have intended to add the rssi field to the telegrams.
Also I have intended to a filtering function that will detect duplicated received telegrams and discard the duplicates.

But you might want to have the opposite, ie all telegrams received and we add two json fields:
recieving_device="rtlwmbus[4422]" rss_db=-12.4
(or whatever the rssi unit and value should be, please let me know if you know.)
then you can plugin in several rtlwmbus dongles with different antennas
and check the rssi for each device. :-)

pswiatki, 2020-10-13:
I will have to look at **`rtlwmbus`** code to make sure what really gets measured there. It may be the raw RF level or perhaps the signal to noise ratio, or SNR (which I am not so sure is easy to do as there could always be stations transmitting in the channel, so I wonder how the noise level could be estimated/measured). In the former case (RF level) in general the unit might be dBm (the reference level being 1 mW), in the latter (SNR) - just dB. I am pretty sure the RF level cannot be easily expressed in real units as the AGC is probably set to auto by **`rtlwmbus`**, so it is difficult to get the reference for the interpretation of sample values. I will comment a bit more when I check to code, though.

weetmuts, 2020-10-14:
Just pushed to master.
@pswiatki I have added two new values in the json, for example:
"device":"im871a[1234]", "rssi_dbm"=-88

The rssi:s are implemented now for im871a,amb8465 where the translations from bytes in the protocol to dBm
is described in the manuals.

As you say, the rtlwmbus is unknown, perhaps we will never be able to actually translate it into an even reasonbly
correct dBm. So now its just report as +1 to +255, hovering around +125 , which if you place the suffix dBm
after is physically silly. 

Maybe the rssi from such dongles should be reported as just "rssi":123
and where the dongle can actually go to dBm "rssi_dbm":-80
What do you think?

pswiatki, 2020-10-14:
Makes sense, although I shall report on **`rtlwmbus`** when I know more. Do you know if some kind of C/N oraz SNR is also available for imXX or ambXX dongles? It makes a lot of sense to also monitor those to have some idea of the signal quality.

weetmuts, 2020-10-14:
Sorry, no push to master. Push to NewDetectionCode of course. I have not seen any mention of snr or cnr in the manuals, but i will check again.

weetmuts, 2020-10-25:
Ok, now merged NewDetectionCode to master. Bumped version number to 1.0.0.
Please read the CHANGES document for the important changes. :-)

slawekzz, 2020-10-26:
After uploading version 1.0.0 wmbusmeter stopped working. On another port, / dev / ttyUSB0, I have a FTDI USB Serial Device converter. After starting wmbusmeters, the transmission on the ttyUSB0 port is interrupted and the device stops working. Call from the command line rtl_sdr -f 868.95M -s 1600000 - | rtl_wmbus works correctly and shows the reading from the water meter. Calling from the wmbusmeters service ends with disconnecting the ttyUSB0 port and in the wmbusmeter.log log there is:
(wmbusmeters) waiting for telegrams
(memory) rss 3710 976 peak 3.53 MiB
(alarm SpecifiedDeviceNotFound) the device rtlwmbus: 868.9M is not working
(alarm SpecifiedDeviceNotFound) the device rtlwmbus: 868.9M is not working
in wmbusmeters.conf calls the device:
device = rtlwmbus: 868.9M
FTDI USB Serial Devices detachment does not affect wmbusmeters operation

weetmuts, 2020-10-26:
@slawekzz Can you check that the user wmbusmeters is a member of plugdev?

If you stop the daemon (sudo systemctl stop wmbusmeters.service)
and run:
wmbusmeters --debug rtlwmbus:868.9M
what is the output?

slawekzz, 2020-10-26:
wmbusmeters --debug rtlwmbus:868.9M
(wmbusmeters) version: 1.0.0
(config) using device: rtlwmbus:868.9M
(config) number of meters: 0
(main) device /dev/ttyS0 not currently used, detect contents...
(serial) before config: /dev/ttyS0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) EVENT thread interrupted
(serial) after config:  /dev/ttyS0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS0 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyS0) sent "FF0A02008077"
(serial /dev/ttyS0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyS0 (detect amb8465)
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS0 fd 3 (detect rc1180)
(serial /dev/ttyS0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyS0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyS0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a) are you there? no.
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyS0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B115200) input(IGNPAR) output() control(CLOCAL|CREAD|CSIZE|CSTOPB|HUPCL) local() special_chars(1) modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 5 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "FF0A02008077FF0A02008077"
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 5 (detect rc1180)
(serial /dev/ttyUSB0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary "00"
(serial) EVENT thread interrupted
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 5 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5010500"
[... 26 lines cut ...]
(rtlwmbus) received full frame
(wmbus) parseDLL @0 111
(telegram) DLL L=6e C=44 (from meter SND_NR) M=0601 (APA) A=00931285 VER=05 TYPE=07 (Water meter) (driver apator162) DEV=rtlwmbus[00000001] RSSI=71
(wmbus) parseELL @10 101
(wmbus) parseAFL @10 101
(wmbus) parseTPL @10 101
(telegram) TPL CI=7a ACC=67 STS=00 CFG=8560 (bidirectional AES_CBC_IV nb=6 cntn=0 ra=0 hc=0)
Received telegram from: 00931285
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Water meter (0x07)
                   ver: 0x05
                device: rtlwmbus[00000001]
                  rssi: 71 dBm
                driver: apator162
(wmbus) 00: 6e length (110 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 0106 dll-mfct (APA)
(wmbus) 04: 85129300 dll-id (00931285)
(wmbus) 08: 05 dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: 67 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 6085 tpl-cfg 8560 (bidirectional AES_CBC_IV nb=6 cntn=0 ra=0 hc=0 )
(wmbus) "telegram=||6E4401068512930005077A67006085EA894759686AA0331D38084DB70D0D4008341EDE7A6D139607BFE5DEFF2EC745B8C6CE6296EC15BAE079FDC883555DDAE0E797D75D0A0C80C7E8F3A23B5350FEDF08B0776DAF098B50739ABF53F9725D5357E1C6008E9840D716281F7160E1EA|+41"

slawekzz, 2020-10-26:
> Can you check that the user wmbusmeters is a member of plugdev?

no, user wmbusmeters is not in the plugdev group

weetmuts, 2020-10-26:
Ok @slawekzz I have pushed a fix to master. The new install script makes sure the wmbusmeters user is in the plugdev
group. If its not in that group, then the daemon cannot get access to the rtlsdr dongle.
But it works when you run with your own user, since that user always has plugdev access.

slawekzz, 2020-10-26:
ok, it works. What about disrupting devices on other ports? e.g. ttyUSB0

weetmuts, 2020-10-26:
So, it must probe the serial devices. Wmbusmeters sends different byte strings and listens for answers to detect the dongles. If no appropriate answer is found, then the serial tty is ignored and will not be probed again, until it detects
that it has disappeared and re-appeared.

The modem-manager does something similar, sending bytes to try to detect if there is a modem there.
Most devices should survive this..... :-)

slawekzz, 2020-10-26:
the problem is that after wmbusmeter detects, I have to reboot the device on ttyUSB0. is it possible to indicate which device should be used? or ports that should not be tested?

weetmuts, 2020-10-27:
@slawekzz Yes, you can now use /dev/ttyUSB0:im871a:c1 as a device and this will prevent
wmbusmeters from probing other ttys and rtlsdr. However I am not sure what happens
when you say just rtlwmbus:c1,t1 
I think it might trigger an auto-probe of tty devices, will check that.

It should be easy to add --donotprobe=/dev/ttyUSB0 
to avoid probing such a device.

Is your serial thingy that gets upset used by a program/daemon?
Does that software try to flock the /dev/ttyUSB0 ? wmbusmeters
does use  flock and should avoid ttys which it cannot lock.

Perhaps it gets upset because wmbusmeters changes the baud/bps
for the serial port and does not restore it....

weetmuts, 2020-11-12:
@slawekzz Several bug fixes pushed!

If you specify just: wmbusmeters rtlwmbus:c1,t1
then it will not probe serial devices.
If you specify /dev/ttyUSB0:im871a:c1
then it will not probe serial devices.

If specifying auto or im871a (ie a tty dongle without /dev/xxx)
then it will probe all ttys to find all im871a:s.

If you only want to start one im871a you type im871a[12345678]:c1
to select that one. But it still needs to probe all ttys to find it.
  
Please test!

weetmuts, 2020-12-01:
It seems to be working for me, closing this for now.

closed 2020-12-01 by weetmuts

makuser, 2021-06-22:
So what seems to be the current new method to run multiple dongles?

weetmuts, 2021-06-22:
Just plug them in! Should work out of the box when you use auto as the driver.

weetmuts, 2021-06-22:
The code has been in the main branch a long time now. I think I bumped the release to version 1.0.0 to signify this big change.