wmbusmeters/wmbusmeters #171: Deamon not getting saving any readings.

Kosygor, 2020-10-09:


weetmuts, 2020-10-10:
Can you try this using the NewDetectionCode branch which will be the
new master in a few weeks.

Run ./configure; make; sudo make install
Now there only a single daemon, makes it easier to stop and start.

Set the device to auto:c1,t1 t
since your are using rtlwmbus which can detect both c1 and t1 at the same time.

To start and stop.
sudo systemctl start wmbusmeters.service
To enable/start at boot.
sudo systemctl enable wmbusmeters.service

weetmuts, 2020-10-10:
If you need to adjust the frequency from 868.95M to 868.9M, you should use this device instead:

rtlwmbus:868.9M:c1,t1

Kosygor, 2020-10-10:
Thank You for Your response
If 
`sudo git clone -b NewDetectionCode https://github.com/weetmuts/wmbusmeters.git`
is right command i did that.Since i do not know how to use uninstall.sh it gave me an error i jus delated previous embusmeters foders. 
I can see change in htop i have three processes called:
`/usr/sbin/wmbusmetersd /run/wmbusmeters/wbusmeters.pid`

sudo systemctl status wmbusmeters.service

`pi@raspberrypi:~ $ sudo systemctl status wmbusmeters.service
● wmbusmeters.service - "wmbusmeters service (no udev trigger)"
   Loaded: loaded (/etc/systemd/system/wmbusmeters.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-10-10 13:41:29 CEST; 54s ago
     Docs: https://github.com/weetmuts/wmbusmeters
           man:wmbusmeters(1)
  Process: 721 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings (code=exited, status=0/SUCCESS)
  Process: 723 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 726 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 731 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 734 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited, status=0/SUCCESS)
 Main PID: 736 (wmbusmetersd)
    Tasks: 3 (limit: 3862)
   CGroup: /system.slice/wmbusmeters.service
           └─736 /usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid

paź 10 13:41:28 raspberrypi systemd[1]: Starting "wmbusmeters service (no udev trigger)"...
paź 10 13:41:29 raspberrypi wmbusmetersd[734]: (wmbusmeters) started /run/wmbusmeters/wmbusmeters.pid
paź 10 13:41:29 raspberrypi systemd[1]: Started "wmbusmeters service (no udev trigger)".

`

weird thing that in `/etc/wmbusmetrs.sd` is now duplicating meter firle *.save

still i do not have any files in `/var/log/wmbusmeters/meter_readings/`

weetmuts, 2020-10-10:
set
loglevel=debug
restart daemon and
tail -f /var/log/wmbusmeters/wmbusmeters.log

Kosygor, 2020-10-10:
it gave:
`(shell) arg "wmbusmeters"
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 1
(shell) 749 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
(serial) waiting for stop
(serial) EVENT thread interrupted
`

weetmuts, 2020-10-10:
Ok, stop the daemon. Unplug your rtlsdr dongle. Test with something simple.
Start wmbusmeters from the command line like this:
wmbusmeters auto:c1 
It prints:
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!

Now insert your rtlsdr dongle, and it will print:
Started auto rtlwmbus on 0_Realtek_RTL2838UHIDIR_00000001 listening on c1

Does this happen?

Kosygor, 2020-10-10:
`sudo systemctl stop wmbusmeters.service`
freezes on empty line
but htop does not show any wmbusmeters service active do i think it works

`wmbusmeters auto:c1` prints:
```
No wmbus device detected, waiting for a device to be plugged in.
Notice! Wmbusmeters daemon (pid 742) is running and it might hog any wmbus devices.
No meters configured. Printing id:s of all telegrams heard!
```
after pluging dongle 
`Started auto rtlwmbus on 0_Realtek_RTL2838UHIDIR_00000001 listening on c1`
htop shows only running only: `wmbusmeters auto:c1`

weetmuts, 2020-10-10:
So to get rid of the daemon:
sudo kill -9 742
sudo killall -9 rtl_sdr
sudo killall -9 rtl_wmbus

weetmuts, 2020-10-10:
Since it printed "Started...", that means that it is listening on 868.95MHz ! Yay!
Now run:
wmbusmeters rtlwmbus:868.9M:c1
this will tune your receiver (rtlwmbus) to this fq instead of 868.95M which is the default.
Wait and see if you get any received telegrams.
(There is a silly alarm logging something about lost device, which is a bug, I will fix that, ignore it for the moment.)

Kosygor, 2020-10-10:
```
sudo kill -9 742
sudo killall -9 rtl_sdr
sudo killall -9 rtl_wmbus
```
Gives `there is no such process` and `could not find such process. `
I have not seen them since installing branch.

`wmbusmeters rtlwmbus:868.9M:c1` prints
```
Received telegram from: 74159398
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
           device type: Cold water meter (0x16)
            device ver: 0x1b
         device driver: multical21
(alarm) device_not_found: rtlwmbus:868.9M:c1
Received telegram from: 74159398
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
           device type: Cold water meter (0x16)
            device ver: 0x1b
         device driver: multical21
(alarm) device_not_found: rtlwmbus:868.9M:c1
```

I am sorry to boter You. I am trying to figure something on my own reading logs etc. but I only acassionaly move outside docker with Home Assistant and therefore I do not have much expierience with linux.

weetmuts, 2020-10-11:
Ah, there are known difficulties with the rtlsdr and docker and the current wmbusmeters master. 
I am not an expert on the docker stuff, but @BIBOLV is an expert. :-)
 
The NewDetectionCode branch is better since it does not require specifically named /dev/ files
for detection. We intend to make better docker version when this branch is merged.

Can you remove your wmbusmeters docker and run as a plain daemon instead?

//Fredrik

Kosygor, 2020-10-11:
Im not an expert but since it it running on raspbian buster and I have installed it via ssh it is not in docker. 
Am I right ? 
Only thing that i have In docker is Home Assistant and accompanying addons .

BIBOLV, 2020-10-11:
@Kosygor looking at your previous comment, looks like you are receiving meter readings ok, so now you just need to adjust configuration to adopt what you are using in command line.
But have you tried running wmbusmeters docker image and for HA integration [this](https://weetmuts.github.io/wmbusmeterswiki/HASS.html)?

Kosygor, 2020-10-11:
@BIBOLV 
This is exactly what i was trying to do but neither of device configuration worked for me.
Yest I have tried 2 ways of getting data to HA: 
- one You mentioned ;
- https://github.com/zibous/ha-watermeter;
But I still belive there is something wrong im my particular case because there are no readings inside /var/log/wmbusmeters/meter_readings directory. 

No I did not. I just found it out after reading Your post. 
Hope I will menage to install It. I`m currently far from home and have not made any ssh access from outside. I will start testing on Friday. 
Sorry for being little clumsy I am doing my best :) .

BIBOLV, 2020-10-11:
No worries, I`m sure we will get you up and running sooner or later :wink:
When running docker image read trough readme carefully as you are running rtl_sdr and it needs specific docker config.
Let us know how it goes.

weetmuts, 2020-10-12:
Ok, I have pushed a fix for the buggy alarm when you specify the device. Please compile and install.

Stop the daemon, sudo systemctl stop wmbusmeters.service
Do:
sudo rm /var/log/wmbusmeters/wmbusmeters.log
then do 
sudo systemctl start wmbusmeters.service

email me the full  /var/log/wmbusmeters/wmbusmeters.log
email me the /etc/wmbusmeters.conf 
(you can cut out the key if you like)

Thanks!

Kosygor, 2020-10-15:
I just got back to master branch left everything intect except for device configuration (initially I was  using conf from  [ha-watermeters](https://github.com/zibous/ha-watermeter/blob/master/docs/build-wmbusmeters.md) on it is running perfectly stable now. Just one thing left rtl-wmbus is using 40-100% of cpu so I am trying to clean up all copies that i made.

closed 2020-10-20 by Kosygor