wmbusmeters/wmbusmeters #173: Not a valid meter type ""

hlnstepanova, 2020-10-26:
I've been trying to run wmbusmeters with a config file on MacOS and keep getting this message
@Lenas-MacBook-Pro ~ % wmbusmeters --useconfig=/
Not a valid meter type ""
Not a valid meter id nor a valid meter match expression ""

I'm very new to this and I think I have messed up the paths somehow, so that the program cannot find the meter file. Or may be I have wrong config files.

wmbusmeters.conf in /etc/
loglevel=normal
device=/dev/tty.usbserial-00B93EB9:im871a
listento=t1
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
meterfilesnaming=name
meterfilestimestamp=day
logfile=/var/log/wmbusmeters/wmbusmeters.log

MySonexHCA in /etc/wmbusmeters.d/
name=MySonexHCA
type=fhkvdataiii
id=22334464
key=NOKEY

It's also probably worth noting that I ran "make" to build the program, but then there was no appropriate folders created in /etc/ and /var/log/, so I created them myself and also added the executable wmbusmeters to usr/local/bin

I'd really appreciate some help with solving how I can run the program with config files

weetmuts, 2020-10-26:
Does it work without config files? E.g.
wmbusmeters /dev/tty.usbserial-00B93EB9:im871a:t1 
and
wmbusmeters /dev/tty.usbserial-00B93EB9:im871a:t1 MySonexHCA fhkvdataiii 22334464 NOKEY
If it does not, then please add --debug

hlnstepanova, 2020-10-26:
I get
wmbusmeters /dev/tty.usbserial-00B93EB9:im871a:t1
Unknown device suffix im871a:t1

But this works
wmbusmeters --t1 /dev/tty.usbserial-00B93EB9:im871a

weetmuts, 2020-10-26:
Please upgrade to the latest master on github.
Then we can debug more.

hlnstepanova, 2020-10-26:
After updating:

wmbusmeters /dev/tty.usbserial-00B93EB9:im871a:t1
Started config im871a on /dev/tty.usbserial-00B93EB9 listening on t1
Warning! Did not get confirmation on set link mode for im871a
No meters configured. Printing id:s of all telegrams heard!

But this time I see no telegrams. The same with this

wmbusmeters /dev/tty.usbserial-00B93EB9:im871a:t1 MySonexHCA fhkvdataiii 22334464 NOKEY
Started config im871a on /dev/tty.usbserial-00B93EB9 listening on t1
Warning! Did not get confirmation on set link mode for im871a

weetmuts, 2020-10-26:
Ah, found and fixed the bug that caused this problem. Fix pushed to master.
Please test again.

hlnstepanova, 2020-10-27:
Now I get data with both commands, for example
wmbusmeters /dev/tty.usbserial-00B93EB9:im871a:t1 MySonexHCA fhkvdataiii 22334464 NOKEY
Started config im871a on /dev/tty.usbserial-00B93EB9 listening on t1
(meter) MySonexHCA: meter detection did not match the selected driver fhkvdataiii! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
MySonexHCA	22334464	0 hca	2020-12-06T02:00:00Z	17399 hca	2048-11-12T02:00:00Z	228.66 °C	0 °C	2020-10-27 09:11.50

But when using 
wmbusmeters --useconfig=/ 
still no reaction, no files changed in /var/log/wmbusmeters

hlnstepanova, 2020-10-27:
I have just noticed changes in the var/log/, probably it took time for the files to refresh
So, apparently, now everything works with config!
Thank you for the help, @weetmuts

weetmuts, 2020-10-27:
Glad to hear that it works now! 

You can always get more information by setting
loglevel=verbose
or 
loglevel=debug
in the wmbusmeters.conf file

This extra info should be logged in /var/log/wmbusmeters/wmbusmeters.log

or use --verbose --debug on the command line.

weetmuts, 2020-10-27:
If you have the time, run with loglevel=debug for a while
and collect one of your telegram=|.....|
lines and email to me. It contains a new telegram version/type/mfct for the fhkvdataiii
the warning notifies us that your meter is not in the known list of meters yet.
Though it could still be compatible with the fhkvdataiii driver.

Remember to disable the debug mode when running for a long time, it generates a lot of output. :-)

hlnstepanova, 2020-10-31:
emailed you the logs, let me know if any other info is needed

weetmuts, 2020-10-31:
Thanks for the logs! I added the new heat cost allocator sontex868, which is the one you have.

closed 2020-11-12 by weetmuts