wmbusmeters/wmbusmeters #174: Amiplus device version 00

DeDragonSlayer, 2020-10-26:
Hi, so I finally managed to activate wmbus on my Tauron provided power meter. I installed wmbusmeters, and the decoding works fine, but the particular mfct, media, version combination is not on the whitelist, and wmbusmeters suggested to open an issue here regarding that, which is what I'm doing. :)
The meter parameters are: M=10b6 (DEV) VER=00 TYPE=02
Below is some hopefuly useful debug output:

```
(wmbusmeters) version: 1.0.0-9-gf4ad901
(config) number of meters: 1
(main) configured "Prad" "amiplus" "00254358" encrypted
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(main) rtlsdr device 00000001 not currently used.
(main) using globally set link modes t1
Started auto rtlwmbus[00000001] listening on t1
(rtlwmbus) using command: rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh pid 2074 fd 7 (rtlwmbus)
(main) regular reset of RTLWMBUS  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for RTLWMBUS
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) received ascii "T1;1;1;2020-10-26 

08:13:59.000;130;138;00254358;0x4e44b6105843250000027a69004005611b068732ea30454069c2070bbef7a54c094c18c86bcacd619afaefba18af58ca8e5c99e75876baa662fd2b5d1159378c360ffda48f190a59ed15ca8cb0d379<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-10-26 

08:13:59.000;130;138;00254358;0x4e44b6105843250000027a69004005611b068732ea30454069c2070bbef7a54c094c18c86bcacd619afaefba18af58ca8e5c99e75876baa662fd2b5d1159378c360ffda48f190a59ed15ca8cb0d379<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 79
(meter) Prad: for me? 00254358
(meter) Prad: meter detection did not match the selected driver amiplus! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Prad: yes for me
(meter) Prad amiplus handling telegram from 00254358
(meter) Prad 00254358 "4E44B6105843250000027A69004005611B068732EA30454069C2070BBEF7A54C094C18C86BCACD619AFAEFBA18AF58CA8E5C99E75876BAA662FD2B5D1159378C360FFDA48F190A59ED15CA8CB0D379"
(wmbus) parseDLL @0 79
(telegram) DLL L=4e C=44 (from meter SND_NR) M=10b6 (DEV) A=00254358 VER=00 TYPE=02 (Electricity meter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=130
(wmbus) parseELL @10 69
(wmbus) parseAFL @10 69
(wmbus) parseTPL @10 69
(TPL) decrypting "611B068732EA30454069C2070BBEF7A54C094C18C86BCACD619AFAEFBA18AF58CA8E5C99E75876BAA662FD2B5D1159378C360FFDA48F190A59ED15CA8CB0D379"
(TPL) num encrypted blocks 4 (64 bytes and remaining unencrypted 0 bytes)
(TPL) IV B6105843250000026969696969696969
(TPL) decrypted  "2F2F0C7835221400066D014E099A2A400E031562420100000E833C0000000000000B2B2909000BAB3C0000000AFDC9FC0139020BABC8FC100000002F2F2F2F2F"
(dvparser) found new format "0C78066D0E030E833C0B2B0BAB3C0AFDC9FC010BABC8FC10" with hash 5b12, remembering!
(telegram) TPL CI=7a ACC=69 STS=00 CFG=0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0)
(amiplus) log "telegram=|4E44B6105843250000027A690040052F2F|0C7835221400066D014E099A2A400E031562420100000E833C0000000000000B2B2909000BAB3C0000000AFDC9FC0139020BABC8FC100000002F2F2F2F2F|+56"
(amiplus) 00: 4e length (78 bytes)
(amiplus) 01: 44 dll-c (from meter SND_NR)
(amiplus) 02: b610 dll-mfct (DEV)
(amiplus) 04: 58432500 dll-id (00254358)
(amiplus) 08: 00 dll-version
(amiplus) 09: 02 dll-type (Electricity meter)
(amiplus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(amiplus) 0b: 69 tpl-acc-field
(amiplus) 0c: 00 tpl-sts-field
(amiplus) 0d: 4005 tpl-cfg 0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
(amiplus) 0f: 2f2f decrypt check bytes
(amiplus) 11: 0C dif (8 digit BCD Instantaneous value)
(amiplus) 12: 78 vif (Fabrication no)
(amiplus) 13: 35221400
(amiplus) 17: 06 dif (48 Bit Integer/Binary Instantaneous value)
(amiplus) 18: 6D vif (Date and time type)
(amiplus) 19: * 014E099A2A40 device datetime (2020-10-26 09:14)
(amiplus) 1f: 0E dif (12 digit BCD Instantaneous value)
(amiplus) 20: 03 vif (Energy Wh)
(amiplus) 21: * 156242010000 total energy (1426.215000 kwh)
(amiplus) 27: 0E dif (12 digit BCD Instantaneous value)
(amiplus) 28: 83 vif (Energy Wh)
(amiplus) 29: 3C vife (backward flow)
(amiplus) 2a: * 000000000000 total energy returned (0.000000 kwh)
(amiplus) 30: 0B dif (6 digit BCD Instantaneous value)
(amiplus) 31: 2B vif (Power W)
(amiplus) 32: * 290900 current power (0.929000 kw)
(amiplus) 35: 0B dif (6 digit BCD Instantaneous value)
(amiplus) 36: AB vif (Power W)
(amiplus) 37: 3C vife (backward flow)
(amiplus) 38: * 000000 current power returned (0.000000 kw)
(amiplus) 3b: 0A dif (4 digit BCD Instantaneous value)
(amiplus) 3c: FD vif (Second extension of VIF-codes)
(amiplus) 3d: C9 vife (10^0 Volts)
(amiplus) 3e: FC vife (Reserved)
(amiplus) 3f: 01 vife (Credit of 10^-2 of the nominal local legal currency units)
(amiplus) 40: 3902
(amiplus) 42: 0B dif (6 digit BCD Instantaneous value)
(amiplus) 43: AB vif (Power W)
(amiplus) 44: C8 vife (upper limit)
(amiplus) 45: FC vife (additive correction constant: unit of VIF * 10^-3)
(amiplus) 46: 10 vife (?)
(amiplus) 47: 000000
(amiplus) 4a: 2F skip
(amiplus) 4b: 2F skip
(amiplus) 4c: 2F skip
(amiplus) 4d: 2F skip
(amiplus) 4e: 2F skip
Prad    00254358        1426.215 kWh    0.929 kW        0 kWh   0 kW    2020-10-26 09:13.59

pswiatki, 2020-10-26:
Just a quick question: how did you manage to get the key necessary to decrypt the telegrams from Tauron? Is it doable at all? By the way: I pick up plenty of devices (mostly water meters) on 868.95 MHz in my neighbourhood.  No telegram I could suspect as coming from my power meter (also ami+ from Tauron). Does wireless broadcast have to be enabled on the meter by Tauron?
Finally: those readings that actually get decoded in your case - do they match the reading on the meter, or are completely false?

majekw, 2020-10-26:
@pswiatki you need to login into https://elicznik.tauron-dystrybucja.pl/, click on 'Licznik / HANplus', click 'Aktywuj HAN' and wait even day/two to enable. After successful enable in the same place you will get decryption key and meter should start broadcast readings.

pswiatki, 2020-10-26:
Nice. I will do just that. The only issue is that the owners (and account users) are located overseas, so it might be a bit difficult to get it done. We'll see - maybe I can manage to convince them to follow this procedure. Thanks a lot!
@majekw what are your results with ami+ and wmbusmeters? Can I successfully get at least the basic readings (1.8.0 for example)?

DeDragonSlayer, 2020-10-26:
@pswiatki - in answer to your earlier question - yes, the readings match what's on my meter. :) The only issue is what I reported, i.e. for example running the following command:
`~/wmbusmeters/build/wmbusmeters --usestdoutforlogging auto:t1 Prad amiplus 00254358 XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX`

I get some warnings in the output as below:
```
Started auto rtlwmbus[00000001] listening on t1
(meter) Prad: meter detection did not match the selected driver amiplus! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
Prad    00254358        1428.485 kWh    0.631 kW        0 kWh   0 kW    2020-10-26 12:22.58
```
I guess that impacts only automatic meter detection.

Regarding getting the key from Tauron, it took me months(!) of clicking through their web page, only to get "HAN activation request - unsuccessful" response each time. I guess they had some bug on their back end, possibly related to the fact that my meter was replaced a few months ago. Looks like they fixed it now.

pswiatki, 2020-10-26:
@DeDragonSlayer Thanks for your confirmation. I let the owners know about the procedure - will see if they help me out with it or not. My meter hasn't been changed recently, so - I hope! - everything will go smoothly.

majekw, 2020-10-26:
As for activation, my meter was also replaced recently and I had to open ticket in Tauron for fixing this, so you need wait few days, check if it works and then push them and fill complaints.

@weetmuts, is it also possible to add decoding of: dif 0a, vif FD C9 FC 01 (current voltage)? Just to remind https://github.com/weetmuts/wmbusmeters/issues/13#issuecomment-479968995 :-p

weetmuts, 2020-10-26:
Thanks @DeDragonSlayer for the new meter that is Tauron/Amiplus compatible.

I have added the new mftct,type,version to the detection in meters.h
It is not quite a whitelist, it helps to find the proper driver for a telegram that arrives
when you run "wmbusmeters auto:t1" it will print things like:

Received telegram from: 00254358
          manufacturer: (DEV) Develco Products, Denmark (0x10b6)
                  type: Electricity meter (0x02)
                   ver: 0x00
                driver: amiplus

weetmuts, 2020-10-26:
@majekw I know, the voltage decoding is part of the dif vif parser rewrite! 
It has been postponed during the major rewrite of the detection code. :-)
It is my next task.

DeDragonSlayer, 2020-10-26:
Thanks @weetmuts for your fast response! Works fine now.

closed 2020-10-26 by DeDragonSlayer

pswiatki, 2020-10-28:
@DeDragonSlayer @majekw Something went perfect, but other thing went wrong. First: I got the WMbus activated on my meter in just.... 17 seconds (time it took Tauron to activate the wireless transmission). 
However, I get this:

```
(serial) received ascii "T1;1;1;2020-10-28 11:39:01.000;122;118;00109314;0x4e44b6101493100002377a740040059efcf287c54f247452ba76c4fec19d89bafc12d06daf52ed8ea701df38e92babe
f16991dc2002a975c450392ec8252af515dc6db39c4076417d613e842bb830c<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-10-28 11:39:01.000;122;118;00109314;0x4e44b6101493100002377a740040059efcf287c54f247452ba76c4fec19d89bafc12d06daf52ed8ea701df38e
92babef16991dc2002a975c450392ec8252af515dc6db39c4076417d613e842bb830c<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 79
(meter) P15_Prund: for me? 00109314
(meter) P15_Prund: yes for me
(meter) P15_Prund amiplus handling telegram from 00109314
(meter) P15_Prund 00109314 "4E44B6101493100002377A740040059EFCF287C54F247452BA76C4FEC19D89BAFC12D06DAF52ED8EA701DF38E92BABEF16991DC2002A975C450392EC8252AF515DC6DB39C40764
17D613E842BB830C"
(wmbus) parseDLL @0 79
(telegram) DLL L=4e C=44 (from meter SND_NR) M=10b6 (DEV) A=00109314 VER=02 TYPE=37 (Radio converter (meter side)) (driver amiplus)
(wmbus) parseELL @10 69
(wmbus) parseAFL @10 69
(wmbus) parseTPL @10 69
(TPL) decrypting "9EFCF287C54F247452BA76C4FEC19D89BAFC12D06DAF52ED8EA701DF38E92BABEF16991DC2002A975C450392EC8252AF515DC6DB39C4076417D613E842BB830C"
(TPL) num encrypted blocks 4 (64 bytes and remaining unencrypted 0 bytes)
(TPL) IV B6101493100002377474747474747474
(TPL) decrypted  "AC934DEAAB8258F3D6723B7393B1C5C22DFAA3F9ADF0FA8958F4C4AA6C37692705725F1297A1BFEE2DA6D9EDA72491566A6A14501188E046EF4121C6A6B3AAC4"
(wmbus) decrypted content failed check, did you use the correct decryption key? Ignoring telegram.
(telegram) TPL CI=7a ACC=74 STS=00 CFG=0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0)
```
Just double-checked. The key is exactly as displayed in the HANplus tab on Tauron website.
Is there anything I did incorrectly?

pswiatki, 2020-10-28:
Hmmm.... I see this fundamental difference:
In @DeDragonSlayer 's dump:
`(telegram) DLL L=4e C=44 (from meter SND_NR) M=10b6 (DEV) A=00254358 VER=00 TYPE=02 (Electricity meter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=130`

in my case it is:
`(telegram) DLL L=4e C=44 (from meter SND_NR) M=10b6 (DEV) A=00109314 VER=02 TYPE=37 (Radio converter (meter side)) (driver amiplus)`

Something is definitely different.

DeDragonSlayer, 2020-10-28:
@pswiatki That should not be an issue, it has been discussed that some of these meters have the radio built in, some have it as a separate module, hence the difference.

I still think something is wrong with the decryption key in your case. If there is more than one meter under your account on Tauron's site, make sure you're looking at the right one. :) No idea what else could be wrong.

pswiatki, 2020-10-28:
I have only one meter. Not good. Either they gave me the incorrect key (doubtful) or I just hit some issue within `wmbusmeters`.

weetmuts, 2020-10-28:
@DeDragonSlayer yes you are right. Its just an indication of the particular vendor type. 

The decryption code is common for all meter drivers, so it is probably working. Check the serialnr.

By the way, unknown driver? I thought I added that combo to the lookup list. Perhaps you have not pulled the latest release?

pswiatki, 2020-10-28:
Just checked the serial number and verified the key (the HEX string) with the QR code they also provide. Everything matches exactly. I will ask TAURON for assistance - maybe they can double-check on their side.

pswiatki, 2020-10-28:
@weetmuts Just to be sure: by serial no you mean the address on the WMbus, right?
This is my meter definition in `/etc/wmbusmeters.d`:
```
name=P15_Prund
type=amiplus
id=00109314
key=xxxxxxxxxxxxxxxxxxxxxxxxxxx
```
Do I need anything else?

weetmuts, 2020-10-28:
Nope, that is enough.

weetmuts, 2020-10-28:
You can run a --debug log for a while and email it to me and the key, then I can double check everything.

pswiatki, 2020-10-28:
I have been running --debug level log for quite some time. Just sent you an email with the details, @weetmuts . Thank you much for your help.

majekw, 2020-10-28:
I have very similar config and what @pswiatki provided should work. Decryption key should have 32 chars.
You can also check using this: https://github.com/CBrunsch/scambus/
In reader.py, in `keys` variable (line 16) you should add your meter like that:
'\x00\x10\x93\x14': \x...' (your key divided into bytes)
and run it like that: 
```
python reader.py -v -t '4e44b6101493100002377a740040059efcf287c54f247452ba76c4fec19d89bafc12d06daf52ed8ea701df38e92babe
f16991dc2002a975c450392ec8252af515dc6db39c4076417d613e842bb830c'
```
argument is raw telegram from rtl_wmbus without 0x at the beginning

pswiatki, 2020-10-28:
@majekw Thanks, I will try to do that.

pswiatki, 2020-10-28:
This was expected, I guess:

```
verbosity:  2
filename:  
txt:  4E44B6101493100002377A4A0040059C8B8916C9189BCF723F67C99E89F0A257E92BA5720F4AD242BEA5C0F2D3DB5529742BE0942DB2C815AD09DFCF1511B2F2AAB8E71234648DE7D547378E051A11
hex:  4E 44 B6 10 14 93 10 00 02 37 7A 4A 00 40 05 9C 8B 89 16 C9 18 9B CF 72 3F 67 C9 9E 89 F0 A2 57 E9 2B A5 72 0F 4A D2 42 BE A5 C0 F2 D3 DB 55 29 74 2B E0 94 2D B2 C8 15 AD 09 DF CF 15 11 B2 F2 AA B8 E7 12 34 64 8D E7 D5 47 37 8E 05 1A 11
dec:  AE 9C 7F 05 47 E0 21 37 42 9A 22 35 F1 8C 27 5E 4B B7 49 36 ED D5 D3 F3 9D DC BC F6 9D D4 2D C2 56 9E 6E DC B4 68 E1 CC ED B9 34 53 3A 1B B6 9B 8F 4D E0 21 7B 7C C7 72 37 31 97 73 62 2A 4B 54
AE 9C 7F 05 47 E0 21 37 42 9A 22 35 F1 8C 27 5E 4B B7 49 36 ED D5 D3 F3 9D DC BC F6 9D D4 2D C2 56 9E 6E DC B4 68 E1 CC ED B9 34 53 3A 1B B6 9B 8F 4D E0 21 7B 7C C7 72 37 31 97 73 62 2A 4B 54
Traceback (most recent call last):
  File "./reader.py", line 50, in <module>
    main(sys.argv[1:])
  File "./reader.py", line 45, in main
    frame.parse(capture, keys)
  File "/home/pswiatki/devel/scambus/wmbus.py", line 99, in parse
    raise Exception("Decryption failed")
Exception: Decryption failed 
```

Sooo..... bad key (not matching the one that actually put in the meter itself) ?

pswiatki, 2020-10-28:
Deactivated the WMbus and now reactivated it. Got a different key this time. Please, keep your fingers crossed. I will report soon.

pswiatki, 2020-10-28:
and........
**success!!**

```
(serial) received ascii "T1;1;1;2020-10-28 19:00:01.000;123;125;00109314;0x4e44b6101493100002377ac9004005e34a7e6ee6bc9e80799f47b89d15b49052f44986dfd854e7167647cbdd68e037c445bcb116486b2666b6ae1869bc0053aec189fd195ca93d3ba5c97b928c30aa<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-10-28 19:00:01.000;123;125;00109314;0x4e44b6101493100002377ac9004005e34a7e6ee6bc9e80799f47b89d15b49052f44986dfd854e7167647cbdd68e037c445bcb116486b2666b6ae1869bc0053aec189fd195ca93d3ba5c97b928c30aa<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 79
(meter) P15_Prund: for me? 00109314
(meter) P15_Prund: yes for me
(meter) P15_Prund amiplus handling telegram from 00109314
(meter) P15_Prund 00109314 "4E44B6101493100002377AC9004005E34A7E6EE6BC9E80799F47B89D15B49052F44986DFD854E7167647CBDD68E037C445BCB116486B2666B6AE1869BC0053AEC189FD195CA93D3BA5C97B928C30AA"
(wmbus) parseDLL @0 79
(telegram) DLL L=4e C=44 (from meter SND_NR) M=10b6 (DEV) A=00109314 VER=02 TYPE=37 (Radio converter (meter side)) (driver amiplus)
(wmbus) parseELL @10 69
(wmbus) parseAFL @10 69
(wmbus) parseTPL @10 69
(TPL) decrypting "E34A7E6EE6BC9E80799F47B89D15B49052F44986DFD854E7167647CBDD68E037C445BCB116486B2666B6AE1869BC0053AEC189FD195CA93D3BA5C97B928C30AA"
(TPL) num encrypted blocks 4 (64 bytes and remaining unencrypted 0 bytes)
(TPL) IV B610149310000237C9C9C9C9C9C9C9C9
(TPL) decrypted  "2F2F0C7816130600066D0040149C2A400E038912001000000E833C0100000000000B2B9903000BAB3C0000000AFDC9FC0135020BABC8FC100000002F2F2F2F2F"
(dvparser) found new format "0C78066D0E030E833C0B2B0BAB3C0AFDC9FC010BABC8FC10" with hash 5b12, remembering!
(telegram) TPL CI=7a ACC=c9 STS=00 CFG=0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0)
(amiplus) log "telegram=|4E44B6101493100002377AC90040052F2F|0C7816130600066D0040149C2A400E038912001000000E833C0100000000000B2B9903000BAB3C0000000AFDC9FC0135020BABC8FC100000002F2F2F2F2F|+28"
(amiplus) 00: 4e length (78 bytes)
(amiplus) 01: 44 dll-c (from meter SND_NR)
(amiplus) 02: b610 dll-mfct (DEV)
(amiplus) 04: 14931000 dll-id (00109314)
(amiplus) 08: 02 dll-version
(amiplus) 09: 37 dll-type (Radio converter (meter side))
(amiplus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(amiplus) 0b: c9 tpl-acc-field
(amiplus) 0c: 00 tpl-sts-field
(amiplus) 0d: 4005 tpl-cfg 0540 (AES_CBC_IV nb=4 cntn=0 ra=0 hc=0 )
(amiplus) 0f: 2f2f decrypt check bytes
(amiplus) 11: 0C dif (8 digit BCD Instantaneous value)
(amiplus) 12: 78 vif (Fabrication no)
(amiplus) 13: 16130600
(amiplus) 17: 06 dif (48 Bit Integer/Binary Instantaneous value)
(amiplus) 18: 6D vif (Date and time type)
(amiplus) 19: * 0040149C2A40 device datetime (2020-10-28 20:00)
(amiplus) 1f: 0E dif (12 digit BCD Instantaneous value)
(amiplus) 20: 03 vif (Energy Wh)
(amiplus) 21: * 891200100000 total energy (10001.289000 kwh)
(amiplus) 27: 0E dif (12 digit BCD Instantaneous value)
(amiplus) 28: 83 vif (Energy Wh)
(amiplus) 29: 3C vife (backward flow)
(amiplus) 2a: * 010000000000 total energy returned (0.001000 kwh)
(amiplus) 30: 0B dif (6 digit BCD Instantaneous value)
(amiplus) 31: 2B vif (Power W)
(amiplus) 32: * 990300 current power (0.399000 kw)
(amiplus) 35: 0B dif (6 digit BCD Instantaneous value)
(amiplus) 36: AB vif (Power W)
(amiplus) 37: 3C vife (backward flow)
(amiplus) 38: * 000000 current power returned (0.000000 kw)
(amiplus) 3b: 0A dif (4 digit BCD Instantaneous value)
(amiplus) 3c: FD vif (Second extension of VIF-codes)
(amiplus) 3d: C9 vife (10^0 Volts)
(amiplus) 3e: FC vife (Reserved)
(amiplus) 3f: 01 vife (Credit of 10^-2 of the nominal local legal currency units)
(amiplus) 40: 3502
(amiplus) 42: 0B dif (6 digit BCD Instantaneous value)
(amiplus) 43: AB vif (Power W)
(amiplus) 44: C8 vife (upper limit)
(amiplus) 45: FC vife (additive correction constant: unit of VIF * 10^-3)
(amiplus) 46: 10 vife (?)
(amiplus) 47: 000000
(amiplus) 4a: 2F skip
(amiplus) 4b: 2F skip
(amiplus) 4c: 2F skip
(amiplus) 4d: 2F skip
(amiplus) 4e: 2F skip
```

weetmuts, 2020-10-28:
Woho! Congratulations!